Koenkk/zigbee2mqtt #5371: Silvercrest HG06338-FR "undefined model"

drdada, 2020-12-22:
## Information about the device + link
Hello Koenkk,
Thank you for your AWESOME job.
I have bought 2x _Silvercrest Smart extention lead_ (from lidl) FR Plugs. Both are not recognized via using the master branch.
Since you fix this: https://github.com/Koenkk/zigbee2mqtt/issues/5204
On the dev branch, I'm able to use one but the second is still shown as unsupported.

Log seems to report same manufacturer name but with different model.
The **undefined** one (first entry) is not working
```
Zigbee2MQTT:warn  2020-12-22 13:54:14: Device '0xbc33acfffe47ee1f' with Zigbee model 'undefined' and manufacturer name '_TZ3000_vzopcetz' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
Zigbee2MQTT:warn  2020-12-22 14:21:05: Device '0xbc33acfffe453469' with Zigbee model 'TS011F' and manufacturer name '_TZ3000_vzopcetz' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

```
[image]

## data/database.db entry of the device

```
{"id":22,"type":"Router","ieeeAddr":"0xbc33acfffe47ee1f","nwkAddr":21543,"manufId":4098,"manufName":"_TZ3000_vzopcetz","powerSource":"Mains (single phase)","epList":[1,2,242,3],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,5,6],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"manufacturerName":"_TZ3000_vzopcetz","powerSource":1,"zclVersion":3,"appVersion":69,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":266,"inClusterList":[3,4,5,6],"outClusterList":[],"clusters":{},"binds":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":266,"inClusterList":[3,4,5,6],"outClusterList":[],"clusters":{},"binds":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"meta":{}}},"appVersion":69,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1608641654085}
{"id":23,"type":"Router","ieeeAddr":"0xbc33acfffe453469","nwkAddr":37854,"manufId":4098,"manufName":"_TZ3000_vzopcetz","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,2,242,3],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,5,6],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_vzopcetz","powerSource":1,"zclVersion":3,"appVersion":69,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":266,"inClusterList":[3,4,5,6],"outClusterList":[],"clusters":{},"binds":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":266,"inClusterList":[3,4,5,6],"outClusterList":[],"clusters":{},"binds":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"meta":{}}},"appVersion":69,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1608643265584}
```

StonikM, 2020-12-22:
Hello.
I have exactly the same problem, if you can solve it, I will be happy if you share it.

on4ami, 2020-12-22:
Hi,
Upgraded to v 1.16.2dev running in domoticz.
When pairing in domoticz log:
2020-12-22 17:46:47.766 Error: (Zigbee) 'onMessage' failed 'TypeError':''NoneType' object is not subscriptable'.
2020-12-22 17:46:47.766 Error: (Zigbee) ----> Line 279 in '/home/pi/domoticz/plugins/zigbee2mqtt/plugin.py', function onMessage
2020-12-22 17:46:47.766 Error: (Zigbee) ----> Line 124 in '/home/pi/domoticz/plugins/zigbee2mqtt/plugin.py', function onMessage
2020-12-22 17:46:47.766 Error: (Zigbee) ----> Line 167 in '/home/pi/domoticz/plugins/zigbee2mqtt/mqtt.py', function onMessage
2020-12-22 17:46:47.766 Error: (Zigbee) ----> Line 153 in '/home/pi/domoticz/plugins/zigbee2mqtt/plugin.py', function onMQTTPublish
2020-12-22 17:46:47.766 Error: (Zigbee) ----> Line 24 in '/home/pi/domoticz/plugins/zigbee2mqtt/devices_manager.py', function set_devices

ZIGBEE2MQTT log:

info  2020-12-22 17:46:23: Device '0xbc33acfffe479c47' joined
info  2020-12-22 17:46:23: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xbc33acfffe479c47","ieee_address":"0xbc33acfffe479c47"},"type":"device_joined"}'
info  2020-12-22 17:46:23: Starting interview of '0xbc33acfffe479c47'
info  2020-12-22 17:46:23: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xbc33acfffe479c47","ieee_address":"0xbc33acfffe479c47","status":"started"},"type":"device_interview"}'
info  2020-12-22 17:46:23: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0xbc33acfffe479c47"},"type":"device_connected"}'
info  2020-12-22 17:46:23: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0xbc33acfffe479c47"},"type":"pairing"}'
info  2020-12-22 17:46:23: MQTT publish: topic 'zigbee2mqtt/bridge/config/devices', payload '[{"dateCode":"20200925","friendly_name":"Coordinator","ieeeAddr":"0x00124b002160023c","lastSeen":1608655583056,"networkAddress":0,"softwareBuildID":"zStack3x0","type":"Coordinator"},{"dateCode":"20170627","description":"Aqara human body movement and illuminance sensor","friendly_name":"PIR Dressing","hardwareVersion":1,"ieeeAddr":"0x00158d00047eafd1","lastSeen":1608653496442,"manufacturerID":4151,"manufacturerName":"LUMI","model":"RTCGQ11LM","modelID":"lumi.sensor_motion.aq2","networkAddress":52269,"powerSource":"Battery","softwareBuildID":"3000-0001","type":"EndDevice","vendor":"Xiaomi"},{"dateCode":"","description":"10A UK or 16A EU smart plug","friendly_name":"Wallplug 11","hardwareVersion":1,"ieeeAddr":"0x588e81fffeff5bc0","lastSeen":1608655543644,"manufacturerID":4098,"manufacturerName":"_TZ3000_g5xawfcq","model":"TS0121_plug","modelID":"TS0121","networkAddress":18517,"powerSource":"Mains (single phase)","type":"Router","vendor":"TuYa"},{"dateCode":"20161128","description":"Aqara door & window contact sensor","friendly_name":"Window Dressing","hardwareVersion":2,"ieeeAddr":"0x00158d00054da010","lastSeen":1608652671991,"manufacturerID":4151,"manufacturerName":"LUMI","model":"MCCGQ11LM","modelID":"lumi.sensor_magnet.aq2","networkAddress":38549,"powerSource":"Battery","softwareBuildID":"3000-0001","type":"EndDevice","vendor":"Xiaomi"},{"dateCode":"","description":"10A UK or 16A EU smart plug","friendly_name":"Wallplug 10","hardwareVersion":1,"ieeeAddr":"0x588e81fffeff5cb0","lastSeen":1608655413468,"manufacturerID":4098,"manufacturerName":"_TZ3000_g5xawfcq","model":"TS0121_plug","modelID":"TS0121","networkAddress":44179,"powerSource":"Mains (single phase)","type":"Router","vendor":"TuYa"},{"dateCode":"","description":"10A UK or 16A EU smart plug","friendly_name":"Wallplug 12","hardwareVersion":1,"ieeeAddr":"0x842e14fffe138a46","lastSeen":1608655468355,"manufacturerID":4098,"manufacturerName":"_TZ3000_g5xawfcq","model":"TS0121_plug","modelID":"TS0121","networkAddress":41251,"powerSource":"Mains (single phase)","type":"Router","vendor":"TuYa"},{"dateCode":"20191205","description":"Aqara temperature, humidity and pressure sensor","friendly_name":"Badkamer","hardwareVersion":30,"ieeeAddr":"0x00158d00057847f0","lastSeen":1608654345911,"manufacturerID":4151,"manufacturerName":"LUMI","model":"WSDCGQ11LM","modelID":"lumi.weather","networkAddress":38746,"powerSource":"Battery","softwareBuildID":"3000-0001","type":"EndDevice","vendor":"Xiaomi"},{"dateCode":"20191205","description":"Aqara temperature, humidity and pressure sensor","friendly_name":"Woonkamer","hardwareVersion":30,"ieeeAddr":"0x00158d00057820da","lastSeen":1608654700964,"manufacturerID":4151,"manufacturerName":"LUMI","model":"WSDCGQ11LM","modelID":"lumi.weather","networkAddress":7051,"powerSource":"Battery","softwareBuildID":"3000-0001","type":"EndDevice","vendor":"Xiaomi"},{"description":"Aqara human body movement and illuminance sensor","friendly_name":"PIR WC","hardwareVersion":1,"ieeeAddr":"0x00158d00058a7cd4","lastSeen":1608654321829,"manufacturerID":4151,"manufacturerName":"LUMI","model":"RTCGQ11LM","modelID":"lumi.sensor_motion.aq2","networkAddress":62142,"powerSource":"Battery","type":"EndDevice","vendor":"Xiaomi"},{"dateCode":"20161128","description":"Aqara door & window contact sensor","friendly_name":"Window Badkamer","hardwareVersion":2,"ieeeAddr":"0x00158d00044ee64e","lastSeen":1608653200713,"manufacturerID":4151,"manufacturerName":"LUMI","model":"MCCGQ11LM","modelID":"lumi.sensor_magnet.aq2","networkAddress":5540,"powerSource":"Battery","softwareBuildID":"3000-0001","type":"EndDevice","vendor":"Xiaomi"},{"dateCode":"20161128","descr [... cut ...] 2","networkAddress":44653,"powerSource":"Battery","softwareBuildID":"3000-0001","type":"EndDevice","vendor":"Xiaomi"},{"dateCode":"","description":"-","friendly_name":"0xbc33acfffe479c47","hardwareVersion":1,"ieeeAddr":"0xbc33acfffe479c47","lastSeen":1608655607582,"manufacturerID":4098,"manufacturerName":"_TZ3000_wamqdr3f","networkAddress":35788,"powerSource":"Mains (single phase)","type":"Router","vendor":"-"}],"type":"devices"}'
info  2020-12-22 17:47:04: Zigbee: disabling joining new devices.

Koenkk, 2020-12-22:
@drdada can you remove `0xbc33acfffe47ee1f` from your network (https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigremove). Repair it and provide the herdsman debug log of this?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

drdada, 2020-12-22:
@Koenkk removing and repairing the device did the tricks.
The model is now displayed
`Zigbee2MQTT:debug 2020-12-22 19:48:58: Received Zigbee message from '0xbc33acfffe47ee1f', type 'readResponse', cluster 'genBasic', data '{"modelId":"TS011F"}' from endpoint 1 with groupID 0` 

Thanks a lot

closed 2020-12-22 by drdada