Koenkk/zigbee2mqtt #5372: aqara magic cube failed at interview

Bios-dude, 2020-12-22:
# Bug Report
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00158d0002a67128","ieee_address":"0x00158d0002a67128"},"type":"device_joined"}'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00158d0002a67128","ieee_address":"0x00158d0002a67128","status":"started"},"type":"device_interview"}'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x00158d0002a67128"},"type":"device_connected"}'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x00158d0002a67128"},"type":"pairing"}'
ERROR: Failed to interview '0x00158d0002a67128', device has not successfully been paired
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00158d0002a67128","ieee_address":"0x00158d0002a67128","status":"failed"},"type":"device_interview"}'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/devices', payload '[{"definition":null,"endpoints":{"1":{"bindings":[],"clusters":{"input":[],"output":[]}},"11":{"bindings":[],"clusters":{"input":["ssIasAce"],"output":["ssIasZone","ssIasWd"]}},"110":{"bindings":[],"clusters":{"input":[],"output":[]}},"12":{"bindings":[],"clusters":{"input":[],"output":[]}},"13":{"bindings":[],"clusters":{"input":["genOta"],"output":[]}},"2":{"bindings":[],"clusters":{"input":[],"output":[]}},"242":{"bindings":[],"clusters":{"input":[],"output":[]}},"3":{"bindings":[],"clusters":{"input":[],"output":[]}},"4":{"bindings":[],"clusters":{"input":[],"output":[]}},"47":{"bindings":[],"clusters":{"input":[],"output":[]}},"5":{"bindings":[],"clusters":{"input":[],"output":[]}},"6":{"bindings":[],"clusters":{"input":[],"output":[]}},"8":{"bindings":[],"clusters":{"input":[],"output":[]}}},"friendly_name":"Coordinator","ieee_address":"0x00124b0014d98ca2","interview_completed":true,"interviewing":false,"network_address":0,"supported":false,"type":"Coordinator"},{"definition":null,"endpoints":{},"friendly_name":"0x00158d0002a67128","ieee_address":"0x00158d0002a67128","interview_completed":false,"interviewing":false,"network_address":19291,"supported":false,"type":"Unknown"}]'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_failed","meta":{"friendly_name":"0x00158d0002a67128"},"type":"pairing"}'
## What happened
aqara magic cube show as unsupported when it should be supported
## What did you expect to happen
it should be supported like usual
## How to reproduce it (minimal and precise)
i bought a new aqara magic cube, press link for 5 second. it start connected and interview but failed
## Debug Info
Zigbee2MQTT version:
Adapter hardware: CC2531
Adapter firmware version: newest

Koenkk, 2020-12-22:
Please check: https://www.zigbee2mqtt.io/information/FAQ.html#interview-fails

gsolem, 2020-12-23:
Had the same problem today, and repairing several times finally it went through. I am suspecting is was because of low battery (29%). Also, make sure to hold the reset button until the blinking stops for pairing to start successfully.

Lebo-Wang, 2020-12-26:
Specifically for Xiaomi cube, I guess it is important to press the button every 2-3 second to keep it alive after the interview started, according to my experience.

closed 2021-02-03 by github-actions[bot]