Koenkk/zigbee2mqtt #5373: xiaomi gas detector JTQJ-BF-01LM/BW always report gas=true

fireport, 2020-12-22:
Xiaomi JTQJ-BF-01LM/BW gas detector report gas leakage even if not true

## What did you expect to happen
Gas leakage must be reported only if there is a true gas leak
## How to reproduce it (minimal and precise)
The problem can be reproduced always, even if the device is powered off and on
## Debug info
Zigbee2MQTT version 1.16.2 (commit #0514204)
Adapter hardware: CC2531
Adapter firmware version: Coordinator firmware version: '{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190608,"transportrev":2},"type":"zStack12"}'

Koenkk, 2020-12-22:
- Stupid question: are you sure the device is not broken?
- Can you provide the debug log when gas: true is published?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

fireport, 2020-12-30:
I think that the device works correctly because identify gas leakage and emits an alarm accordingly.
This is the relevant part of the log. The attribute gas it's true even on boot.
The device name is "sensore perdita gas"
Thank you for your effort

`info  2020-12-30 17:25:22: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"sensore perdita gas","ieee_address":"0x00158d00042d4b14"},"type":"device_announce"}'
info  2020-12-30 17:25:22: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"sensore perdita gas"},"type":"device_announced"}'
debug 2020-12-30 17:25:39: Received Zigbee message from 'clima esterno', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":2955,"10":33558,"100":1184,"101":8165,"102":100056,"4":17320,"5":5,"6":[2,17039363]}}' from endpoint 1 with groupID 0
info  2020-12-30 17:25:39: MQTT publish: topic 'zigbee2mqtt/clima esterno', payload '{"battery":74,"humidity":81.65,"linkquality":18,"pressure":1000.6,"temperature":11.84,"voltage":2955}'
debug 2020-12-30 17:26:13: Received Zigbee message from 'sensore presenza', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":8}' from endpoint 1 with groupID 0
info  2020-12-30 17:26:13: MQTT publish: topic 'zigbee2mqtt/sensore presenza', payload '{"battery":97,"illuminance":8,"illuminance_lux":8,"linkquality":18,"occupancy":true,"voltage":2995}'
debug 2020-12-30 17:26:13: Received Zigbee message from 'sensore presenza', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":1}' from endpoint 1 with groupID 0
info  2020-12-30 17:26:13: MQTT publish: topic 'zigbee2mqtt/sensore presenza', payload '{"battery":97,"illuminance":8,"illuminance_lux":8,"linkquality":15,"occupancy":true,"voltage":2995}'
debug 2020-12-30 17:26:13: Received Zigbee message from 'router zigbee', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 8 with groupID 0
info  2020-12-30 17:26:13: MQTT publish: topic 'zigbee2mqtt/router zigbee', payload '{"led":false,"linkquality":34}'
debug 2020-12-30 17:27:14: Received Zigbee message from 'router zigbee', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 8 with groupID 0
info  2020-12-30 17:27:14: MQTT publish: topic 'zigbee2mqtt/router zigbee', payload '{"led":false,"linkquality":18}'
info  2020-12-30 17:27:43: MQTT publish: topic 'zigbee2mqtt/sensore presenza', payload '{"battery":97,"illuminance":8,"illuminance_lux":8,"linkquality":15,"occupancy":false,"voltage":2995}'
debug 2020-12-30 17:28:14: Received Zigbee message from 'router zigbee', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 8 with groupID 0
info  2020-12-30 17:28:14: MQTT publish: topic 'zigbee2mqtt/router zigbee', payload '{"led":false,"linkquality":18}'
debug 2020-12-30 17:29:14: Received Zigbee message from 'router zigbee', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 8 with groupID 0
info  2020-12-30 17:29:14: MQTT publish: topic 'zigbee2mqtt/router zigbee', payload '{"led":false,"linkquality":15}'
debug 2020-12-30 17:29:22: Received Zigbee message from 'sensore perdita gas', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
debug 2020-12-30 17:30:08: Saving state to file /opt/zigbee2mqtt/data/state.json
debug 2020-12-30 17:30:14: Received Zigbee message from 'router zigbee', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 8 with groupID 0
info  2020-12-30 17:30:14: MQTT publish: topic 'zigbee2mqtt/router zigbee', payload '{"led":false,"linkquality":18}'
debug 2020-12-30 17:30:22: Received Zigbee message from 'sensore perdita gas', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
debug 2020-12-30 17:30:29: Received Zigbee message from 'sensore perdita gas', type 'attributeReport', cluster 'genBasic', data '{"65281":{"100":0,"150":2147483648,"3":36,"5":22,"8":8456,"9":256}}' from endpoint 1 with groupID 0
info  2020-12-30 17:30:29: MQTT publish: topic 'zigbee2mqtt/sensore perdita gas', payload '{"battery_low":false,"gas":true,"gas_density":0,"linkquality":18,"sensitivity":"medium","tamper":false}'
`

Koenkk, 2020-12-31:
@fireport can you stop z2m, remove `gas` from the `data/state.json` file and provide the debug log when gas turns true again?

fireport, 2021-01-01:
Here the log after removing gas entry in data/state.json and triggering  the sensor using a gas ligther
```
debug 2021-01-01 10:37:25: Received Zigbee message from 'sensore perdita gas', type 'attributeReport', cluster 'genBasic', data '{"65281":{"100":237,"150":1107296256,"3":36,"5":22,"8":8456,"9":256}}' from endpoint 1 with groupID 0
info  2021-01-01 10:37:25: MQTT publish: topic 'zigbee2mqtt/sensore perdita gas', payload '{"battery_low":false,"gas_density":237,"linkquality":21,"sensitivity":"medium","tamper":false}'
debug 2021-01-01 10:37:26: Received Zigbee message from 'sensore perdita gas', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":1}' from endpoint 1 with groupID 0
info  2021-01-01 10:37:26: MQTT publish: topic 'zigbee2mqtt/sensore perdita gas', payload '{"battery_low":false,"gas":true,"gas_density":237,"linkquality":15,"sensitivity":"medium","tamper":false}'
debug 2021-01-01 10:37:44: Received Zigbee message from 'sensore perdita gas', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
```
thank yoy and happy new year

Koenkk, 2021-01-01:
`debug 2021-01-01 10:37:26: Received Zigbee message from 'sensore perdita gas', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":1}' from endpoint 1 with groupID 0` was the message I was searching for. You can see the `zonestatus` is 1 (which means gas true). Can you also catch the same message when the sensor is expected to go from gas true -> gas false?

fireport, 2021-01-01:
Dear Koenkk, I close this issue. In the meanwhile I updated to   1.17.0 commit: 07cdc9d and then sensor starts to work as expected. 
Thank you

closed 2021-01-01 by fireport