Koenkk/zigbee2mqtt #5374: Saswell SEA801-Zigbee is recognized as SEA802-Zigbee

salopette, 2020-12-22:
Hello, my thermostat is Saswell SEA801-Zigbee is recognized as SEA802-Zigbee.

[image]
[image]

Koenkk, 2020-12-22:
Fixed.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-12-22 by Koenkk

drzony, 2020-12-24:
@Koenkk 801 and 802 have the same model and manufaturer now. So now my 802 are recognized as 801. I'll make a pull request to do a single 80x. Is it possible to add 2 pictures to zigbee2mqtt.io?