Koenkk/zigbee2mqtt #5377: XBee 3 responding to on/off command from Hass.io

byrleyar, 2020-12-22:
# Bug Report
## What happened
z2m is running in hass.io.  I have an xbee3 set up as an end device, I have it added to zigbee2mqtt as a 'supported' device via an [external converter](https://pastebin.com/DTAfucfu) -- a simple on/off switch.  When I click the toggle in home assistant, I can see (via Digi XCTU) that my xbee receives an "[Explicit RX Indicator](https://pastebin.com/XBkegq0D)," but after around 30 seconds or so, zigbee2mqtt [throws an error](https://pastebin.com/Hvp5m8k2) because it doesn't get a response back from the xbee.  My question:  what does z2m expect to receive from the xbee?

## What did you expect to happen
I expected zigbee2mqtt to register the end device being in the ON state.

## How to reproduce it (minimal and precise)
This most definitely isn't a bug -- it's my ignorance.  Feel free to delete if this isn't appropriate, but I was just following the link on the How To Support New Devices Page.  Thank you!

## Debug Info
Zigbee2MQTT version: 1.16.2.2
Adapter hardware: deconz conbeeII
Adapter firmware version:

Koenkk, 2020-12-23:
Try disabling the default response in the external converters like this: https://github.com/Koenkk/zigbee-herdsman-converters/blob/ef578140a56b7ff1489916e865f826975f655e6f/devices.js#L4343

byrleyar, 2020-12-28:
> Try disabling the default response in the external converters like this: https://github.com/Koenkk/zigbee-herdsman-converters/blob/ef578140a56b7ff1489916e865f826975f655e6f/devices.js#L4343

This was it.  Thank you!

closed 2020-12-28 by byrleyar

Lavaerius, 2021-12-06:
@byrleyar 
Hi there, I've been trying to follow your example all morning.  I can't get the xbee unit responding to the exposed switch command.  

I'd comment on your project, but all I see is this thread.  Do you have your work anywhere.?

Thanks!

byrleyar, 2021-12-06:
@Lavaerius you can find more project details here: https://hackaday.io/project/178435-mr-miffy-rgb-led-module

That said, a recent z2m update broke my admittedly-jury-rigged setup, so I abandoned the xbee and implemented an esp8266 w/ WLED, which works quite will with Home Assistant

corteXnzt, 2022-06-22:
@byrleyar Hello, I have read your research sources and managed to create a zigbee switch based on your project. The power, voltage, current etc are also exposed but currently with null value.  The problem I have now is how to send data back to coordinator like power, voltage, current, etc. 

I'm playing with the xbee.transmit() in micropython but with no luck. I guess I need to fill up some other details or there is a needed format. I know based on you're comment here you abandoned the project but just in case you continued or you have idea on how can I send back those payload? TIA

Lavaerius, 2022-06-22:
Hey, I had to start in a few places, but I've built an xbee 3 rf device that's acting as a garage barrier with micropython
https://gitea.wagshome.duckdns.org/publicWagsHome/garageDoorZigbee

this has been running in my garage now for a couple months and it's been great.  there might be some old code in there for pushing some other zcl values you're looking for.

Lavaerius, 2022-06-23:
Z2m is looking for data that tracks the ZigBee cluster library (zcl), and a bit from the ZigBee device object (zdo) a recent-ish release of the xbee firmware allowed for zdo to be exposed to micropython, which enabled responding to z2m with specific device describers. (Clusters, endpoints, etc.). Then, when it's identified what the device is, it starts operating against the zcl to communicate with the device.
If you look through z2m devices, it's not terrifyingly hard to mimic an already defined device at this point.  But if you want to do something not defined in the herdsman, or not defined by a converter, then z2m doesn't know how to handle it.

Part of getting the project above working required a PR into herdsman for defining the barrier type, and a converter I've not merged in (but is in the project) to defined the conversions between incoming/outgoing data, and what z2m knows how to ship out to HA.

It was a whole thing.

corteXnzt, 2022-06-23:
@byrleyar  Yeah! Thank you for your example in hackday, it pointed me out here.

@Lavaerius Thank you for sharing examples code and brief explanation on how it works. It seems your project is more advance that what I'm trying to achieve (which is better). I will study your code now.

here is my converter , I hope I'm in the right track

> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;

>const homeAssistantPlug = {
    type: 'switch',
    object_id: 'switch',
    discovery_payload: {
        state: "OFF",    
        power: 0,
        current: '0',        
        voltage: 0,
        energy: '0.00',
        linkquality: 160,
        schema: 'json',
        command_topic: true,
    },
};

>const device = {
    zigbeeModel: ['XBEE3'],
    model: 'XBee',
    vendor: 'Digi',
    description: 'Smart Socket(100A)',
    fromZigbee: [fz.on_off],
    toZigbee: [tz.on_off, tz.ignore_transition, tz.ignore_rate],
    meta: {disableDefaultResponse: true},
    exposes: [e.switch(), e.power(), e.current(), e.voltage(), e.energy(), e.linkquality()],
};

>module.exports = device;

corteXnzt, 2022-06-25:
> Z2m is looking for data that tracks the ZigBee cluster library (zcl), and a bit from the ZigBee device object (zdo) a recent-ish release of the xbee firmware allowed for zdo to be exposed to [...]

@Lavaerius Thank you! I got it working. 

You are right --> "If you look through z2m devices, it's not terrifyingly hard to mimic an already defined device ". 

Though my application is not supported on the spec library I managed to make it work by reading the zcl library pdf link on the spec library and a lots of searches and trial and errors. 

Cheers!