Koenkk/zigbee2mqtt #5378: Cover group doesn't seem to work?

Zebble, 2020-12-23:
## What happened
Created a cover group with IKEA Fyrtur blinds (4 blinds) using the web UI.  The group appears in the RAW config and appears to be responding normally through MQTT and the Logs.  However, when sending a set = {"position": XX}, nothing happens.  The same logic and configuration using lights works as expected.

After sending the set command, an update is published for all grouped blinds, as expected but the position values don't change on individual blinds.

## What did you expect to happen
Position of blinds should change after a set position is sent to the group.

## How to reproduce it (minimal and precise)
- Create cover group using Web UI.
- Send a {"position": XX} to the zigbee2mqtt/<group name>/set topic.
- Topics are updated for individual blinds, but position # doesn't change.
- All blinds in the group should set their position to XX.

## Debug info
Zigbee2MQTT version:  1.16.2-dev commit: be6b693
Adapter hardware: zigazigah CC2652R stick
Adapter firmware version:  20201026

Koenkk, 2020-12-23:
This is a known issue which cannot be fixed. Technical details can be found here: https://github.com/Koenkk/zigbee2mqtt/issues/892#issuecomment-635275634

closed 2020-12-23 by Koenkk

maciekb, 2021-02-26:
I don't quite understand, does this only apply to Ikea blinds or any non-router device? I'm asking because I'm having a problem with a group consisting of https://www.zigbee2mqtt.io/devices/TS0601_cover.html devices working. Individually they work, but I can't get any response to commands sent to the group.

King-Cole, 2022-02-10:
Peripheral subject, but I was disheartened reading this thread and almost gave up. **For anyone having trouble with binding their remote to the blind group directly**, I have some advice. It took me a while, but finally got it. I have 3 blinds (thus 3 remotes), and I did a bad thing on the first remote that screwed me up for a while: upgrading via OTA. The second two remotes I did not do this, and they are still able to command groups just fine. Up, down, stopping partway, the works.

So, in short:

1. For firmwares: blinds to newest (2.3.088), button to old (shipped with 2.2.010)
2. Create group via z2m UI and add all your blinds. FYI, I have never gotten scenes or group commands to work here. I've tried setting the group to "optimistic" and every kinda MQTT payload imaginable, just doesn't work.
3. Via the z2m UI, go to the "Bind" page under the button device.
4. Unbind the default group (default_bind_group), press either up or down on the button to ensure it receives the setting.
5. Bind your newly created blinds group with only the "Closures" cluster. Again, press either up or down on the button.
6. With any luck (trust me you need some in this game), you can now control all your blinds!

jonhammer, 2022-05-16:
@King-Cole thanks for the detailed guide, this was super helpful and worked flawlessly for me!