Koenkk/zigbee2mqtt #5380: Supported sonoff becomes unsupported

gald33, 2020-12-23:
## What happened
I had a Sonoff basiczbr3 connected to the coordinator, it was recognized as a Sonoff switch.
Today I've added a cc2530 router and moved the Sonoff to a farther room.
I didn't know how to make the Sonoff communicate with the new router instead of the coordinator so I reset it.
Now, it is found by zigbee2mqtt and the interview passes but the Sonoff is not recognized.
I repeated the process multiple times and put the Sonoff near the router, but it didn't help.

## What did you expect to happen
To have the Sonoff recognized.

## How to reproduce it (minimal and precise)
Resetting the Sonoff with 5 sec press on it's button.

## Debug info
```INFO: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"friendly_name":"0x00124b001b79882e","supported":false},"type":"pairing"}'```

Zigbee2MQTT version:
Adapter hardware: Sonoff basiczbr3
Adapter firmware version:

Koenkk, 2020-12-23:
Can you provide the data/databse.db entry of that device? (`0x00124b001b79882e`)

gald33, 2020-12-23:
Gladly, where can I find it? (I'm using the HA add on, but I can work with the terminal if more convenient)

gald33, 2020-12-23:
From the logs I can see this info if that what you mean:
```{"ieeeAddr":"0x00124b001b79882e","networkAddress":62577},"sourceIeeeAddr":"0x00124b001b79882e","sourceNwkAddr":62577,"target":```
and 
```["lqi"],"friendlyName":"0x00124b001b79882e","ieeeAddr":"0x00124b001b79882e","lastSeen":1608719263767,"networkAddress":62577,"type":"Router"}]}},"status":"ok"}'```
what's more, I've restarted the device and the map shows it's not linked to anything as a child or parent.

gald33, 2020-12-23:
I managed to get it working again by connecting the device to the coordinator (and powering off the router).
I still want to get it the data you asked about because the problem is just circumvented now but not really solved.
If I discover something new I'll update the post.

Koenkk, 2020-12-23:
What is logged above: `INFO: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"friendly_name":"0x00124b001b79882e","supported":false},"type":"pairing"}'` (also enable the debug logging)

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

gald33, 2020-12-23:
I copied what's above... and even some below.
https://pastebin.com/RyhAQ1TC

hope that's that, tell me if you need anything else

Koenkk, 2020-12-24:
> I copied what's above... and even some below.

In that case I would expect the log to contains `INFO: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"friendly_name":"0x00124b001b79882e","supported":false},"type":"pairing"}'` but it doesn't (??)

gald33, 2020-12-24:
I'll copy the full log next time. The think is I now connected the device not through the router it is shown as supported this way.
I'll be able to reset it again and copy the entire log only in two weeks, so  I'll update after I have it.

by the way, I copied from the log tab in the add-on (not it's web ui).

frogale, 2021-01-09:
Hi.. 

I have same problem now.. I have 4 sonoff as routers for all my home.. but today after update of my docker image to 1.17.0 I have  all my network very unstable.. It looks like Sonoffs does not work.. so I have removed one of them and after new repair only "unsupported" is shown:

`{"id":23,"type":"Unknown","ieeeAddr":"0x00124b002225522f","nwkAddr":30567,"epList":[],"endpoints":{},"interviewCompleted":false,"meta":{},"lastSeen":1610214969516}`

Sonoff constantly trying to join (green led is blinking) but without success.. I have downgraded back to 1.16.2 with same result. I have restarted my raspberry, no change. Not sure what to do next..

`info  2021-01-09 18:54:35: Device '0x00124b002225522f' joined
info  2021-01-09 18:54:35: Starting interview of '0x00124b002225522f'
info  2021-01-09 18:54:35: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b002225522f","ieee_address":"0x00124b002225522f"},"type":"device_joined"}'
info  2021-01-09 18:54:35: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b002225522f","ieee_address":"0x00124b002225522f","status":"started"},"type":"device_interview"}'
info  2021-01-09 18:54:35: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x00124b002225522f"},"type":"device_connected"}'
info  2021-01-09 18:54:35: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x00124b002225522f"},"type":"pairing"}'`

frogale, 2021-01-09:
1 hour of debugging and trying.. and now it is working again (already paired devices and even new sonoff).. Don't know why..

frogale, 2021-01-24:
Same problem again.. cannot pair sonoff zigbee zbr3.. It started with very strange behavior.. Z2M was showing device's state changes but it cannot set with error 225. After reset of ZBR3 it cannot connect and there is no log message in the console. In this case there is no "unsupported" messages.

frogale, 2021-01-24:
Very strange. It looks like some interference on USB cable to slae'sh dongle. If I touch the cable it is working. Hope it will helps someone.

fluffyanvil, 2021-03-01:
same problem with basiczbr3 after 1 week using
1) big latency in 1 meter of coordinator
2) blinking led and "interview_failed"

frogale, 2021-03-03:
I have had problems with Ikea Tradfri bulbs.. Do you have any bulbs in your Zigbee networks? Remove them. And try different USB cable if you are using it for dongle connection.

closed 2021-04-10 by github-actions[bot]