Koenkk/zigbee2mqtt #5381: zigbee_publish_error on Osram smart+ plug

flubitz, 2020-12-23:
Hi All,

I have problems running 2 Osram smart+ plugs stable in my zigbee2mqtt enviroment. After some time (hours to days) I got this error:

Zigbee2MQTT:info  2020-12-23 13:57:33: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'zb_Osram1' failed: 'Error: Command 0x7cb03eaa0a09f244/3 genOnOff.toggle({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Data request failed with error: 'MAC channel access failure' (225))'","meta":{"friendly_name":"zb_Osram1"},"type":"zigbee_publish_error"}'

I don't get rid of this problem by: 
- rejoining all my 7 devices 
- "rebuild" zigbe2mqtt 
-  reseting CC26X2R1 or reloading firmware and erasing complete memory.

my Enviroment:
- Raspi Pi 4 / orginal raspi power supply
- running zigbee2mqtt on Docker
- Fhem

## What happened
Osram smart+ plug does not respond to on/off/toggle commands.
## What did you expect to happen
plug should change state
## How to reproduce it (minimal and precise)
It seems to work for a while after repairing/resetting the plug (hours to days) -> Then the device is not responding anymore.

## Debug info
Zigbee2MQTT version: 1.16.2-dev
Adapter hardware:  TI CC26X2R1
Adapter firmware version: CC26X2R1_20201026.hex

Any ideas?

Kind regards, 

Florian

Koenkk, 2020-12-23:
MAC channel access failure' (225)) typically indicated interference problems

- Since you have a pi 4, make sure the CC26X2R1 is placed away from the pi (as far as possible)
- Check if you wifi channels interfere with the Zigbee ones: https://www.zigbee2mqtt.io/how_tos/how_to_improve_network_range_and_stability.html#reduce-wifi-interference-by-changing-the-zigbee-channel
- Try switching to this firmware which increases transmit power: https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin

flubitz, 2020-12-25:
Hi Koenekk,

thanks for your help. I tryed out a longer extension cable and placed the CC26X2R1 farther away from the pi. This works fine for 2 days now. 

Florian

closed 2021-02-02 by github-actions[bot]