Koenkk/zigbee2mqtt #5382: Lonsonho 1-gang wall switch unable to see changing statuses. TS0001 by _TZ3000_gidy6sjs.

Mistral, 2020-12-23:
## What happened
I've bought some wall switches from Lonsonho, 1-gang, 2-gang and 3-gang (all with neutral line). Each devices were successfully paired, but I don't receive any changing statuses from 1-gang version which is discovered as: `TS0001 by _TZ3000_gidy6sjs.` rest devices are 2gang-> `TS0002 by _TZ3000_svoqrno4` (works correctly), 3gang -> `TS0003 by _TYZB01_b8cr31hp` (also works correctly). From 1-gang version I don't see any updates.

## What did you expect to happen
See notification after pressing button.

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: 1.16.2
Adapter hardware:  Conbee II
Adapter firmware version: 0x26660700

Koenkk, 2020-12-23:
Can you try changing the definition in devices.js to below and see if it works?

```js
{
    zigbeeModel: ['TS0001'],
    model: 'TS0001',
    vendor: 'TuYa',
    description: '1 gang switch',
    extend: preset.switch,
    whiteLabel: [{vendor: 'CR Smart Home', model: 'TS0001', description: 'Valve control'}, {vendor: 'Lonsonho', model: 'X701'},
        {vendor: 'Bandi', model: 'BDS03G1'}],
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint) => {
        await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
    },
},
```

Mistral, 2020-12-23:
I had to make some changes (probably my dependency of converter library is different as it is on the master branch on Github).

Here is the code which worked for me:
```
{
    zigbeeModel: ['TS0001'],
    model: 'TS0001',
    vendor: 'TuYa',
    description: '1 gang switch',
    extend: generic.switch,
    whiteLabel: [{vendor: 'CR Smart Home', model: 'TS0001', description: 'Valve control'}, {vendor: 'Lonsonho', model: 'X701'},
        {vendor: 'Bandi', model: 'BDS03G1'}],
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint) => {
        await bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
    },
},
```
I believe your code after merging to master branch of zigbee-herdsman-converters should also work properly.

Should I open a pull request with your fix, or will you do it?

Koenkk, 2020-12-24:
@Mistral thanks for confirming, added it.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-12-24 by Koenkk