Koenkk/zigbee2mqtt #5384: Aqara GZCGQ01LM - seen but won'tpair

z4rn0x, 2020-12-23:
## What happened
Can't pair the Aqara GZCGQ01LM light intensity sensor
Even after mutliple tentatives, and with two different sensors, I can't get to pair my new Aqara light intensity sensor. It's seen by Z2M but won't pair.
I attached the log file

## What did you expect to happen
Pairing of the sensors

## How to reproduce it (minimal and precise)
Multiple tentatives, same result:
- long press on the button until the blue light blinks, then wait for the pairing
- long press on the button until the blue light blinks, then click press the button every second until it's pairing

## Debug info
Zigbee2MQTT version: 1.16.2
Adapter hardware: CC2531
Adapter firmware version: zStack12
Coordinator version: 20190608

https://pastebin.com/7UH2Vh0U

Koenkk, 2020-12-23:
Make sure to check: https://www.zigbee2mqtt.io/information/FAQ.html#interview-fails (also reflashing the CC2531 helps sometimes)

z4rn0x, 2020-12-23:
> Make sure to check: https://www.zigbee2mqtt.io/information/FAQ.html#interview-fails (also reflashing the CC2531 helps sometimes)

I tried everything on the link you provided, without success, so I reflashed my CC2531, I have no idea why, but it did the trick!
Thank you!

closed 2020-12-23 by z4rn0x