Koenkk/zigbee2mqtt #5385: Support Ledvance A60 multicolour bulb - attempts and info in here

DaOurZ, 2020-12-23:
Hello fellow Zigbee lovers,

I'm trying to add a (kinda) unsupported device: 

## Device: Ledvance Led bulb
_Information about the device + link_
[Ledvance Smart+ Classic Multicolour 60 10W e27](https://shop.ledvance.com/en/products/ledvance-smart-classic-multicolour-60-10-w-e27-energieklasse-a?variant=32922873987133)
Zigbee smart multicolour led bulb, from Ledvance (providing also Osram bulbs). If relevant, seems to be Zigbee 3.0 - not written, but it behaves like it (pairing process shows green - blue - red, and not blink - cf 3.3.8 on [Ledvance FAQ](https://www.ledvance.com/consumer/smart/faq/faq-for-smart-zigbee-products/index.jsp))

## Database.db
_data/database.db entry of the device_

`{"id":5,"type":"Router","ieeeAddr":"0xf0d1b8000018a364","nwkAddr":41640,"manufId":4489,"manufName":"LEDVANCE","powerSource":"Unknown","modelId":"A60 RGBW Value II","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,2821,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"A60 RGBW Value II","manufacturerName":"LEDVANCE","powerSource":0,"zclVersion":3,"appVersion":0,"stackVersion":0,"hwVersion":2,"swBuildId":"00203045"}}},"binds":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"meta":{}}},"appVersion":0,"stackVersion":0,"hwVersion":2,"swBuildId":"00203045","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1608736565930}
`

## My attempts
"Kinda unsupported", as it is very similar to its fellow OSRAM bulbs. I used the tutorial to add the device to my Home Assistant instance (running on Home Assistant OS). I then imported "devices.js" file from zigbee-herdsman-converters github, put it under "/share/zigbee2mqtt", and added the following lines to it:
`
    {
        zigbeeModel: ['A60 RGBW Value II'],
        model: 'AC25697',
        vendor: 'LEDVANCE',
        description: 'SMART+ CLASSIC MULTICOLOUR 60 10W E27',
        extend: preset.ledvance.light_onoff_brightness_colortemp_colorhs,
        ota: ota.ledvance,
    },
`

... but as I am running Home Assistant OS, it was not an easy part: the zigbee2mqtt plugin did not want to start as it was looking to files not here (_require(".converters/..."_ for example, npm was not happy). I tested by putting under "/share/zigbee2mqtt" the whole zigbee-herdsman-converters github but with no more success.

## Bottom line
Is there any chance this device would be added to this beloved system? If by any chance and with my limited material it would be interesting I perform more test don't hesitate, I can do! :)

Any help is appreciated :)

Koenkk, 2020-12-23:
There is a high chance this config works so i've added it.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

DaOurZ, 2020-12-25:
Hi Koen!
I waited a bit to see if the miracle of Christmas happened and.... it is here! A thousand thanks - I can now play with my lights, and with the Light Entity Card, how nice it is! :)
Cheers!

closed 2020-12-25 by DaOurZ

DaOurZ, 2020-12-25:
PS: by the way if you need me to enhance the wiki (zigbee2mqtt.info) with this new device, I can do it or provide the info needed, if it helps.