Koenkk/zigbee2mqtt #5386: zzh! paring issues with zemismart switch - 3 gang v2.0 (HGZB-43) and 2 gang v2.0 (HGZB-42)

juslex, 2020-12-23:
What happened
I am currently using zzh as a coordinator. My first coordinator was conbee 2 for the reason of this issue here (https://github.com/Koenkk/zigbee2mqtt/issues/4461) I decided to acquire zzh.
On zzh, I have the update on MQTT when I press the zemismart switch - 3 gang v2.0 (HGZB-43) and  - 2 gang v2.0 (HGZB-42).
However, I am experiencing another type of problem and I think it is also related to zemismart switch.
After configuring and pairing the devices, in that first moment, I can add any end device that I want without any problem, doing the pairing procedure in the place where the end device will be. However, after some time, I cannot add any device to the network. I can only do it if I approach the coordinator, so the pairing is successful. 
However, after some time, I cannot add any device to the network. I can only do it if I approach the coordinator, so the pairing is successful. However, when taking the end device to its destination, it cannot connect to the nearest router (zemismart switch - 3 gang v2.0 (HGZB-43) and Smart light switch - 2 gang v2.0 (HGZB-42 )), even admitting the entry of new devices and pressing the physical button of the end device to perform the interview again. 
The only way to overcome this problem is by deleting the router closest to where the end device will be, re-adding the router and then being able to pair the end device at its destination location. In this case, the end device connects to re-added router

What did you expect to happen
End device join in network
How to reproduce it (minimal and precise)
Just add any end device using zzh! as coordinator using zemismart switch button - 3 gang v2.0 (HGZB-43) and Smart light switch - 2 gang v2.0 (HGZB-42) as routers. 
Debug info
ZigbeeMQTT version: 1.16.2-dev commit: 2468d61
Adapter hardware: zzh!
Adapter firmware version: 20201026 and 20201113 (develop)

_Originally posted by @juslex in https://github.com/Koenkk/zigbee2mqtt/discussions/5266#discussioncomment-233455_

juslex, 2020-12-24:
I did a herdsman debug log trying to add the devices below and in the following order:

1) SNZB-02 - 0x00124b002214cc3a - Sensor Presença Escritório
2) SJCGQ11LM - 0x00158d00042d3335 - Sensor de Chuva
3) SNZB-03 - 0x00124b001faf5394 - Sensor Presença Fundos

[Zigbee-herdsman debug 12-23-20 - 1.txt](https://github.com/Koenkk/zigbee2mqtt/files/5737879/Zigbee-herdsman.debug.12-23-20.-.1.txt)
[Zigbee-herdsman debug 12-23-20 - 2.txt](https://github.com/Koenkk/zigbee2mqtt/files/5737880/Zigbee-herdsman.debug.12-23-20.-.2.txt)
[Zigbee-herdsman debug 12-23-20 - 3.txt](https://github.com/Koenkk/zigbee2mqtt/files/5737881/Zigbee-herdsman.debug.12-23-20.-.3.txt)
[Zigbee-herdsman debug 12-23-20 - 4.txt](https://github.com/Koenkk/zigbee2mqtt/files/5737882/Zigbee-herdsman.debug.12-23-20.-.4.txt)

juslex, 2020-12-24:
@Koenkk

juslex, 2020-12-24:
[image]

juslex, 2020-12-24:
I make a herdsman debug reconfiguring all of my HGZB-42 and HGZB-43 routers.
[image]
[Zigbee-herdsman debug 12-24-20 - reconfigure HGZB-42 e HGZB-43.txt](https://github.com/Koenkk/zigbee2mqtt/files/5737982/Zigbee-herdsman.debug.12-24-20.-.reconfigure.HGZB-42.e.HGZB-43.txt)

Koenkk, 2020-12-24:
> However, after some time, I cannot add any device to the network. I can only do it if I approach the coordinator, so the pairing is successful. However, when taking the end device to its destination, it cannot connect to the nearest router (zemismart switch - 3 gang v2.0 (HGZB-43) and 2 gang v2.0 (HGZB-42 )), even admitting the entry of new devices and pressing the physical button of the end device to perform the interview again.

The only way to debug this is by sniffing the traffic: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html . 

To do this, put the sniffer close to the end device, start sniffing, trigger an action on the end device and share the sniff together with your network key here.

juslex, 2021-01-07:
Dear Koenkk,

The sniff log: https://www.dropbox.com/sh/ow6byf84wixv164/AABbqdGPLB6aCkDz6vQtvdrVa?dl=0

The network key: 01:03:05:07:09:0B:0D:0F:00:02:04:06:08:0A:0C:0D

Koenkk, 2021-01-07:
I see multiple. In which sniff did you do the following:

> To do this, put the sniffer close to the end device, start sniffing, trigger an action on the end device

juslex, 2021-01-10:
Dear Koenkk,

I try again. The network key is the same. Please look this image:

[image]

Sniff file:

https://www.dropbox.com/s/wrheeebf0syaced/trying%20to%20paring%20snzb-04.pcapng?dl=0

juslex, 2021-01-10:
[image]

juslex, 2021-01-10:
[image]

Koenkk, 2021-01-10:
- I see the device tries to join via a router (0x158d00054edd44), what kind of device is this? 
-  Try joining it directly to the coordinator by pairing it close to that

juslex, 2021-01-10:
Dear Koenkk,

I see the device tries to join via a router (0x158d00054edd44), what kind of device is this? 

Is a zemismart 3 gangs switch.
[image]

The end device is a sonoff SNZB-04.

Try joining it directly to the coordinator by pairing it close to that.

pairing in these conditions occurs without a problem.  likewise, when the router is not a zemismart switch.  yesterday after the test I managed to pair it on another router on my network TS0601_roller_blind

As i mentioned at the beginning of the issue the problem only occurs with zemismart swtichs.  even if pairing with the coordinator or next to the cover, when taking the end device to its destination, it does not communicate with the next zemismart router, making it unusable in view of the distance from the cover or coordinator

Koenkk, 2021-01-10:
So looks like we have 2 issues here (which we cannot solve from Zigbee2MQTT)
1. zemiswart switches don't allow joining of devices -> looks like a bug in the fw of the zemismart switches
2. SNZB-04 doesn't choose a new parent. If this only happens when the desired router is a zemismart switch it may also be a bug in the zemismart switch firmware.

juslex, 2021-01-10:
> So looks like we have 2 issues here (which we cannot solve from Zigbee2MQTT)
> 
> 1. zemiswart switches don't allow joining of devices -> looks like a bug in the fw of the zemismart switches
> 2. SNZB-04 doesn't choose a new parent. If this only happens when the desired router is a zemismart switch it may also be a bug in the zemismart switch firmware (not allowing new children)

I agree. One more thing:  these issues do not happen when conbee 2 is the coordinator and CC2531.  I think there may be some incompatibility between the zzh and zemismart firmware.

juslex, 2021-01-10:
is it not possible to try to correct it via Z-Stack?

juslex, 2021-01-10:
One more thing: if i change the pan id and do a new pairing, zemismart will work normally for a while, pairing and bonding with children will normally take place after a while.

Koenkk, 2021-01-10:
@juslex maybe the pan_id is indeed the key to solving this. In https://github.com/Koenkk/zigbee2mqtt/issues/5386#issuecomment-757420629 the response of the zemismart switch is " pan full". The Conbee II uses a different pan id by default so this could explain why you don't have this issue.

juslex, 2021-01-10:
would it be possible to use the conbee panid in zzh to see if we can work around this problem?

Koenkk, 2021-01-10:
If you share a sniff with the conbee I can check it

juslex, 2021-01-13:
Dear Koenkk,

Conbee2 sniff paring with the same end device:

https://www.dropbox.com/s/nclbpgx0797thkv/paring%20with%20conbee2.pcapng?dl=0

Look this pictures:

[image]

[image]

Paring success without error from panid.

Koenkk, 2021-01-13:
I see the default pan_id is used here (0x1a62) , with zzh 0x1a74 is used (what you specified in your `configuration.yaml`, otherwise it would also use `0x1a62`

juslex, 2021-01-13:
yes, i used different panid.  I created a vm with a home assistant to connect the conbee to the switch, in order to avoid problems in my main installation where zzh is the coordinator.

do you need me to keep the same panid in the conbee?  I also used different channels.

one more thing: it happened that if i removed the switch and re-added it, i would be able to pair end devices through it for some time.

juslex, 2021-01-16:
Dear @Koenkk,

how can i help to try to solve this problem?  do you need any specific sniff?

thank you so much for your precious attention

Koenkk, 2021-01-16:
@juslex I cannot solve this problem since it is a bug in the firmware of your switch. Seems that specifying a different pan_id in your configuration fixes it

juslex, 2021-01-16:
what if i stipulate an ext_pan_id is there any chance to resolve it?  changing pan_id only resolves temporarily.

Koenkk, 2021-01-16:
I don't know, you have to try it.

juslex, 2021-01-20:
Dear @Koenkk 

I managed to do the pairing, but with a juggling. If I restarted zigbee2mqtt and as soon as it started I authorize the pairing of new devices, for a brief moment, the pairing occurs with the nearest zemismart switch, but after a few minutes the problem recurs. After this new fact, isn't there any way to get around this?

Thank you

Koenkk, 2021-01-20:
@juslex if the error is still "PAN full" like https://github.com/Koenkk/zigbee2mqtt/issues/5386#issuecomment-757420629, no

closed 2021-02-27 by github-actions[bot]

k7franklin, 2021-05-12:
Hi, Koenkk. I`m having similar problems with my zzh. I can't set permit_join to a selective router. They doesn't hear any end device pairing try...

If i select permit_join ALL, works fine, but some end devices doens't pair to an router that is about 50cm near, but choose to pair to coordinator that is about 6 or 7 meters...

I use zbmini routers and MOES routers. 

Tried to flash an early firmware, but the same problem is here...

You doesn't think this is an z2m problem?  thanks in advance