Koenkk/zigbee2mqtt #5392: E1743 - IKEA TRADFRI ON/OFF switch - no mqtt messages at all

JacekKac, 2020-12-24:
Device sucessfully paired and configured :
  '0xec1bbdfffeef2dbf':
    friendly_name: ikeaswitch01
    legacy: false

+ zigbee2mqtt/bridge/unbind/ikeaswitch01

after that -  no payload messages at all visible in log or  mosquitto_sub -t 'zigbee2mqtt/ikeaswitch01'. Tried repairing, force_remove and didnt help 

**start log**
Starting Zigbee2MQTT version 1.15.0-dev (commit #unknown)
Coordinator firmware version: '{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20180507,"transportrev":2},"type":"zStack12"}'
 Currently 7 devices are joined:
info  2020-12-24 10:57:49: ikeaswitch01 (0xec1bbdfffeef2dbf): E1743 - IKEA TRADFRI ON/OFF switch (EndDevice)
**cat database.db:** 
{"id":8,"type":"EndDevice","ieeeAddr":"0xec1bbdfffeef2dbf","nwkAddr":36403,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Battery","modelId":"TRADFRI on/off switch","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":2080,"inClusterList":[0,1,3,9,32,4096,64636],"outClusterList":[3,4,6,8,25,258,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TRADFRI on/off switch","manufacturerName":"IKEA of Sweden","powerSource":3,"zclVersion":3,"appVersion":33,"stackVersion":98,"hwVersion":1,"dateCode":"20190723","swBuildId":"2.2.010"}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0018e2a1ed","endpointID":1}],"meta":{}}},"appVersion":33,"stackVersion":98,"hwVersion":1,"dateCode":"20190723","swBuildId":"2.2.010","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1608803259423}

drybrdl, 2020-12-24:
all of my tradfri devices stopped working a couple of days ago

JacekKac, 2020-12-24:
> all of my tradfri devices stopped working a couple of days ago

I use also 4 ikea bulbes  and they are working ok

JacekKac, 2020-12-30:
upgrading coordinator firmware to :
{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"**revision":20190608**,"transportrev":2},"type":"zStack12"}'
debug 2020-12-30 15:18:31: Zigbee network parameters: {"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754}
solved my issue

closed 2020-12-30 by JacekKac