Koenkk/zigbee2mqtt #5394: IKEA LED1537R6 works in group but doesn't work as standalone device

nosorozec, 2020-12-24:
## What happened
I have a three TRADFRI bulbs with the following configuration (fw: 20200312, 2.0.029):

```
devices:
  '0xbc33acfffe61b402':
    friendly_name: LivingTV1
  '0xbc33acfffe5b8d5e':
    friendly_name: LivingTV2
  '0xbc33acfffe5c0a9f':
    friendly_name: LivingTV3

groups:
  '13':
    friendly_name: LivingWallTV
    retain: false
    transition: 0
    optimistic: true
    devices:
      - '0xbc33acfffe61b402'
      - '0xbc33acfffe5b8d5e'
      - '0xbc33acfffe5c0a9f'
```

## What did you expect to happen
I should be able to control each individual bulb with with mqtt topic: `zigbee2mqtt/<friendly name>/set` and also as a group ` zigbee2mqtt\LivingWallTV\set`. The funny thing is that group topics work perfectly well - **but** one of the bulb doesn't work as single unit.

when I want to turn on one of the bulb I got the following error:
```
Zigbee2MQTT:error 2020-12-24 12:38:36: Publish 'set' 'state' to 'LivingTV3' failed: 'Error: Command 0xbc33acfffe5c0a9f/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (no response received)'

Zigbee2MQTT:error 2020-12-24 12:38:41: Publish 'set' 'brightness' to 'LivingTV3' failed: 'Error: Command 0xbc33acfffe5c0a9f/1 genLevelCtrl.moveToLevelWithOnOff({"level":253,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (no response received)'

Zigbee2MQTT:error 2020-12-24 12:38:46: Publish 'set' 'state' to 'LivingTV3' failed: 'Error: Command 0xbc33acfffe5c0a9f/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (no response received)'

Zigbee2MQTT:error 2020-12-24 12:39:41: Publish 'set' 'color_temp' to 'LivingTV3' failed: 'Error: Command 0xbc33acfffe5c0a9f/1 lightingColorCtrl.moveToColorTemp({"colortemp":500,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (no response received)'
```

## How to reproduce it (minimal and precise)

this **doesn't work** and produces error:
```bash
mosquitto_pub -t 'zigbee2mqtt/LivingTV3/set' -m '{"state": "on"}'
```

this **does work**:
```
mosquitto_pub -t 'zigbee2mqtt/LivingWallTV/set' -m '{"state": "on"}'
```

## Debug info
Zigbee2MQTT version: 1.16.2 commit: 04c15f7
Adapter hardware: Conbee II
Adapter firmware version: deCONZ firmware version 26680700

Koenkk, 2020-12-24:
@chrishae looks like a deconz specific issue. 

@nosorozec can you provide the herdsman debug log of this?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

nosorozec, 2020-12-24:
> @ChrisHae looks like a deconz specific issue.
> 
> @nosorozec can you provide the herdsman debug log of this?
> 
> To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

Please have a look the attached file:
[z2m.log.gz](https://github.com/Koenkk/zigbee2mqtt/files/5740511/z2m.log.gz)

this is different bulb:
```yaml
devices:
  '0xbc33acfffe58527e':
    friendly_name: AgaDesk
```
with the exactly same issue (herdsman and z2m were enabled).

frankivo, 2020-12-28:
I had the same thing happening to me (IKEA LED1624G9).

Worked around it by force removing the device that was not working. 
After that, I paired it again and put it back in the group.

ChrisHae, 2021-01-13:
I also think repairing could solve the problem.

closed 2021-02-21 by github-actions[bot]

nikdom, 2021-08-28:
@frankivo does it still work for you? I have the same problem, but also force removing and re-pairing leads to the same problem.

frankivo, 2021-08-30:
> 
> 
> @frankivo does it still work for you? I have the same problem, but also force removing and re-pairing leads to the same problem.

Yep. No new problems for me 🤔