Koenkk/zigbee2mqtt #5395: Zigbee network stops working on new hassio install and full snapshot reload

andrejkolesa, 2020-12-24:
Hi,

i am trying to add USB SSD HDD to Raspberypi4.
Now all works OK. I use Mosquito + zigbe2mqtt (edge).+ and Hassio 5.9 32bit on SD card.
I use USB Electrolama zig-a-zig-ah! (zzh!) ( CC26X2R1) with Coordinator FW: CC26X2R1_20201026.zip

So i installed Hassio 5.9 64bit on SSD drive.
Than i restored snapshot that i took before on SD running system.

After restore all looks ok. BUT my zigbee network does not work as it should.
I can see i receive Aqara action clicks and temperatures, but my Tuy device does not work anymore.

I use couple of TS0601 and TS0011 and TS0012.

When i try to repair this Tuya devices i see message: DEVICEID has left the network.
After that nothing happens. Even when i try to repair again, i get same mssage.
It is impossible to repair it again or make it work.

If i connect old SD card to raspberrypi4 all works OK.

Is there something that must be done for CC26X2R1 devices ?

In logfile i can see for Tuya device messages like: …MAC transaction expired. Or …no network route.

andrejkolesa, 2020-12-24:
MY config is:

data_path: /share/zigbee2mqtt
external_converters: []
devices: devices.yaml
groups: groups.yaml
homeassistant: true
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://192.168.1.10:1883'
  user: mqttdoma
  password: password
serial:
  port: /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0
advanced:
  rtscts: false
  log_level: debug
  pan_id: 6754
  channel: 11
  network_key:
    - 1
    - 3
    - 5
    - 7
    - 9
    - 11
    - 13
    - 15
    - 0
    - 2
    - 4
    - 6
    - 8
    - 10
    - 12
    - 13
  availability_blocklist: []
  availability_passlist: []
device_options: {}
blocklist: []
passlist: []
queue: {}
frontend:
  port: 8099
experimental:
  new_api: true
socat:
  enabled: false
  master: 'pty,raw,echo=0,link=/dev/ttyZ2M,mode=777'
  slave: 'tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5'
  restartdelay: 1
  initialdelay: 1
  options: '-d -d'
  log: false

MalagaInternet, 2020-12-27:
Hi, 
Maybe not related ... but have you tried a USB extension cable? . Check this: https://www.youtube.com/watch?v=2PK3TrOGWNs

andrejkolesa, 2020-12-29:
> Hi,
> Maybe not related ... but have you tried a USB extension cable? . Check this: https://www.youtube.com/watch?v=2PK3TrOGWNs

Uau this is insane!  After all the hours is spend braking my head.
This actually works. I got USB extender cable, move USB unit away fro PI and it works.

Looks like USB drive really produce many interference. Also my BLE (Bluetooth) devices range is much shorter after adding SSD drive.

Thanks MalagaInternet for great tip.

MalagaInternet, 2020-12-29:
Great to hear!

Koenkk, 2020-12-29:
@MalagaInternet thanks for helping, assuming this can be closed.

closed 2020-12-29 by Koenkk