Koenkk/zigbee2mqtt #5396: In Devices docs there is CO2 everywhere while talking about CO (carbon monoxide)

rduivenvoorde, 2020-12-24:
Before doing a pull  request I'd thought to ask here first.

In the docs I see for all carbon monoxide something in the text docs like:
**"Indicates if CO2 (carbon monoxide) is detected."**
For example:
https://www.zigbee2mqtt.io/devices/W2-Module.html
https://www.zigbee2mqtt.io/devices/SCA01ZB.html
Not sure if it is common to write it like this. To me it would be better to write:
**"Indicates if CO (carbon monoxide) is detected."**
as opposed to (not yet available?) carbon DIoxide sensors: CO2

But as said, maybe this is a common notation ( though not correct in chemical sense :-) )

Koenkk, 2020-12-24:
should indeed be CO, can you make a pr?

rduivenvoorde, 2020-12-24:
Yep, will do

closed 2020-12-24 by rduivenvoorde