Koenkk/zigbee2mqtt #5398: Using synology and latest version of Hass.io the zigbee device is not working anymore

dummys, 2020-12-24:
Hi, I installed the latest version from https://github.com/danielwelch/hassio-zigbee2mqtt and my zigbee device are not working anymore. I can't remember exactly but I think they were a problem the first time I installed the addon, and I had to use the same addon but forked. Is it still the case ? Another question, why when the addon is installed, Hass.io said that the configuration is not correct ?
[image]

Koenkk, 2020-12-24:
Please create an issue about this here: https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/issues

closed 2020-12-24 by Koenkk

dummys, 2020-12-24:
yes but I don't know which one I have to use to be honest .. @Koenkk there is no documentation about this. Why two addons ?

Koenkk, 2020-12-24:
Good question (I don't know the answer tbh). @ciotlosm is https://github.com/danielwelch/hassio-zigbee2mqtt deprecated now and should https://github.com/zigbee2mqtt/hassio-zigbee2mqtt be used?

dummys, 2020-12-24:
I tried to use zigbee2mqtt locally on my computer and it works. The only things that is weird, is that even using disable_led: false  the led is not up. But I can successfully see a device in pairing mode. Is there something to do to made the led up ?

Koenkk, 2020-12-25:
The 3.0 firmware isn't recommended for the CC2531 (https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator#im-a-zigbee2mqtt-user-which-one-should-i-use). All devices should be able to pair with the 1.2 coordinator since 3.0 and 1.2 are backwards compatible

dummys, 2020-12-25:
ok thanks or your info. I re-oredred 3 more CC to test with firmware 1.2 and see why its not working. I also orderd the home assistant blue to get rid of unsupported synology package. Thanks for help,

ciotlosm, 2021-01-04:
@Koenkk Regarding the addons:
https://github.com/danielwelch/hassio-zigbee2mqtt  is still the "official" one as it's still in all docs & stuff. 
 https://github.com/zigbee2mqtt/hassio-zigbee2mqtt will become the official one soon, but when it's going to be so I will make sure we get PRs for:
- docs update
- Removal of duplicates from https://github.com/danielwelch/hassio-zigbee2mqtt  and keep just simple configs for backward compatibility

The add-on is now stable enough I think to be switched, so I will make a ticket to github to break the "fork" to be able to make adjustments.