Koenkk/zigbee2mqtt #5399: Bitron AV2010/25

kevincaradant, 2020-12-24:
Please, could you fix the value in "watt" ? Since latest version in dev branch, I have some wrong values. I think you forget a division by 10 :)

[image]
 
[image]

Thank you :)

Koenkk, 2020-12-24:
Fixed.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

kevincaradant, 2020-12-25:
still having the issue ;)

Koenkk, 2020-12-27:
Can you try experimenting with the divisor and check when you get correct values? To do this update devices.js and edit the `divisor` value. Currently it's `100`, but try with e.g. `1000` and, restart zigbee2mqtt and do a reconfigure via the zigbee2mqtt ui.

```js
{
    zigbeeModel: ['902010/25'],
    model: 'AV2010/25',
    vendor: 'Bitron',
    description: 'Video wireless socket',
    fromZigbee: [fz.on_off, fz.metering],
    toZigbee: [tz.on_off],
    exposes: [e.switch(), e.power()],
    meta: {configureKey: 3},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'seMetering']);
        await reporting.instantaneousDemand(endpoint);
        endpoint.saveClusterAttributeKeyValue('seMetering', {divisor: 100, multiplier: 1});
    },
},
```

See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html on how to modify devices.js

kevincaradant, 2020-12-27:
I did some tests and I got the real value using divisor at 10000

However, under HA I have a value but honestly I don't know why ... it's not documented and I don't understand what is it 'energy'

[image]

The power consumption is 'power' but 'energy' should not exist ?

Koenkk, 2020-12-28:
Energy is the sum of the measured power in kWh, I've fixed the divisor.

closed 2020-12-28 by kevincaradant