Koenkk/zigbee2mqtt #5404: No power information receive BW-SHP13 (TS0121)

Kalypox, 2020-12-25:
Is it necessary to open issues for the dev branch ?

## What happened
My BW-SHP13 no longer send power informations (the value in W).
It works on the image:latest but not on the image:latest-dev

## How to reproduce it (minimal and precise)
Pull the state of the device with "get" and payload `state : ""` and the returned value for power is 0

1igeza, 2020-12-25:
For me it's working OK on newest dev. 
```
zigbee2mqtt version
    1.16.2-dev commit: 99c77d2
coordinator type
    zStack3x0
coordinator revision
    20201113
frontend version
    0.2.110
```
```
Dec 26 00:01:49 zigbee2mqt npm[19744]: Zigbee2MQTT:debug 2020-12-26 00:01:49: Received MQTT message on 'zigbee2mqtt/SHP13/get/state' with data ''
Dec 26 00:01:49 zigbee2mqt npm[19744]: Zigbee2MQTT:debug 2020-12-26 00:01:49: Publishing get 'get' 'state' to 'SHP13'
Dec 26 00:01:49 zigbee2mqt npm[19744]: Zigbee2MQTT:debug 2020-12-26 00:01:49: Received Zigbee message from 'SHP13', type 'readResponse', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Dec 26 00:01:49 zigbee2mqt npm[19744]: Zigbee2MQTT:info  2020-12-26 00:01:49: MQTT publish: topic 'zigbee2mqtt/SHP13', payload '{"current":0.01,"energy":15.03,"last_seen":"2020-12-26T00:01:49+01:00","linkquality":135,"power":2,"state":"ON","voltage":228}'

```

Kalypox, 2020-12-25:
@1igeza,

I just tried to update my docker : it works now !

Developers are faster than me ^^
My release date from 12/23.

Kalypox, 2020-12-26:
I reopen because the problem is still here.
The power value is stuck on 600W

As you can see:
[image]

The energy doesn't grow up, because the switch is OFF but the power say 600W

Thanks !

Kalypox, 2021-01-04:
The 1.17.0 fixed that

closed 2021-01-04 by Kalypox