Koenkk/zigbee2mqtt #5409: Xiaomi WSDCGQ11LM sensor on Zigbee2MQTT 1.16.2 does not show temperature below -25 Celsius

AndreyOsadchiy, 2020-12-26:
I
[image]
[image]
[image]
[image]

n my region, temperatures were established from -35 to -40 degrees Celsius.

From the Xiaomi WSDCGQ11LM Sensor, I cannot get the temperature below -25 degrees Celsius.
At the same time, I get up-to-date readings of pressure and humidity without any problems, that is, there is a connection with the sensor and it transmits some readings

## Debug info
Zigbee2MQTT version: 1.16.2
Adapter hardware: CC2531

kevincaradant, 2020-12-26:
What idea to live in Russia also :D. I'm joking of course. 

However, I think that the WSDCGQ11LM is not a good idea for you because it can work at very high temperature (60 degrees) but not very low (only -20). You are lucky that it works until -25 but you will be stick with that :(

On google, I found this: `Temperature detection range and accuracy: -20 ℃ ~ + 60 ℃, ± 0.3 ℃.`

AndreyOsadchiy, 2020-12-27:
The battery is not a problem. If the device transmitted at least some temperature readings below -25, then it would be possible to connect it to batteries located inside the room (for example, 2 AA cells via a pair of wires) or use a CR battery with a temperature range of -40 degrees Celsius ...
In the Russian-language forums on zigbee devices, I saw a mention that it is zigbee2mqtt that filters the sensor data and cuts off readings below -25 degrees Celsius.

Interestingly, when the temperature rises, the battery capacity also returns to normal.

kevincaradant, 2020-12-27:
alright maybe there is also an issue with Z2M in this case. I let others investigate ')

Koenkk, 2020-12-27:
The min temperature was indeed limited to -25 as sometimes this sensor provided unrealistic values 
 (e.g. -100). (https://github.com/Koenkk/zigbee2mqtt/issues/798) 

I've extended this range now to -65 till +65. 

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-12-27 by Koenkk