Koenkk/zigbee2mqtt #5410: Add !secret support in config file

caddyblack, 2020-12-26:
In the config file I have to fill in the MQTT server login details.
Request to have !secret support for these parameters.

deosrc, 2020-12-26:
I believe this is already a feature: https://www.zigbee2mqtt.io/information/configuration.html#specifying-mqtt-userpassword-and-network_key-in-a-different-file

deosrc, 2020-12-27:
Probably worth noting that this is not available for the frontend auth_token:

https://github.com/Koenkk/zigbee2mqtt/blob/04c15f7fe8abaa9534ea565b0fa0cd796652d69e/lib/util/settings.js#L318

Koenkk, 2020-12-27:
As @deosrc this is already supported.

closed 2020-12-27 by Koenkk

caddyblack, 2020-12-27:
Thanks for your reply.  The !secret works well. It has to be in quotes like **'!secret mqttpw'**
Issue can be closed

teeling, 2021-02-13:
In my configuration.yaml I have:
auth_token: '!secret frontend_auth_token'
but the front end doesn't let me in with the string in the secret.yaml, but with the string '!secret frontend_auth_token' instead.

Would it be possible to add this for the "auth_token" setting in the configuration.yaml as well?