Koenkk/zigbee2mqtt #5411: Philips Hue Cher ceiling light (with Bluetooth) 4096730P6

WouterJN, 2020-12-26:
# Bug Report
## What happened

The new Philips Hue Cher ceiling light (with Bluetooth) 4096730P6 is not supported 

Can somebody pls add this light, I assume the functions are the same as the existing Cher light:

https://www.zigbee2mqtt.io/devices/4096730U7.html

WouterJN, 2020-12-26:
During pairing the light doesn’t give a “zigbeemodel”. I tried to use “4096730P6” as model but then the pairing is still not working.

WouterJN, 2020-12-27:
Zigbee_herdsman debug log:
`
2020-12-27 11:19:28 | stdout | [31mZigbee2MQTT:error[39m 2020-12-27 11:19:28: Failed to interview '0x0017880108588e57', device has not successfully been paired
-- | -- | --
2020-12-27 11:19:28 | stderr | 2020-12-27T11:19:28.024Z zigbee-herdsman:controller:error Interview failed for '0x0017880108588e57 with error 'Error: Interview failed because can not get node descriptor ('0x0017880108588e57')'
2020-12-27 11:19:28 | stderr | at rejected (/app/node_modules/zigbee-herdsman/dist/controller/model/device.js:25:65)'
2020-12-27 11:19:28 | stderr | at Generator.throw (<anonymous>)
2020-12-27 11:19:28 | stderr | at Device.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/device.js:350:23)
2020-12-27 11:19:28 | stderr | 2020-12-27T11:19:28.022Z zigbee-herdsman:controller:device:log Interview - failed for device '0x0017880108588e57' with error 'Error: Interview failed because can not get node descriptor ('0x0017880108588e57')
2020-12-27 11:19:28 | stderr | 2020-12-27T11:19:28.022Z zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x0017880108588e57', attempt 6
2020-12-27 11:19:24 | stderr | 2020-12-27T11:19:24.581Z zigbee-herdsman:controller:log Not interviewing '0x0017880108588e57', completed 'false', in progress 'true'
2020-12-27 11:19:24 | stderr | 2020-12-27T11:19:24.581Z zigbee-herdsman:controller:log Device '0x0017880108588e57' accepted by handler
2020-12-27 11:19:24 | stderr | 2020-12-27T11:19:24.581Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-12-27 11:19:24 | stderr | 2020-12-27T11:19:24.581Z zigbee-herdsman:controller:log Device '0x0017880108588e57' joined
2020-12-27 11:19:24 | stderr | 2020-12-27T11:19:24.581Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":13465,"extaddr":"0x0017880108588e57","parentaddr":0}
2020-12-27 11:19:24 | stderr | 2020-12-27T11:19:24.581Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 12 - 2 - 5 - 202 - [153,52,87,142,88,8,1,136,23,0,0,0] - 57
2020-12-27 11:19:24 | stderr | 2020-12-27T11:19:24.580Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,12,69,202,153,52,87,142,88,8,1,136,23,0,0,0,57]
2020-12-27 11:19:24 | stderr | 2020-12-27T11:19:24.580Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [88,8,1,136,23,0,0,0,57]
2020-12-27 11:19:24 | stderr | 2020-12-27T11:19:24.578Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,12,69,202,153,52,87,142]
2020-12-27 11:19:24 | stderr | 2020-12-27T11:19:24.578Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,12,69,202,153,52,87,142]
2020-12-27 11:19:22 | stderr | 2020-12-27T11:19:22.082Z zigbee-herdsman:controller:log Not interviewing '0x0017880108588e57', completed 'false', in progress 'true'
2020-12-27 11:19:22 | stderr | 2020-12-27T11:19:22.082Z zigbee-herdsman:controller:log Device '0x0017880108588e57' accepted by handler
2020-12-27 11:19:22 | stderr | 2020-12-27T11:19:22.081Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-12-27 11:19:22 | stderr | 2020-12-27T11:19:22.081Z zigbee-herdsman:controller:log Device '0x0017880108588e57' joined
2020-12-27 11:19:22 | stderr | 2020-12-27T11:19:22.081Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":13465,"extaddr":"0x0017880108588e57","parentaddr":0}
2020-12-27 11:19:22 | stderr | 2020-12-27T11:19:22.081Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 12 - 2 - 5 - 202 - [153,52,87,142,88,8,1,136,23,0,0,0] - 57
2020-12-27 11:19:22 | stderr | 2020-12-27T11:19:22.081Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,12,69,202,153,52,87,142,88,8,1,136,23,0,0,0,57]
2020-12-27 11:19:22 | stderr | 2020-12-27T11:19:22.081Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [88,8,1,136,23,0,0,0,57]
2020-12-27 11:19:22 | stderr | 2020-12-27T11:19:22.080Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,12,69,202,153,52,87,142]
2020-12-27 11:19:22 | stderr | 2020-12-27T11:19:22.080Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,12,69,202,153,52,87,142]
2020-12-27 11:19:21 | stderr | 2020-12-27T11:19:21.524Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []

`

WouterJN, 2020-12-27:
I got it working, by locating the light next to the Zigbee-stick. Normally this is not working for Hue lights.

In the next few days I will make a pull request at the zigbee_herdsman Github Page.

`    {
        zigbeeModel: ['4096730P6'],
        model: '4096730P6',
        vendor: 'Philips',
        description: 'Hue Cher ceiling light',
        meta: {turnsOffAtBrightness1: true},
        extend: preset.hue.light_onoff_brightness_colortemp,
        ota: ota.zigbeeOTA,
    },`
[image]

Koenkk, 2020-12-27:
Closing this as https://github.com/Koenkk/zigbee-herdsman-converters/pull/1982 is merged.

closed 2020-12-27 by Koenkk