Koenkk/zigbee2mqtt #5413: Brightness set to 1 on turn on after several turn off commands

Mr-Groch, 2020-12-26:
## What happened
When I publish `{"state":"OFF"}` and then next `{"state":"OFF"}` to light then after `{"state":"ON}` it will have `brightness = 1` instead off previous brightness value. This happens when I manually turn some light off, and next I have routine to turn off all lights when I go to sleep. Next day lights will be turned on at `brightness = 1`

## What did you expect to happen
Multiple turn off commands should not affect previously stored brigtness value

## How to reproduce it (minimal and precise)
1. I have light with "`brightness":254`, and then send `{"state":"OFF"}`
```
debug 2020-12-26 18:18:09: Received MQTT message on 'zigbee2mqtt/Taśma LED RGBW - prysznic/set' with data '{"state":"OFF"}'
debug 2020-12-26 18:18:09: Publishing 'set' 'state' to 'Taśma LED RGBW - prysznic'
info  2020-12-26 18:18:09: MQTT publish: topic 'zigbee2mqtt/Taśma LED RGBW - prysznic', payload '{"brightness":254,"color":{"hue":0,"x":0.386,"y":0.564},"color_temp":237,"linkquality":66,"state":"OFF"}'
debug 2020-12-26 18:18:09: Received Zigbee message from 'Taśma LED RGBW - prysznic', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":1}' from endpoint 1 with groupID 0
info  2020-12-26 18:18:09: MQTT publish: topic 'zigbee2mqtt/Taśma LED RGBW - prysznic', payload '{"brightness":1,"color":{"hue":0,"x":0.386,"y":0.564},"color_temp":237,"linkquality":66,"state":"OFF"}'
debug 2020-12-26 18:18:09: Received Zigbee message from 'Taśma LED RGBW - prysznic', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
info  2020-12-26 18:18:09: MQTT publish: topic 'zigbee2mqtt/Taśma LED RGBW - prysznic', payload '{"brightness":1,"color":{"hue":0,"x":0.386,"y":0.564},"color_temp":237,"linkquality":66,"state":"OFF"}'
```
2. Next after `{"state":"ON"}` brightness is restored correctly to `"brightness":254`:
```
debug 2020-12-26 18:18:13: Received MQTT message on 'zigbee2mqtt/Taśma LED RGBW - prysznic/set' with data '{"state":"ON"}'
debug 2020-12-26 18:18:13: Publishing 'set' 'state' to 'Taśma LED RGBW - prysznic'
info  2020-12-26 18:18:13: MQTT publish: topic 'zigbee2mqtt/Taśma LED RGBW - prysznic', payload '{"brightness":254,"color":{"hue":0,"x":0.386,"y":0.564},"color_temp":237,"linkquality":66,"state":"ON"}'
debug 2020-12-26 18:18:13: Received Zigbee message from 'Taśma LED RGBW - prysznic', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
info  2020-12-26 18:18:13: MQTT publish: topic 'zigbee2mqtt/Taśma LED RGBW - prysznic', payload '{"brightness":254,"color":{"hue":0,"x":0.386,"y":0.564},"color_temp":237,"linkquality":66,"state":"ON"}'
debug 2020-12-26 18:18:13: Received Zigbee message from 'Taśma LED RGBW - prysznic', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":254}' from endpoint 1 with groupID 0
info  2020-12-26 18:18:13: MQTT publish: topic 'zigbee2mqtt/Taśma LED RGBW - prysznic', payload '{"brightness":254,"color":{"hue":0,"x":0.386,"y":0.564},"color_temp":237,"linkquality":66,"state":"ON"}'
```
3. Send 2x times `{"state":"OFF"}` command:
```
debug 2020-12-26 18:18:19: Received MQTT message on 'zigbee2mqtt/Taśma LED RGBW - prysznic/set' with data '{"state":"OFF"}'
debug 2020-12-26 18:18:19: Publishing 'set' 'state' to 'Taśma LED RGBW - prysznic'
info  2020-12-26 18:18:19: MQTT publish: topic 'zigbee2mqtt/Taśma LED RGBW - prysznic', payload '{"brightness":254,"color":{"hue":0,"x":0.386,"y":0.564},"color_temp":237,"linkquality":66,"state":"OFF"}'
debug 2020-12-26 18:18:19: Received Zigbee message from 'Taśma LED RGBW - prysznic', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
info  2020-12-26 18:18:19: MQTT publish: topic 'zigbee2mqtt/Taśma LED RGBW - prysznic', payload '{"brightness":254,"color":{"hue":0,"x":0.386,"y":0.564},"color_temp":237,"linkquality":66,"state":"OFF"}'
debug 2020-12-26 18:18:19: Received Zigbee message from 'Taśma LED RGBW - prysznic', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":1}' from endpoint 1 with groupID 0
info  2020-12-26 18:18:19: MQTT publish: topic 'zigbee2mqtt/Taśma LED RGBW - prysznic', payload '{"brightness":1,"color":{"hue":0,"x":0.386,"y":0.564},"color_temp":237,"linkquality":69,"state":"OFF"}'
debug 2020-12-26 18:18:23: Received MQTT message on 'zigbee2mqtt/Taśma LED RGBW - prysznic/set' with data '{"state":"OFF"}'
debug 2020-12-26 18:18:23: Publishing 'set' 'state' to 'Taśma LED RGBW - prysznic'
info  2020-12-26 18:18:23: MQTT publish: topic 'zigbee2mqtt/Taśma LED RGBW - prysznic', payload '{"brightness":1,"color":{"hue":0,"x":0.386,"y":0.564},"color_temp":237,"linkquality":66,"state":"OFF"}'
```
4. And then send `{"state":"ON"}` and observe that brightness will not be restored to 254 but it will be 1:
```
debug 2020-12-26 18:18:26: Received MQTT message on 'zigbee2mqtt/Taśma LED RGBW - prysznic/set' with data '{"state":"ON"}'
debug 2020-12-26 18:18:26: Publishing 'set' 'state' to 'Taśma LED RGBW - prysznic'
info  2020-12-26 18:18:26: MQTT publish: topic 'zigbee2mqtt/Taśma LED RGBW - prysznic', payload '{"brightness":1,"color":{"hue":0,"x":0.386,"y":0.564},"color_temp":237,"linkquality":66,"state":"ON"}'
debug 2020-12-26 18:18:26: Received Zigbee message from 'Taśma LED RGBW - prysznic', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
info  2020-12-26 18:18:26: MQTT publish: topic 'zigbee2mqtt/Taśma LED RGBW - prysznic', payload '{"brightness":1,"color":{"hue":0,"x":0.386,"y":0.564},"color_temp":237,"linkquality":66,"state":"ON"}'
```

## Debug info
Zigbee2MQTT version: 1.16.2
Adapter hardware: CC26X2R zzh!
Adapter firmware version: 20201113

Koenkk, 2020-12-27:
This happens because the device itself reports brightness: 1

```debug 2020-12-26 18:18:26: Received Zigbee message from 'Taśma LED RGBW - prysznic', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
```

Do you have https://www.zigbee2mqtt.io/information/report.html enabled?

Mr-Groch, 2020-12-27:
Yes, I have report enabled. But this happens only when set `{"state":"OFF"}` is published twice or more times. When `{"state":"OFF"}` is sended only once, on` {"state":"ON"}` it will have always previous brightness (even after many hours). So it looks that second `{"state":"OFF"}` command changes something. This happens on various bulbs and LED strips (Silvercrest, OSRAM).

Koenkk, 2020-12-27:
Can you provide the herdsman debug log of this?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

Mr-Groch, 2020-12-27:
Yes I can. Here you go:
http://groch.cloud/misc/z2m_herdsman_debug_log_20201227.zip

As you can see:
1. Firstly LED strip brigthness is 254 and it is ON
2.  I'm turning it OFF once, and then ON again - brigthtness after turn on is 254
3.  Next I send 2x OFF command and then ON - brigthness is 1 after turning on this time...

Koenkk, 2020-12-27:
Should be fixed now. Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Also, do you have `transition: 0` set for these devices in configuration.yaml?

Mr-Groch, 2020-12-27:
Yes, currently I have `transition: 0` for all my lights. It's long story... I have been trying with transition > 0  for smooth on/off but this caused some lags in Home Assistant light states when I was controlling lights from there, so now I'm adding transition manually for automations, and ended with `transition: 0` instead of removing it (for easy change if I ever want to change it again)

Koenkk, 2020-12-27:
@Mr-Groch yes there is a difference between not specifying it (without transition it should already work without the fix). 

Assume this can be closed now.

closed 2020-12-27 by Koenkk