Koenkk/zigbee2mqtt #5414: Retained availability topic not cleared after device rename

Mr-Groch, 2020-12-26:
## What happened
When I have availability feature enabled and rename device, then availability topic from old device name will not be deleted

## What did you expect to happen
Availability topic for old device name should be deleted after device rename

## How to reproduce it (minimal and precise)
1. Turn on availability feature
2. Rename any device
3. Observe with MQTT explorer 2 retained availability topics for this device with old and new name

## Debug info
Zigbee2MQTT version: 1.16.2
Adapter hardware:  CC26X2R1 zzh!
Adapter firmware version: 20201113

timoline, 2020-12-28:
I can reproduce this

Koenkk, 2020-12-28:
Fixed.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-12-28 by Koenkk