Koenkk/zigbee2mqtt #5422: OSRAM 4052899926110 wrong color?

jmerifjKriwe, 2020-12-27:
Hi, 

I'm using a https://www.zigbee2mqtt.io/devices/4052899926110.html and realized that if I send

`{"color":{"b":0,"g":0,"r":255},"transition":1}`

It's converted to:

```
color-x  0.7006
color-y | 0.2993

```

Well, at least this is what I got back via MQTT.
Which is rgb(240,18,2) not rgb(255,0,0).

If I try other color codes I also realize that a back-conversion shows me different rgb-color codes.

Do you have an idea why this happens?
Did I miss something during setup?

Thank you in advance.

## Debug info
Zigbee2MQTT version: 1.16.2-dev
Adapter hardware: CC2531
Adapter firmware version: Default Koenkk 08.06.2019

iFloris, 2020-12-28:
Sounds a lot like my #4125 which I have not gotten back to yet

jmerifjKriwe, 2020-12-28:
btw: Debug output:

```
Zigbee2MQTT:debug 2020-12-28 14:46:10: Received MQTT message on 'zigbee2mqtt/Kind1LEDStrip1/set' with data '{"color":{"b":0,"g":0,"r":255},"transition":1}'
Zigbee2MQTT:debug 2020-12-28 14:46:10: Publishing 'set' 'color' to 'Kind1LEDStrip1'
  zigbee-herdsman:controller:endpoint Command 0x7cb03eaa0a005d14/3 lightingColorCtrl.moveToColor({"transtime":10,"colorx":45914,"colory":19615}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) +17s
  zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x7cb03eaa0a005d14:60522/3 (0,0,1) +17s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":60522,"destendpoint":3,"srcendpoint":1,"clusterid":768,"transid":3,"options":0,"radius":30,"len":9,"data":{"type":"Buffer","data":[1,4,7,90,179,159,76,10,0]}} +17s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,19,36,1,106,236,3,1,0,3,3,0,30,9,1,4,7,90,179,159,76,10,0,151] +17s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100] +17s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100 +1ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0} +17s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,3,197] +408ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,3,197] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,3] - 197 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":3} +18s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +2ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,196,106,236,0,4] +31ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,196,106,236,0,4] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 196 - [106,236,0] - 4 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":60522,"relaycount":0,"relaylist":[]} +34ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +2ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,25,68,129,0,0,0,3,106,236,3,1,0,26,0,5,169,100,0,0,5,24,4,11,7,0,106,236,29,7] +17ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,25,68,129,0,0,0,3,106,236,3,1,0,26,0,5,169,100,0,0,5,24,4,11,7,0,106,236,29,7] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 25 - 2 - 4 - 129 - [0,0,0,3,106,236,3,1,0,26,0,5,169,100,0,0,5,24,4,11,7,0,106,236,29] - 7 +1ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":768,"srcaddr":60522,"srcendpoint":3,"dstendpoint":1,"wasbroadcast":0,"linkquality":26,"securityuse":0,"timestamp":6596869,"transseqnumber":0,"len":5,"data":{"type":"Buffer","data":[24,4,11,7,0]}} +21ms
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":4,"manufacturerCode":null,"commandIdentifier":11},"Payload":{"cmdId":7,"statusCode":0}},"address":60522,"endpoint":3,"linkquality":26,"groupID":0}' +18s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +11ms
Zigbee2MQTT:info  2020-12-28 14:46:11: MQTT publish: topic 'zigbee2mqtt/Kind1LEDStrip1', payload '{"brightness":254,"color":{"x":0.7006,"y":0.2993},"color_temp":37,"linkquality":26,"state":"OFF","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:debug 2020-12-28 14:46:11: Publishing 'set' 'transition' to 'Kind1LEDStrip1'
Zigbee2MQTT:info  2020-12-28 14:46:11: MQTT publish: topic 'zigbee2mqtt/Kind1LEDStrip1/state', payload 'OFF'
Zigbee2MQTT:info  2020-12-28 14:46:11: MQTT publish: topic 'zigbee2mqtt/Kind1LEDStrip1/brightness', payload '254'
Zigbee2MQTT:info  2020-12-28 14:46:11: MQTT publish: topic 'zigbee2mqtt/Kind1LEDStrip1/color-x', payload '0.7006'
Zigbee2MQTT:info  2020-12-28 14:46:11: MQTT publish: topic 'zigbee2mqtt/Kind1LEDStrip1/color_temp', payload '37'
Zigbee2MQTT:info  2020-12-28 14:46:11: MQTT publish: topic 'zigbee2mqtt/Kind1LEDStrip1/color-y', payload '0.2993'
Zigbee2MQTT:info  2020-12-28 14:46:11: MQTT publish: topic 'zigbee2mqtt/Kind1LEDStrip1/update-state', payload 'idle'
Zigbee2MQTT:info  2020-12-28 14:46:11: MQTT publish: topic 'zigbee2mqtt/Kind1LEDStrip1/update_available', payload 'false'
Zigbee2MQTT:info  2020-12-28 14:46:11: MQTT publish: topic 'zigbee2mqtt/Kind1LEDStrip1/linkquality', payload '26'
```

jmerifjKriwe, 2021-01-04:
If you need more debug information, just ask :-)

Koenkk, 2021-01-04:
This probably happens because a gamma correction is applied. 

Try disabling it by changing https://github.com/Koenkk/zigbee-herdsman-converters/blob/c9ed6bb75a9682cf60b41e4e7647c653480f9a6c/lib/utils.js#L184 to `const rgb = {r: red, g: green, b: blue};`

See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html on how to modify lib/utils.js

closed 2021-02-12 by github-actions[bot]