Koenkk/zigbee2mqtt #5423: OSRAM Motion unable to Calibrate Temperature

max0x19, 2020-12-27:
Hello.
I have the same problem with the Osram Motion AC01353010G.
Below a part of configuration.yaml:

devices:
'0x00158d00052dcfgh':
	friendly_name: aqaraweather_ext
'0x000d6f000f5e4569':
	friendly_name: 'osram_motion'
	device_options:
	temperature_calibration: 2

The **temperature_calibration** doesn't work.
Could you help me, please?
Regards

_Originally posted by @max0x19 in https://github.com/Koenkk/zigbee2mqtt/issues/2037#issuecomment-750926700_

Koenkk, 2020-12-28:
Can you provide the debug log of the AC01353010G reporting it's temperature?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

max0x19, 2020-12-28:
Hi Koenkk.
here below:

-- Logs begin at Fri 2020-12-25 04:55:15 CET. --
Dec 28 19:02:26 hab npm[675]: Zigbee2MQTT:info  2020-12-28 19:02:26: MQTT publis                                                           h: topic 'zigbee2mqtt/osram_motion/occupancy', payload 'true'
Dec 28 19:02:26 hab npm[675]: Zigbee2MQTT:info  2020-12-28 19:02:26: MQTT publis                                                           h: topic 'zigbee2mqtt/osram_motion/tamper', payload 'true'
Dec 28 19:02:26 hab npm[675]: Zigbee2MQTT:info  2020-12-28 19:02:26: MQTT publis                                                           h: topic 'zigbee2mqtt/osram_motion/temperature', payload '13.7'
Dec 28 19:02:29 hab npm[675]: Zigbee2MQTT:debug 2020-12-28 19:02:29: Received Zi                                                           gbee message from 'osram_motion', type 'commandStatusChangeNotification', cluste                                                           r 'ssIasZone', data '{"extendedstatus":0,"zonestatus":36}' from endpoint 1 with                                                            groupID 0
Dec 28 19:02:29 hab npm[675]: Zigbee2MQTT:info  2020-12-28 19:02:29: MQTT publis                                                           h: topic 'zigbee2mqtt/osram_motion', payload '{"battery_low":false,"linkquality"                                                           :84,"occupancy":false,"tamper":true,"temperature":13.7}'
Dec 28 19:02:29 hab npm[675]: Zigbee2MQTT:info  2020-12-28 19:02:29: MQTT publis                                                           h: topic 'zigbee2mqtt/osram_motion/battery_low', payload 'false'
Dec 28 19:02:29 hab npm[675]: Zigbee2MQTT:info  2020-12-28 19:02:29: MQTT publis                                                           h: topic 'zigbee2mqtt/osram_motion/linkquality', payload '84'
Dec 28 19:02:29 hab npm[675]: Zigbee2MQTT:info  2020-12-28 19:02:29: MQTT publis                                                           h: topic 'zigbee2mqtt/osram_motion/occupancy', payload 'false'
Dec 28 19:02:29 hab npm[675]: Zigbee2MQTT:info  2020-12-28 19:02:29: MQTT publis                                                           h: topic 'zigbee2mqtt/osram_motion/tamper', payload 'true'
Dec 28 19:02:29 hab npm[675]: Zigbee2MQTT:info  2020-12-28 19:02:29: MQTT publis                                                           h: topic 'zigbee2mqtt/osram_motion/temperature', payload '13.7'

Note:
I'm using the option:
experimental:
  output: attribute_and_json

Many thanks

Koenkk, 2020-12-28:
In this Zigbee message occupancy is reported, not temperature, wait for a message like: `Received Zigbee message from 'osram_motion', type 'msTemperatureMeasurement'`.

max0x19, 2020-12-28:
ok Koenkk

debug 2020-12-28 20:01:48: Received Zigbee message from 'osram_motion', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":1504}' from endpoint 1 with groupID 0
info  2020-12-28 20:01:48: MQTT publish: topic 'zigbee2mqtt/osram_motion', payload '{"battery_low":false,"linkquality":44,"occupancy":false,"tamper":true,"temperature":15.04}'
info  2020-12-28 20:01:48: MQTT publish: topic 'zigbee2mqtt/osram_motion/battery_low', payload 'false'
info  2020-12-28 20:01:48: MQTT publish: topic 'zigbee2mqtt/osram_motion/linkquality', payload '44'
info  2020-12-28 20:01:48: MQTT publish: topic 'zigbee2mqtt/osram_motion/occupancy', payload 'false'
info  2020-12-28 20:01:48: MQTT publish: topic 'zigbee2mqtt/osram_motion/tamper', payload 'true'
info  2020-12-28 20:01:48: MQTT publish: topic 'zigbee2mqtt/osram_motion/temperature', payload '15.04'
debug 2020-12-28 20:02:05: Saving state to file /opt/zigbee2mqtt/data/state.json

thanks

Koenkk, 2020-12-28:
I think you made a mistake in your config, it should be:

```yaml
devices:
  '0x00158d00052dcfgh':
    friendly_name: aqaraweather_ext
  '0x000d6f000f5e4569':
    friendly_name: 'osram_motion'
    temperature_calibration: 2
```

max0x19, 2020-12-29:
OK Koenkk.
Now it seems works, after reboot my RPI4 with OH2.5.
How can check if calibration it's applied?
Probably it is better update the documentation
thanks

in attached the reference
https://ibb.co/DfDhDqg

Koenkk, 2020-12-29:
To check if calibration is applied compare the received value against the published one. In the log your provided: ` Received Zigbee message from 'osram_motion', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":1504}' from endpoint 1 with groupID 0` it reports 15.04 degrees, you should see 17 being published.

The documentation is correct, with device_options you can apply it for all devices. e.g. 

```yaml
devices:
  '0x00158d00052dcfgh':
    friendly_name: aqaraweather_ext
  '0x000d6f000f5e4569':
    friendly_name: 'osram_motion'

device_options:
    temperature_calibration: 2
```

closed 2020-12-29 by Koenkk