Koenkk/zigbee2mqtt #5425: Xiaomi  WXKG02LM Aqara double key wireless wall switch no longer works completely since the update

salopette, 2020-12-28:
Hello, since yesterday's update we only sent single_left and single_right  and single_both to domoticz.
the other functions remain unsuccessful.

[image]
[image]
[image]
[image]

Koenkk, 2020-12-28:
What other actions did it provide? In https://github.com/Koenkk/zigbee2mqtt/issues/5132#issuecomment-751332119 it looked like this switch only supports the single action (I've confirmed this with my own WXKG02LM)

salopette, 2020-12-28:
Everything that can be seen in the picture worked before!
All clicks are also displayed in FrontEnd, only they no longer arrive in Domoticz.

[image]

Koenkk, 2020-12-28:
Can you provide the Zigbee2MQTT debug log when triggering those actions?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

salopette, 2020-12-28:
`INFO: MQTT publish: topic 'zigbee2mqtt/SW-Andrej-Bett', payload '{"action":"single_left","battery":100,"click":"left","last_seen":"2020-12-28T16:30:22.427Z","linkquality":69,"voltage":3015}'
INFO: MQTT publish: topic 'zigbee2mqtt/SW-Andrej-Bett', payload '{"action":"double_left","battery":100,"click":"left_double","last_seen":"2020-12-28T16:30:26.751Z","linkquality":69,"voltage":3015}'
INFO: MQTT publish: topic 'zigbee2mqtt/SW-Andrej-Bett', payload '{"action":"double_right","battery":100,"click":"right_double","last_seen":"2020-12-28T16:30:33.985Z","linkquality":69,"voltage":3015}'
INFO: MQTT publish: topic 'zigbee2mqtt/SW-Andrej-Bett', payload '{"action":"double_both","battery":100,"click":"both_double","last_seen":"2020-12-28T16:30:39.059Z","linkquality":105,"voltage":3015}'
INFO: MQTT publish: topic 'zigbee2mqtt/SW-Andrej-Bett', payload '{"action":"single_left","battery":100,"click":"left","last_seen":"2020-12-28T16:30:41.305Z","linkquality":69,"voltage":3015}'
INFO: MQTT publish: topic 'zigbee2mqtt/SW-Andrej-Bett', payload '{"action":"single_left","battery":100,"click":"left","last_seen":"2020-12-28T16:30:42.892Z","linkquality":102,"voltage":3015}'
INFO: MQTT publish: topic 'zigbee2mqtt/SW-Andrej-Bett', payload '{"action":"single_right","battery":100,"click":"right","last_seen":"2020-12-28T16:30:44.154Z","linkquality":105,"voltage":3015}'
INFO: MQTT publish: topic 'zigbee2mqtt/SW-Andrej-Bett', payload '{"action":"single_right","battery":100,"click":"right","last_seen":"2020-12-28T16:30:45.348Z","linkquality":105,"voltage":3015}'
INFO: MQTT publish: topic 'zigbee2mqtt/SW-Andrej-Bett', payload '{"action":"single_left","battery":100,"click":"left","last_seen":"2020-12-28T16:30:46.259Z","linkquality":102,"voltage":3015}'
INFO: MQTT publish: topic 'zigbee2mqtt/SW-Andrej-Bett', payload '{"action":"single_both","battery":100,"click":"both","last_seen":"2020-12-28T16:30:48.394Z","linkquality":102,"voltage":3015}'`

Koenkk, 2020-12-28:
This is not the debug logging, make sure to set it up correctly according to: https://github.com/Koenkk/zigbee2mqtt/issues/5425#issuecomment-751762252

Also annotate what is logged on which action.

salopette, 2020-12-28:
So for me in the config is log_level: debug, but I don't get any other outputs.
Does the state file help you if I download it via the frontend?

[image]

PS: the Sawell thermostats have not received any images since the last update.

[image]

Koenkk, 2020-12-28:
Fixed debug logging not logged on frontend (update to latest dev branch).

> PS: the Sawell thermostats have not received any images since the last update.

Will be fixed once new release is out (1 january)

timoline, 2020-12-28:
works correct now in dev

salopette, 2020-12-28:
> Fixed debug logging not logged on frontend (update to latest dev branch).
> 
> > PS: the Sawell thermostats have not received any images since the last update.
> 
> Will be fixed once new release is out (1 january)

`DEBUG: Received Zigbee message from 'SW-Andrej-Bett', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":1}' from endpoint 1 with groupID 0
DEBUG: Received Zigbee message from 'SW-Andrej-Bett', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":1}' from endpoint 2 with groupID 0
DEBUG: Received Zigbee message from 'SW-Andrej-Bett', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":1}' from endpoint 2 with groupID 0
DEBUG: Received Zigbee message from 'SW-Andrej-Bett', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":1}' from endpoint 3 with groupID 0
DEBUG: Received Zigbee message from 'SW-Andrej-Bett', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":2}' from endpoint 1 with groupID 0
DEBUG: Received Zigbee message from 'SW-Andrej-Bett', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":2}' from endpoint 2 with groupID 0
DEBUG: Received Zigbee message from 'SW-Andrej-Bett', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":1}' from endpoint 1 with groupID 0
DEBUG: Received Zigbee message from 'SW-Andrej-Bett', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":1}' from endpoint 2 with groupID 0
DEBUG: Received Zigbee message from 'SW-Andrej-Bett', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":2}' from endpoint 1 with groupID 0
DEBUG: Received Zigbee message from 'SW-Andrej-Bett', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":2}' from endpoint 2 with groupID 0
DEBUG: Received Zigbee message from 'SW-Andrej-Bett', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":2}' from endpoint 3 with groupID 0
DEBUG: Received Zigbee message from 'SW-Andrej-Bett', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":1}' from endpoint 3 with groupID 0
DEBUG: Received Zigbee message from 'SW-Andrej-Bett', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":0}' from endpoint 3 with groupID 0
DEBUG: Received Zigbee message from 'SW-Andrej-Bett', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":0}' from endpoint 1 with groupID 0
DEBUG: Received Zigbee message from 'SW-Andrej-Bett', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":0}' from endpoint 2 with groupID 0`

Koenkk, 2020-12-28:
Thanks, can you also share the data/database.db entry of this device?

salopette, 2020-12-28:
`{"id":22,"type":"EndDevice","ieeeAddr":"0x00158d0002d7b87a","nwkAddr":39022,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.remote.b286acn01","epList":[1,2,3],"endpoints":{"1":{"epId":1,"inClusterList":[],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"65281":{"1":3015,"3":25,"4":17320,"5":13,"6":[0,44],"8":5129,"10":12993},"modelId":"lumi.remote.b286acn01","appVersion":9}},"genMultistateInput":{"attributes":{"presentValue":1}}},"binds":[],"meta":{}},"2":{"epId":2,"inClusterList":[],"outClusterList":[],"clusters":{"genMultistateInput":{"attributes":{"presentValue":1}}},"binds":[],"meta":{}},"3":{"epId":3,"inClusterList":[],"outClusterList":[],"clusters":{"genMultistateInput":{"attributes":{"presentValue":1}}},"binds":[],"meta":{}}},"appVersion":9,"interviewCompleted":true,"meta":{},"lastSeen":1609177666501}
`

Koenkk, 2020-12-28:
- Do you have legacy: false set for this device in configuration.yaml?
- In https://github.com/Koenkk/zigbee2mqtt/issues/5425#issuecomment-751811663 you provided only the debug log, but please don't filter out the INFO log, I need both.

salopette, 2020-12-28:
> * Haben Sie Legacy: false für dieses Gerät in configuration.yaml festgelegt?
> * In [# 5425 (Kommentar) haben](https://github.com/Koenkk/zigbee2mqtt/issues/5425#issuecomment-751811663) Sie nur das Debug-Protokoll angegeben, aber bitte filtern Sie das INFO-Protokoll nicht heraus, ich brauche beides.

DEBUG: Received Zigbee message from 'SW-Andrej-Bett', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":1}' from endpoint 1 with groupID 0
INFO: MQTT publish: topic 'zigbee2mqtt/SW-Andrej-Bett', payload '{"action":"single_left","battery":100,"click":"left","last_seen":"2020-12-28T18:31:49.052Z","linkquality":51,"voltage":3015}'
DEBUG: Received Zigbee message from 'SW-Andrej-Bett', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":2}' from endpoint 1 with groupID 0
INFO: MQTT publish: topic 'zigbee2mqtt/SW-Andrej-Bett', payload '{"action":"double_left","battery":100,"click":"left_double","last_seen":"2020-12-28T18:32:06.848Z","linkquality":69,"voltage":3015}'
DEBUG: Received Zigbee message from 'SW-Andrej-Bett', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":1}' from endpoint 1 with groupID 0
INFO: MQTT publish: topic 'zigbee2mqtt/SW-Andrej-Bett', payload '{"action":"single_left","battery":100,"click":"left","last_seen":"2020-12-28T18:32:51.715Z","linkquality":66,"voltage":3015}'
DEBUG: Received Zigbee message from 'SW-Andrej-Bett', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":1}' from endpoint 2 with groupID 0
INFO: MQTT publish: topic 'zigbee2mqtt/SW-Andrej-Bett', payload '{"action":"single_right","battery":100,"click":"right","last_seen":"2020-12-28T18:32:54.585Z","linkquality":69,"voltage":3015}'
DEBUG: Received Zigbee message from 'SW-Andrej-Bett', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":2}' from endpoint 1 with groupID 0
INFO: MQTT publish: topic 'zigbee2mqtt/SW-Andrej-Bett', payload '{"action":"double_left","battery":100,"click":"left_double","last_seen":"2020-12-28T18:32:56.430Z","linkquality":69,"voltage":3015}'
INFO: MQTT publish: topic 'zigbee2mqtt/SW-Andrej-Bett', payload '{"action":"double_right","battery":100,"click":"right_double","last_seen":"2020-12-28T18:32:57.773Z","linkquality":72,"voltage":3015}'
INFO: MQTT publish: topic 'zigbee2mqtt/SW-Andrej-Bett', payload '{"action":"hold_left","battery":100,"click":"left_long","last_seen":"2020-12-28T18:32:59.446Z","linkquality":72,"voltage":3015}'
INFO: MQTT publish: topic 'zigbee2mqtt/SW-Andrej-Bett', payload '{"action":"hold_right","battery":100,"click":"right_long","last_seen":"2020-12-28T18:33:01.373Z","linkquality":69,"voltage":3015}'
DEBUG: Received Zigbee message from 'SW-Andrej-Bett', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":1}' from endpoint 3 with groupID 0
INFO: MQTT publish: topic 'zigbee2mqtt/SW-Andrej-Bett', payload '{"action":"single_both","battery":100,"click":"both","last_seen":"2020-12-28T18:33:03.978Z","linkquality":63,"voltage":3015}'
DEBUG: Received Zigbee message from 'SW-Andrej-Bett', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":0}' from endpoint 3 with groupID 0
INFO: MQTT publish: topic 'zigbee2mqtt/SW-Andrej-Bett', payload '{"action":"hold_both","battery":100,"click":"both_long","last_seen":"2020-12-28T18:33:05.848Z","linkquality":69,"voltage":3015}'

Koenkk, 2020-12-28:
- I miss the answer to: Do you have legacy: false set for this device in configuration.yaml?
- I don't understand your problem, in the OP you state only single is published but in your log I also see `double_left`, `hold_right`, etc.

salopette, 2020-12-28:
I didn't make any changes for Legacy.

The problem is that Domoticz only got single_left and single_right and single_both since yesterday. Although in the z2m all other states exist and are also executed.

Koenkk, 2020-12-28:
In that case looks like Zigbee2MQTT is not the issue here, I will close this. I think you have to continue in https://github.com/stas-demydiuk/domoticz-zigbee2mqtt-plugin/issues/531

closed 2020-12-28 by Koenkk

salopette, 2020-12-28:
A day ago z2m sent 12 states to domoticz, now it only sends 3 states.
The Stas plug-in has not been changed during that time!

[image]
[image]

Koenkk, 2020-12-28:
Ah I see, it's about the exposes not about the actual send states. Fixed that.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

salopette, 2020-12-28:
Many Thanks! Now everything is back!

The picture and link will also be maintained on January 1st?

[image]

Koenkk, 2020-12-28:
> The picture and link will also be maintained on January 1st?

yes

salopette, 2020-12-28:
Super!

husnainali09, 2022-08-06:
@Koenkk Hi, Is this fix in the main branch yet (HASS addon). I have the latest and its still only registering single presses.