Koenkk/zigbee2mqtt #5426: GS SSHM-I1

mSapik, 2020-12-28:
GS SSHM-I1 - copy of Heiman smoke sensor, in user guide writen build by Heiman Technology co.
In devices.js i copy heiman hs1sa section and replace model and vendor. The sensor works correctly, transmits its status, only battery level is n/a. Please add support.

## Information about the device + link
https://www.gs.ru/catalog/umnyy-dom/datchik-dyma-gs-sshm-i1/

## data/database.db entry of the device
{"date_code":"2019.12.2","definition":null,"endpoints":{"1":{"bindings":[],"clusters":{"input":["genBasic","genPowerCfg","genIdentify","ssIasZone"],"output":["genOta"]}}},"friendly_name":"0x000d6f00131c50bd","ieee_address":"0x000d6f00131c50bd","interview_completed":true,"interviewing":false,"network_address":54574,"power_source":"Battery","software_build_id":"1.1.1","supported":false,"type":"EndDevice"}]'

Koenkk, 2020-12-28:
Added, thanks!

closed 2020-12-28 by Koenkk

mrmaximas, 2023-12-13:
> Added, thanks!

Hi again @Koenkk 
I have probably same sensor but from Perenio https://perenio.com/catalog/pecss01-smoke-sensor-pecss01
https://perenio.com/userfiles/files/Documentation/Manuals/EN/PECSS01_Smoke_Sensor_Manual_EN.pdf
Detected by z2m like https://www.zigbee2mqtt.io/devices/HS3SA_HS1SA.html
I'm not sure if tamper expose works like here https://www.zigbee2mqtt.io/devices/SSHM-I1.html , but tamper mini-knob is present. I think it can be verified.
DB data
{"id":34,"type":"EndDevice","ieeeAddr":"0x000d6f0011f9f341","nwkAddr":12212,"manufId":4619,"manufName":"HEIMAN","powerSource":"Battery","modelId":"SmokeSensor-EM","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,1280],"outClusterList":[25],"clusters":{"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b002a1aa3c4","zoneState":1}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":185}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b002a1aa3c4","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],"meta":{}}},"appVersion":18,"stackVersion":2,"hwVersion":16,"dateCode":"2018.3.7","swBuildId":"1.1.1","zclVersion":1,"interviewCompleted":true,"meta":{"configured":491743244},"lastSeen":1702487724047,"defaultSendRequestWhen":"immediate"}

Koenkk, 2023-12-13:
`SmokeSensor-EM` is already supported, picture can be different because this is a whitelabel device with the same modelID