Koenkk/zigbee2mqtt #5428: Ubisys D1: Actions from inputs do not register

wb441, 2020-12-28:
## What happened

When pairing the device, it is automatically recognized.
When actuating either of the two inputs, I could not register these inputs in order to couple an action to them. 

## What did you expect to happen

When pushing either of the buttons, at that point there would be a message coming in as is the case for another input device I have, namely containing "action" and "linkquality" values.
 I consulted devices.js, toZigbee and fromZigbee, and as far as I can see, all is well defined in there. 
When briefly running the dev branch, I noticed that the 'manuSpecific..' cluster is not the one from Ubisys as expected, but the one from Legrand. Perhaps this is a clue regarding the setup in herdsman/herdsman-converters?

## How to reproduce it (minimal and precise)

1) Pairing Ubisys D1 to the system
2) Applying 230V to the first input. This results in the light wired from this device to also turn on so on the device, the input is registered.
3) Checking the zigbee2mqtt log and/or mosquitto subscribed to zigbee2mqtt/<FRIENDLY_NAME>, no feedback there.

## Debug info
Zigbee2MQTT version: latest, not dev, downloaded two days ago.
Adapter hardware: CC2652 from Tindie 
Adapter firmware version: CC26X2R1_20201026.hex

wb441, 2020-12-28:
Further to the above, I switched to the dev branch to more easily check out the bindings via the Bind tab.
I expected the above to work based on the doc: https://www.zigbee2mqtt.io/devices/C4.html#configuring-inputs 

> When used with Zigbee2MQTT all endpoints get bound to the coordinator automatically.

Therefore, should I be seeing these bindings here (Axelle is the friendly name I gave the device)?
[image]

wb441, 2020-12-28:
As extra info, this is the log when pairing a D1 on a fresh install
https://hastebin.com/yuminijihu.apache

wb441, 2020-12-29:
@Koenkk  Could you give me some pointers? Many thanks!!

Koenkk, 2020-12-29:
@wb441 I don't know this device so can't help you with the specific part, but have you checked: https://www.zigbee2mqtt.io/devices/D1.html ?

@felixstorm do you maybe know what is wrong here?

wb441, 2020-12-29:
@guydriesen Dear Guy, I see you recently created the option to set the dim method for this device. I'm currently trying to get information from the other end, the inputs, to publish their "action" property if they are actuated. Do you perhaps know if this is possible?
My goal is for the second input of a dimmer to act as an 'alles uit' for multiple D1's. I was hoping that when the input becomes high, this would show up as a message from zigbee2mqtt so I could couple actions to it.

wb441, 2020-12-29:
When I check in devices.js, I find the device: zigbeeModel: ['D1 (5503)', 'D1-R (5603)'].
Incontrast to the S1 and C4, it does not expose 'action' related to the inputs.
`{
        zigbeeModel: ['D1 (5503)', 'D1-R (5603)'],
        model: 'D1',
        vendor: 'Ubisys',
        description: 'Universal dimmer D1',
        fromZigbee: [fz.on_off, fz.brightness, fz.metering],
        toZigbee: [tz.light_onoff_brightness, tz.ballast_config, tz.ubisys_device_setup],
        exposes: [e.light_brightness(), e.power()],`

As reference, here is the S2's definition which shows the inputs being exposed:

`{
        zigbeeModel: ['S2 (5502)', 'S2-R (5602)'],
        model: 'S2',
        vendor: 'Ubisys',
        description: 'Power switch S2',
        exposes: [
            e.switch().withEndpoint('l1'), e.switch().withEndpoint('l2'), e.power(),
            e.action(['toggle_s1', 'toggle_s2', 'on_s1', 'on_s2', 'off_s1', 'off_s2', 'recall_*_s1', 'recal_*_s2', 'brightness_move_up_s1',
                'brightness_move_up_s2', 'brightness_move_down_s1', 'brightness_move_down_s2', 'brightness_stop_s1',
                'brightness_stop_s2'])],
        fromZigbee: [fz.on_off, fz.metering, fz.command_toggle, fz.command_on, fz.command_off, fz.command_recall, fz.command_move,
            fz.command_stop],
        toZigbee: [tz.on_off, tz.ubisys_device_setup],
        endpoint: (device) => {
            return {'l1': 1, 'l2': 2, 's1': 3, 's2': 4};
        },
        meta: {configureKey: 3, multiEndpoint: true},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(5);
            await reporting.bind(endpoint, coordinatorEndpoint, ['seMetering']);
            await reporting.readMeteringMultiplierDivisor(endpoint);
            await reporting.instantaneousDemand(endpoint);
        },
        onEvent: async (type, data, device) => {
            /*
             * As per technical doc page 20 section 7.4.4 and
             *                      page 22 section 7.5.4
             * https://www.ubisys.de/wp-content/uploads/ubisys-s2-technical-reference.pdf
             *
             * This cluster uses the binding table for managing command targets.
             * When factory fresh, this cluster is bound to endpoint #1 to
             * enable local control.
             *
             * This cluster uses the binding table for managing command targets.
             * When factory fresh, this cluster is bound to endpoint #2 to
             * enable local control
             *
             * We use addBinding to 'record' this default binding.
             */
            if (type === 'deviceInterview') {
                const ep1 = device.getEndpoint(1);
                const ep2 = device.getEndpoint(2);
                const ep3 = device.getEndpoint(3);
                const ep4 = device.getEndpoint(4);
                ep3.addBinding('genOnOff', ep1);
                ep4.addBinding('genOnOff', ep2);
            }
        },`

Conclusion seems to be I should expand the D1's device definition similar to the S2's definition.

sjorge, 2020-12-29:
@wb441 I added the actions to S1 and S2 recently as I noticed they are reporting them (even when bound to a group).

I do not own a D1 so I didn't add them there as I am not sure it also reports them, but you can easily test.

Try with this device block instead of the one that is there now:

```
    {
        zigbeeModel: ['D1 (5503)', 'D1-R (5603)'],
        model: 'D1',
        vendor: 'Ubisys',
        description: 'Universal dimmer D1',
        fromZigbee: [fz.on_off, fz.brightness, fz.metering, fz.command_toggle, 
            fz.command_on, fz.command_off, fz.command_move, fz.command_stop],
        toZigbee: [tz.light_onoff_brightness, tz.ballast_config, tz.ubisys_device_setup],
        exposes: [e.light_brightness(), e.power(), e.action([
            'toggle', 'on', 'off',
            'brightness_move_up', 'brightness_move_down', 'brightness_stop',
        ])],
        meta: {configureKey: 3},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(4);
            await reporting.bind(endpoint, coordinatorEndpoint, ['seMetering']);
            await reporting.readMeteringMultiplierDivisor(endpoint);
            await reporting.instantaneousDemand(endpoint);
        },
        ota: ota.ubisys,
    },
```

You can check with debug logging enabled if I forgot to add one. This device could use some more love though as the default binding don't get added yet either. (Well the device has them, z2m doesn't know about them yet)

wb441, 2020-12-29:
@sjorge Got action to come in after also manually binding the input endpoint (2) to the coordinator as well. Progress!

```
{
        zigbeeModel: ['D1 (5503)', 'D1-R (5603)'],
        model: 'D1',
        vendor: 'Ubisys',
        description: 'Universal dimmer D1',
        fromZigbee: [fz.on_off, fz.brightness, fz.metering, fz.command_toggle, fz.command_on, fz.command_off, fz.command_recall, fz.command_move, fz.command_stop],
        toZigbee: [tz.light_onoff_brightness, tz.ballast_config, tz.ubisys_device_setup],
        exposes: [e.light_brightness(), e.power(), e.action([
            'toggle', 'on', 'off',
            'brightness_move_up', 'brightness_move_down', 'brightness_stop',
        ])],
        meta: {configureKey: 3},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(4);
            await reporting.bind(endpoint, coordinatorEndpoint, ['seMetering']);
            await reporting.readMeteringMultiplierDivisor(endpoint);
            await reporting.instantaneousDemand(endpoint);
        },
        onEvent: async (type, data, device) => {
            /*
             * As per technical doc page 18 section 7.3.4
             * https://www.ubisys.de/wp-content/uploads/ubisys-s1-technical-reference.pdf
             *
             * This cluster uses the binding table for managing command targets.
             * When factory fresh, this cluster is bound to endpoint #1 to
             * enable local control.
             *
             * We use addBinding to 'record' this default binding.
             */
            if (type === 'deviceInterview') {
                const ep1 = device.getEndpoint(1);
                const ep2 = device.getEndpoint(2);
                ep2.addBinding('genOnOff', ep1);
            }
        },
        ota: ota.ubisys,
    },
```
Now I'll look deeper how to add the second input as well.
Once it's satisfactory, should I somehow submit this to make it available for other users?

closed 2020-12-29 by wb441

sjorge, 2020-12-30:
Should be in the next dev release now.

guydriesen, 2020-12-30:
> @guydriesen Dear Guy, I see you recently created the option to set the dim method for this device. I'm currently trying to get information from the other end, the inputs, to publish their "action" [...]

Didn't came to that part of the dimmer (yet?) as it is installed behind a single push button. But good to see that it's already fixed :+1: