Koenkk/zigbee2mqtt #5430: Availability: incorrect state upon Z2M startup (with Home Assistant)

schmittner, 2020-12-28:
## What happened

With the availability extension enabled (`availability_timeout` set to a positive non-zero value), Z2M announces every device as available upon startup. Offline devices are shown as `offline` only after the first scan interval expires. This essentially means that Z2M lies about the state of offline devices, which might break automations that rely on the availability state. For me, this problem currently only applies to light bulbs that are turned off via a physical switch. But there might be other device types where where this type of misinformation might wreak more havoc. My argument goes along the lines the discussion in #3496.

## What did you expect to happen

Pingable Zigbee devices should be announced as `offline` until Z2M has made sure that the device is online.

## How to reproduce it (minimal and precise)

1. Restart Z2M
2. When connected to MQTT, Z2M announces all devices as `online`
3. After the `availability_timeout` interval, offline devices appear as `offline`.

## Proposed fix

I want to propose a fix that  involves changes in both the availability and home assistant extensions.

1. We announce pingable devices as `offline` upon startup, and immediately start a ping sweep such that online devices will be detected and announced as such shortly after. Availability state of non-pingable devices should also be checked for immediately after startup and announce availability based on last seen time (we don't know how long Z2M has been shut down).
https://github.com/Koenkk/zigbee2mqtt/blob/d1842ab69ae62a8a59a0de1b0caa86f73d614763/lib/extension/availability.js#L75

2. With (1) fixed, I still see a short `online` period of offline devices when using the home assistant integration because the availability topics include the bridge state. I would argue that if `availability_timeout` is set, the device's availability topics should not include the bridge state. Is there a reason why a device should have two availability topics? However, I believe that this fix would only work for new devices, right? Maybe there is a better solution.
https://github.com/Koenkk/zigbee2mqtt/blob/d1842ab69ae62a8a59a0de1b0caa86f73d614763/lib/extension/homeassistant.js#L684-L687

## Debug info
Zigbee2MQTT version: 1.16.2

Koenkk, 2020-12-28:
> We announce pingable devices as offline upon startup, and immediately start a ping sweep such that online devices will be detected and announced as such shortly after.

Pinging a lot of devices at startup can introduce various problems, due to the big amount of request pings could fail. This would mean that devices are not controllable for the period of availabilty_timeout.

> Is there a reason why a device should have two availability topics?

A mqtt client can only have one last will topic, in Zigbee2MQTT this is zigbee2mqtt/bridge/state, if Zigbee2MQTT crashes no offline will be published to the device availability topics and the devices will remain online.

schmittner, 2020-12-28:
Thanks for the quick response.

> Pinging a lot of devices at startup can introduce various problems, due to the big amount of request pings could fail. This would mean that devices are not controllable for the period of availabilty_timeout.

As far as I understand the code, this *ping storm* happens anyway, just later, because all pings for each device are scheduled after availability_timeout. Does deferring the pings somehow mitigate the collisions? In any case, one could spread the pings across the ping interval window.

Also, the Z2M downtime usually happens due to updates which take a couple of minutes on a Raspberry Pi. I‘d rather accept a slightly extended downtime (60s is the default) than having incorrect information send to HA.
I understand that such a change would make things worse for setups that usually do not have unavailable devices, so this may be a new configuration option?

> A mqtt client can only have one last will topic, in Zigbee2MQTT this is zigbee2mqtt/bridge/state, if Zigbee2MQTT crashes no offline will be published to the device availability topics and the devices will remain online.

Do you know how HA handles multiple availability topic? Does it always take the latest message on any of the topics? If so, could one somehow make the Z2M device-specific availability topic be the most up to date upon connection?

Btw: really appreciate your work 👍

Koenkk, 2020-12-28:
> Do you know how HA handles multiple availability topic? Does it always take the latest message on any of the topics? If so, could one somehow make the Z2M device-specific availability topic be the most up to date upon connection?

It takes the latest published message. But this is kinda problematic. 

If we set the availability only to the device specific one, the device will never be marked offline when z2m crashes. Other solution would be to keep both topics and not publish `online` to `zigbee2mqtt/bridge/state` but this isn't nice either (so I currently don't know how to continue here)

schmittner, 2020-12-29:
> It takes the latest published message. But this is kinda problematic.

Why is this problematic?

> If we set the availability only to the device specific one, the device will never be marked offline when z2m crashes. Other solution would be to keep both topics and not publish online to zigbee2mqtt/bridge/state but this isn't nice either (so I currently don't know how to continue here)

This really appears to be a generic problem of this bridge architecture, as there is only one last will message for all devices. I think a (cleaner) solution could be to configure only the device-specific topic and then use a watchdog that will change the availability topics of all individual devices as soon as `zigbee2mqtt/bridge/state` turns `offline`. In HA, this could be a done using a simple automation using the already available `sensor.zigbee2mqtt_bridge_state`.
What do you think?

Koenkk, 2020-12-29:
> then use a watchdog that will change the availability topics of all individual devices as soon as zigbee2mqtt/bridge/state turns offline

Can you provide some example of this?

schmittner, 2020-12-29:
Something like this should work (obviously it would need one action per device):
```yaml
- alias: 'Zigbee2MQTT: mark devices as unavailable when bridge becomes offline'
  trigger:
  - platform: state
    entity_id: sensor.zigbee2mqtt_bridge_state
    to: offline
  action:
  - service: mqtt.publish
    data:
      topic: zigbee2mqtt/<DEVICE_NAME>/availability
      payload: offline
      retain: true
```
One could probably make this even nicer using loops in a dedicated script.

Koenkk, 2020-12-29:
The per device solution is not acceptable in my opinion. With this we trade one problem for another one. If we can come up with a simple solution to mark devices offline when zigbee2mqtt/bridge/state offline is published it's acceptable for me.

schmittner, 2020-12-29:
Agreed, so it seems this is something that HA needs to change. According to https://github.com/home-assistant/core/pull/37418, multiple availability topics were introduced for gateways. In this case, the behavior of only using the latest message does not make sense IMHO. Instead, HA should mark a device as available only if *all* topics are set to `online`. Do you agree? I would then open an issue with HA.

Koenkk, 2020-12-29:
@schmittner that indeed makes much more sense for this use case.

schmittner, 2020-12-29:
@Koenkk great! Then I'll open an issue there.

However, there remains this problem that Z2M could address:

> As far as I understand the code, this ping storm happens anyway, just later, because all pings for each device are scheduled after availability_timeout. Does deferring the pings somehow mitigate the [...]
I understand that such a change would make things worse for setups that usually do not have unavailable devices, **so this may be a new configuration option?**

Can you comment in this?

Koenkk, 2020-12-29:
That can indeed be fixed from Zigbee2MQTT. Currently planning some availability feature improvements (https://github.com/Koenkk/zigbee2mqtt/issues/5330) and will also fix this here.

schmittner, 2021-01-29:
un-stale since this is still an issue

Koenkk, 2021-02-14:
Closing in favour of https://github.com/Koenkk/zigbee2mqtt/issues/6281

closed 2021-02-14 by Koenkk