Koenkk/zigbee2mqtt #5441: UNREPORTABLE_ATTRIBUTE Error when configuring supported Sinope TH1123ZB Thermostat device

3gyptian, 2020-12-29:
## What happened
Configuration error reported when attempting to connect.  I'm new at using Zigbee2MQTT so my apologies if I'm missing something obvious here.  

**Error message:**
Zigbee2MQTT:error 2020-12-28 16:41:15: Failed to configure '0x500b9140000108ee', attempt 1 (Error: ConfigureReporting 0x500b9140000108ee/1hvacThermostat([{"attribute":"systemMode","minimumReportInterval":1,"maximumReportInterval":0,"reportableChange":null}],{"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Status **'UNREPORTABLE_ATTRIBUTE'**)

## How to reproduce it (minimal and precise)
Two attempts from scratch have produced identical results

## Debug info
Zigbee2MQTT version: version 1.16.2 (commit #d1842ab)
Adapter hardware: Comidox CC2531 Sniffer USfB Dongle 
Adapter firmware version: CC2531_DEFAULT_20190608
MacOS Catalina 10.15.7

Device DB entry:
```
{"id":2,"type":"Router","ieeeAddr":"0x500b9140000108ee","nwkAddr":45891,"manufId":4508,"manufName":"Sinope Technologies\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","powerSource":"Mains (single phase)","modelId":"TH1123ZB\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","epList":[1,196],"endpoints":{"1":{"profId":260,"epId":1,"devId":769,"inClusterList":[0,3,4,5,513,516,1026,2820,2821,65281],"outClusterList":[65281,25],"clusters":{"genBasic":{"attributes":{"modelId":"TH1123ZB\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","manufacturerName":"Sinope Technologies\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","powerSource":1,"zclVersion":1,"appVersion":33,"stackVersion":3,"hwVersion":0}},"hvacThermostat":{"attributes":{"localTemp":2143,"occupiedHeatingSetpoint":1950}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00124b001cce285b","endpointID":1},{"cluster":3,"type":"endpoint","deviceIeeeAddress":"0x00124b001cce285b","endpointID":1},{"cluster":4,"type":"endpoint","deviceIeeeAddress":"0x00124b001cce285b","endpointID":1},{"cluster":513,"type":"endpoint","deviceIeeeAddress":"0x00124b001cce285b","endpointID":1},{"cluster":516,"type":"endpoint","deviceIeeeAddress":"0x00124b001cce285b","endpointID":1},{"cluster":1026,"type":"endpoint","deviceIeeeAddress":"0x00124b001cce285b","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b001cce285b","endpointID":1}],"meta":{}},"196":{"profId":49757,"epId":196,"devId":769,"inClusterList":[1],"outClusterList":[],"clusters":{},"binds":[],"meta":{}}},"appVersion":33,"stackVersion":3,"hwVersion":0,"zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1609204162034}
```

Initial log output during pairing:

```
Zigbee2MQTT:info  2020-12-28 16:40:43: Logging to console and directory: '/Users/mcc/homeassistant/zigbeemqtt/zigbee2mqtt/data/log/2020-12-28.16-40-43' filename: log.txt
Zigbee2MQTT:info  2020-12-28 16:40:43: Starting Zigbee2MQTT version 1.16.2 (commit #d1842ab)
Zigbee2MQTT:info  2020-12-28 16:40:43: Starting zigbee-herdsman (0.13.37)
Zigbee2MQTT:info  2020-12-28 16:40:45: zigbee-herdsman started
Zigbee2MQTT:info  2020-12-28 16:40:45: Coordinator firmware version: '{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190608,"transportrev":2},"type":"zStack12"}'
Zigbee2MQTT:info  2020-12-28 16:40:45: Currently 0 devices are joined:
Zigbee2MQTT:warn  2020-12-28 16:40:45: `permit_join` set to  `true` in configuration.yaml.
Zigbee2MQTT:warn  2020-12-28 16:40:45: Allowing new devices to join.
Zigbee2MQTT:warn  2020-12-28 16:40:45: Set `permit_join` to `false` once you joined all devices.
Zigbee2MQTT:info  2020-12-28 16:40:45: Zigbee: allowing new devices to join.
Zigbee2MQTT:info  2020-12-28 16:40:45: Connecting to MQTT server at mqtt://192.168.86.32
Zigbee2MQTT:info  2020-12-28 16:40:45: Connected to MQTT server
Zigbee2MQTT:info  2020-12-28 16:40:45: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
Zigbee2MQTT:info  2020-12-28 16:40:45: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"d1842ab","coordinator":{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190608,"transportrev":2},"type":"zStack12"},"log_level":"info","network":{"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754},"permit_join":true,"version":"1.16.2"}'
Zigbee2MQTT:info  2020-12-28 16:40:59: Device '0x500b9140000108ee' joined
Zigbee2MQTT:info  2020-12-28 16:40:59: Starting interview of '0x500b9140000108ee'
Zigbee2MQTT:info  2020-12-28 16:40:59: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x500b9140000108ee"},"type":"device_connected"}'
Zigbee2MQTT:info  2020-12-28 16:40:59: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x500b9140000108ee"},"type":"pairing"}'
Zigbee2MQTT:info  2020-12-28 16:40:59: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x500b9140000108ee"},"type":"device_announced"}'
Zigbee2MQTT:info  2020-12-28 16:41:14: Successfully interviewed '0x500b9140000108ee', device has successfully been paired
Zigbee2MQTT:info  2020-12-28 16:41:14: Device '0x500b9140000108ee' is supported, identified as: Sinope Zigbee line volt thermostat (TH1123ZB)
Zigbee2MQTT:info  2020-12-28 16:41:14: Configuring '0x500b9140000108ee'
Zigbee2MQTT:info  2020-12-28 16:41:14: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"Zigbee line volt thermostat","friendly_name":"0x500b9140000108ee","model":"TH1123ZB","supported":true,"vendor":"Sinope"},"type":"pairing"}'
Zigbee2MQTT:error 2020-12-28 16:41:15: Failed to configure '0x500b9140000108ee', attempt 1 (Error: ConfigureReporting 0x500b9140000108ee/1 hvacThermostat([{"attribute":"systemMode","minimumReportInterval":1,"maximumReportInterval":0,"reportableChange":null}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Status 'UNREPORTABLE_ATTRIBUTE')
    at Endpoint.checkStatus (/Users/mcc/homeassistant/zigbeemqtt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:173:23)
    at Endpoint.<anonymous> (/Users/mcc/homeassistant/zigbeemqtt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:396:26)
    at Generator.next (<anonymous>)
    at fulfilled (/Users/mcc/homeassistant/zigbeemqtt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:24:58))
Zigbee2MQTT:info  2020-12-28 16:41:18: Configuring '0x500b9140000108ee'
Zigbee2MQTT:error 2020-12-28 16:41:21: Failed to configure '0x500b9140000108ee', attempt 2 (Error: ConfigureReporting 0x500b9140000108ee/1 hvacThermostat([{"attribute":"systemMode","minimumReportInterval":1,"maximumReportInterval":0,"reportableChange":null}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Status 'UNREPORTABLE_ATTRIBUTE')
    at Endpoint.checkStatus (/Users/mcc/homeassistant/zigbeemqtt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:173:23)
    at Endpoint.<anonymous> (/Users/mcc/homeassistant/zigbeemqtt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:396:26)
    at Generator.next (<anonymous>)
    at fulfilled (/Users/mcc/homeassistant/zigbeemqtt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:24:58))
Zigbee2MQTT:info  2020-12-28 16:41:24: MQTT publish: topic 'zigbee2mqtt/0x500b9140000108ee', payload '{"linkquality":49,"local_temperature":22.54}'
Zigbee2MQTT:info  2020-12-28 16:41:24: Configuring '0x500b9140000108ee'
Zigbee2MQTT:error 2020-12-28 16:41:25: Failed to configure '0x500b9140000108ee', attempt 3 (Error: ConfigureReporting 0x500b9140000108ee/1 hvacThermostat([{"attribute":"systemMode","minimumReportInterval":1,"maximumReportInterval":0,"reportableChange":null}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disab [... cut ...]
    at Endpoint.<anonymous> (/Users/mcc/homeassistant/zigbeemqtt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:396:26)
    at Generator.next (<anonymous>)
    at fulfilled (/Users/mcc/homeassistant/zigbeemqtt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:24:58))
^CZigbee2MQTT:info  2020-12-28 16:42:02: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'offline'
Zigbee2MQTT:info  2020-12-28 16:42:02: Disconnecting from MQTT server
Zigbee2MQTT:info  2020-12-28 16:42:03: zigbee-herdsman stopped
```

thank you!

3gyptian, 2020-12-29:
I forgot to include the zigbee-herdsman debug from the end of the logging:

```
zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":513,"srcaddr":45891,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":55,"securityuse":0,"timestamp":12381076,"transseqnumber":0,"len":4,"data":{"type":"Buffer","data":[24,5,7,0]}} +26ms
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":5,"manufacturerCode":null,"commandIdentifier":7},"Payload":[{"status":0}]},"address":45891,"endpoint":1,"linkquality":55,"groupID":0}' +56ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:controller:endpoint ConfigureReporting 0x500b9140000108ee/1 hvacThermostat([{"attribute":"systemMode","minimumReportInterval":1,"maximumReportInterval":0,"reportableChange":null}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) +57ms
  zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x500b9140000108ee:45891/1 (0,0,1) +57ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":45891,"destendpoint":1,"srcendpoint":1,"clusterid":513,"transid":5,"options":0,"radius":30,"len":11,"data":{"type":"Buffer","data":[16,6,6,0,28,0,48,1,0,0,0]}} +56ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,21,36,1,67,179,1,1,1,2,5,0,30,11,16,6,6,0,28,0,48,1,0,0,0,238] +56ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100] +14ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0} +60ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,5,195] +6ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,5,195] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,5] - 195 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":5} +22ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,27,68,129,0,0,1,2,67,179,1,1,0,52,0,43,236,188,0,0,7,24,6,7,140,0,28,0,67,179,9,21] +27ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,27,68,129,0,0,1,2,67,179,1,1,0,52,0,43,236,188,0,0,7,24,6,7,140,0,28,0,67,179,9,21] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 27 - 2 - 4 - 129 - [0,0,1,2,67,179,1,1,0,52,0,43,236,188,0,0,7,24,6,7,140,0,28,0,67,179,9] - 21 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":513,"srcaddr":45891,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":52,"securityuse":0,"timestamp":12381227,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[24,6,7,140,0,28,0]}} +27ms
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":6,"manufacturerCode":null,"commandIdentifier":7},"Payload":[{"status":140,"direction":0,"attrId":28}]},"address":45891,"endpoint":1,"linkquality":52,"groupID":0}' +49ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:controller:endpoint ConfigureReporting 0x500b9140000108ee/1 hvacThermostat([{"attribute":"systemMode","minimumReportInterval":1,"maximumReportInterval":0,"reportableChange":null}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Status 'UNREPORTABLE_ATTRIBUTE') +0ms
Zigbee2MQTT:error 2020-12-28 17:40:22: Failed to configure '0x500b9140000108ee', attempt 1 (Error: ConfigureReporting 0x500b9140000108ee/1 hvacThermostat([{"attribute":"systemMode","minimumReportInterval":1,"maximumReportInterval":0,"reportableChange":null}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Status 'UNREPORTABLE_ATTRIBUTE')
    at Endpoint.checkStatus (/Users/mcculloc/homeassistant/zigbeemqtt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:173:23)
    at Endpoint.<anonymous> (/Users/mcculloc/homeassistant/zigbeemqtt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:396:26)
    at Generator.next (<anonymous>)
    at fulfilled (/Users/mcculloc/homeassistant/zigbeemqtt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:24:58))
```

Please let me know if you need any further info ..

Koenkk, 2020-12-29:
Does updating the definition in devices.js to below solve it?

```js
{
    zigbeeModel: ['TH1123ZB'],
    model: 'TH1123ZB',
    vendor: 'Sinope',
    description: 'Zigbee line volt thermostat',
    fromZigbee: [fz.legacy.sinope_thermostat_att_report, fz.legacy.hvac_user_interface, fz.metering, fz.ignore_temperature_report,
        fz.legacy.sinope_thermostat_state],
    toZigbee: [tz.thermostat_local_temperature, tz.thermostat_occupied_heating_setpoint, tz.thermostat_unoccupied_heating_setpoint,
        tz.thermostat_temperature_display_mode, tz.thermostat_keypad_lockout, tz.thermostat_system_mode, tz.thermostat_running_state,
        tz.sinope_thermostat_occupancy, tz.sinope_thermostat_backlight_autodim_param, tz.sinope_thermostat_time,
        tz.sinope_thermostat_enable_outdoor_temperature, tz.sinope_thermostat_outdoor_temperature],
    exposes: [e.local_temperature(), e.keypad_lockout(), e.power(), e.energy(),
        exposes.climate().withSetpoint('occupied_heating_setpoint', 7, 30, 0.5).withLocalTemperature()
            .withSystemMode(['off', 'auto', 'heat']).withRunningState(['idle', 'heat'])],
    meta: {configureKey: 3},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        const binds = ['genBasic', 'genIdentify', 'genGroups', 'hvacThermostat', 'hvacUserInterfaceCfg', 'msTemperatureMeasurement',
            'seMetering'];

        await reporting.bind(endpoint, coordinatorEndpoint, binds);
        await reporting.thermostatTemperature(endpoint, {min: 10, max: 300, change: 20});
        await reporting.thermostatPIHeatingDemand(endpoint, {min: 10, max: 301, change: 5});
        await reporting.thermostatOccupiedHeatingSetpoint(endpoint, {min: 1, max: 302, change: 50});

        await reporting.readMeteringMultiplierDivisor(endpoint);

        try {
            await reporting.thermostatSystemMode(endpoint, {min: 1, max: 0});
            await reporting.thermostatKeypadLockMode(endpoint, {min: 1, max: 0});
            await reporting.instantaneousDemand(endpoint, {min: 10, max: 303, change: 1});
        } catch (error) {
            // Not all support this:
            // https://github.com/Koenkk/zigbee2mqtt/issues/3760
            // https://github.com/Koenkk/zigbee2mqtt/issues/5441
        }

        // Disable default reporting
        await reporting.temperature(endpoint, {min: 1, max: 0xFFFF});
        await endpoint.configureReporting('msTemperatureMeasurement', [{
            attribute: 'tolerance', minimumReportInterval: 1, maximumReportInterval: 0xFFFF, reportableChange: 1}]);
    },
},
```

3gyptian, 2020-12-29:
Hi - I tried but it's throwing an undefined for the first legacy property right at startup ..

```
$ npm start

> zigbee2mqtt@1.16.2 start /opt/zigbee2mqtt
> node index.js

/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices.js:11970
        fromZigbee: [fz.legacy.sinope_thermostat_att_report, fz.legacy.hvac_user_interface, fz.metering, fz.ignore_temperature_report,
                               ^

TypeError: Cannot read property 'sinope_thermostat_att_report' of undefined
    at Object.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices.js:11970:32)
    at Module._compile (internal/modules/cjs/loader.js:999:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1027:10)
    at Module.load (internal/modules/cjs/loader.js:863:32)
    at Function.Module._load (internal/modules/cjs/loader.js:708:14)
    at Module.require (internal/modules/cjs/loader.js:887:19)
    at require (internal/modules/cjs/helpers.js:74:18)
    at Object.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/index.js:3:17)
    at Module._compile (internal/modules/cjs/loader.js:999:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1027:10)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.16.2 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.16.2 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-12-29T20_49_53_831Z-debug.log
```

By the way I found things do show up in home assistant using the original devices.js and it seems I'm able to to control the thermostat.  Haven't got to far along with that yet though but either way I would like to clear up the unreportable attribute error.

thanks again for your help with this

Koenkk, 2020-12-30:
Forgot to mention you have to update to the latest dev branch first. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-02-06 by github-actions[bot]