Koenkk/zigbee2mqtt #5442: problem with UseeLink SM-AZ713 device after upgrade to 1.16.2.2 version

eljotpl, 2020-12-29:
Dears,

Im using supervisor version of Home Assistant based on docker and Debian as core.
I have installed 1.16.2.2 Zigbee2mqtt official addon.
I use conbee2 usb stick as coordinator.

After upgrade Zigbee2mqtt from 1.16.1 my UseeLink SM-AZ713 device is marked as unsupported (previously work fine):

WARNING: Device '0xbc33acfffe6e80f4' with Zigbee model 'TS011F' and manufacturer name '_TZ3000_tvuarksa' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

I saw that users have similar issues with TS011F devices after upgrade - and always manufacturer name is different.

Can I help me give info how to bring this device to life again ?

Koenkk, 2020-12-29:
How was it previously recognised? (what model) If you don't know can you provide a link the the product?

eljotpl, 2020-12-29:
Dear Koen,

I'm not 100% sure, but i think that was recognized as this device:

https://www.zigbee2mqtt.io/devices/TS011F_socket_module.html

Link to my product:

https://www.aliexpress.com/item/4000061156491.html?spm=a2g0s.9042311.0.0.216f4c4dECPNNq

PS. You are doing great job !

Koenkk, 2020-12-29:
Re-added the device

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html) (or wait for next z2m release on 1 january)

closed 2020-12-29 by Koenkk

eljotpl, 2020-12-29:
Dear Koen,

Many thanks for support !

I wish You everything best, and Happy New Year !