Koenkk/zigbee2mqtt #5459: Silvercrest switch (HG06337?) stopped to work on 1.16.2.2 - unsupported?

Seba200086, 2020-12-29:
## What happened
My Silvercrest (Lidl) switch stopped to work after update to 1.16.2.2

## What did you expect to happen
I would like to operate this device

## How to reproduce it (minimal and precise)
Update Zigbee2mqtt to 1.16.2.2

## Debug info
Zigbee2MQTT version: 1.16.2.2
Adapter hardware: CC2531
Adapter firmware version: 20190619 (tried also on default, 20190608)

I cannot operate the device, in log there is:
Zigbee2MQTT:info  2020-12-29 19:47:41: silvercrest_socket_1 (0xbc33acfffe47f0e0): Not supported (Router)

I tried to add it manually (according to https://www.zigbee2mqtt.io/devices/HG06337.html), but I do not see any new entities...

On zigbee2mqtt 1.16.1 everything is fine, device working (auto discovery) without any issue. Problem is on 1.16.2.2.
When I restoring to 1.16.1 all is fine again.

How to add and operate it on 1.16.2.2?

Koenkk, 2020-12-29:
Probably already fixed in the dev branch (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Seba200086, 2020-12-29:
Confirmed. Thank you so much for your work and best support!
Hope, that it will be fixed in next update. God bless you!

closed 2020-12-29 by Seba200086

marcgarciamarti, 2021-03-14:
@Koenkk sorry to post this on a closed issue but I'm at a loss. I have like 30 devices from mixed brands running just fine on Z2M, but I'm trying to pair a few HG06637s to my network and I can't. I've tested the current branch and the dev branch and the device is not recognised at all. Z2M acts if I wouldn't see the devices at all. I've made sure I'm initiating the process sitting very nearby the usb stick but it won't work either. Not a single trace in the log file.