Koenkk/zigbee2mqtt #5463: "manufId":0 for orvibo curtain 'AM68'

vapopov, 2020-12-29:
The Device seems to be paired incorrectly, so any configuration in devices.js are ignored

After pairing this device are stored in db *zigbee2mqtt/data/database.db* like that:
```
{"id":4,"type":"Router","ieeeAddr":"0x00124b0010026fa3","nwkAddr":8877,"manufId":0,"epList":[1],"endpoints":{"1":{"epId":1,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":true,"meta":{},"lastSeen":1609277569360}
```

Debug log in pairing time, `my_curtain` is an alias for the friendly name

```log
Dec 29 21:32:47 raspberrypi npm[4037]: Zigbee2MQTT:info  2020-12-29 21:32:47: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"friendly_name":"my_curtain","supported":false},"type":"pairing"}'
Dec 29 21:32:47 raspberrypi npm[4037]: Zigbee2MQTT:debug 2020-12-29 21:32:47: Device 'my_curtain' announced itself
Dec 29 21:32:47 raspberrypi npm[4037]: Zigbee2MQTT:info  2020-12-29 21:32:47: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"my_curtain","ieee_address":"0x00124b0010026fa3"},"type":"device_announce"}'
Dec 29 21:32:47 raspberrypi npm[4037]: Zigbee2MQTT:info  2020-12-29 21:32:47: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"my_curtain"},"type":"device_announced"}'
Dec 29 21:32:49 raspberrypi npm[4037]: Zigbee2MQTT:debug 2020-12-29 21:32:49: Received Zigbee message from 'my_curtain', type 'attributeReport', cluster 'genBasic', data '{"dateCode":"2017-05-19","swBuildId":"v2.2.9"}' from endpoint 1 with groupID 0
Dec 29 21:32:49 raspberrypi npm[4037]: Zigbee2MQTT:warn  2020-12-29 21:32:49: Received message from unsupported device with Zigbee model 'undefined' and manufacturer name 'undefined'
Dec 29 21:32:49 raspberrypi npm[4037]: Zigbee2MQTT:warn  2020-12-29 21:32:49: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Dec 29 21:32:49 raspberrypi npm[4037]: Zigbee2MQTT:debug 2020-12-29 21:32:49: Device 'my_curtain' announced itself
Dec 29 21:32:49 raspberrypi npm[4037]: Zigbee2MQTT:info  2020-12-29 21:32:49: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"my_curtain","ieee_address":"0x00124b0010026fa3"},"type":"device_announce"}'
Dec 29 21:32:49 raspberrypi npm[4037]: Zigbee2MQTT:info  2020-12-29 21:32:49: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"my_curtain"},"type":"device_announced"}'

Dec 29 21:33:44 raspberrypi npm[4037]: Zigbee2MQTT:debug 2020-12-29 21:33:44: Received Zigbee message from 'my_curtain', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":0}' from endpoint 1 with groupID 0
Dec 29 21:33:44 raspberrypi npm[4037]: Zigbee2MQTT:warn  2020-12-29 21:33:44: Received message from unsupported device with Zigbee model 'undefined' and manufacturer name 'undefined'
Dec 29 21:33:44 raspberrypi npm[4037]: Zigbee2MQTT:warn  2020-12-29 21:33:44: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Dec 29 21:34:44 raspberrypi npm[4037]: Zigbee2MQTT:debug 2020-12-29 21:34:44: Received Zigbee message from 'my_curtain', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":0}' from endpoint 1 with groupID 0
```

## What happened

The device all time marked as undefined
`Received message from unsupported device with Zigbee model 'undefined' and manufacturer name 'undefined'`

## What did you expect to happen

a working connection to a device, and recognized command from it

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: fresh from dev, take on time when this issue is created
Adapter hardware: CC2531

Thanks for any help!

vapopov, 2020-12-29:
after adding

```
        fingerprint: [
                {type: 'Router', manufacturerID: 0},
        ],
```
device become matched, but if there are more than one `unnamed` device, how I can match them by different parameters?

Koenkk, 2020-12-30:
Can you provide the herdsman debug log of pairing this device?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

vapopov, 2021-01-05:
@Koenkk thanks for support!

```log
Jan 05 14:03:48 raspberrypi npm[460]: 2021-01-05T14:03:48.969Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,12,69,202,237,165,163,111,2,16,0,75,18,0,0,0,76]
Jan 05 14:03:48 raspberrypi npm[460]: 2021-01-05T14:03:48.975Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,12,69,202,237,165,163,111,2,16,0,75,18,0,0,0,76]
Jan 05 14:03:48 raspberrypi npm[460]: 2021-01-05T14:03:48.976Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 12 - 2 - 5 - 202 - [237,165,163,111,2,16,0,75,18,0,0,0] - 76
Jan 05 14:03:48 raspberrypi npm[460]: 2021-01-05T14:03:48.978Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":42477,"extaddr":"0x00124b0010026fa3","parentaddr":0}
Jan 05 14:03:48 raspberrypi npm[460]: 2021-01-05T14:03:48.980Z zigbee-herdsman:controller:log Device '0x00124b0010026fa3' joined
Jan 05 14:03:48 raspberrypi npm[460]: 2021-01-05T14:03:48.982Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
Jan 05 14:03:48 raspberrypi npm[460]: 2021-01-05T14:03:48.983Z zigbee-herdsman:controller:log Device '0x00124b0010026fa3' accepted by handler
Jan 05 14:03:48 raspberrypi npm[460]: 2021-01-05T14:03:48.984Z zigbee-herdsman:controller:log New device '0x00124b0010026fa3' joined
Jan 05 14:03:49 raspberrypi npm[460]: 2021-01-05T14:03:48.999Z zigbee-herdsman:controller:log Creating device '0x00124b0010026fa3'
Jan 05 14:03:49 raspberrypi npm[460]: Zigbee2MQTT:info  2021-01-05 14:03:49: Device 'my_curtain' joined
Jan 05 14:03:49 raspberrypi npm[460]: Zigbee2MQTT:info  2021-01-05 14:03:49: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"my_curtain","ieee_address":"0x00124b0010026fa3"},"type":"device_joined"}'
Jan 05 14:03:49 raspberrypi npm[460]: 2021-01-05T14:03:49.066Z zigbee-herdsman:controller:log Interview '0x00124b0010026fa3' start
Jan 05 14:03:49 raspberrypi npm[460]: Zigbee2MQTT:info  2021-01-05 14:03:49: Starting interview of 'my_curtain'
Jan 05 14:03:49 raspberrypi npm[460]: Zigbee2MQTT:info  2021-01-05 14:03:49: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"my_curtain","ieee_address":"0x00124b0010026fa3","status":"started"},"type":"device_interview"}'
Jan 05 14:03:49 raspberrypi npm[460]: 2021-01-05T14:03:49.101Z zigbee-herdsman:controller:device:log Interview - start device '0x00124b0010026fa3'
Jan 05 14:03:49 raspberrypi npm[460]: 2021-01-05T14:03:49.178Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":42477,"nwkaddrofinterest":42477}
Jan 05 14:03:49 raspberrypi npm[460]: 2021-01-05T14:03:49.180Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,237,165,237,165,35]
Jan 05 14:03:49 raspberrypi npm[460]: Zigbee2MQTT:info  2021-01-05 14:03:49: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"my_curtain"},"type":"device_connected"}'
Jan 05 14:03:49 raspberrypi npm[460]: Zigbee2MQTT:info  2021-01-05 14:03:49: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"my_curtain"},"type":"pairing"}'
Jan 05 14:03:49 raspberrypi npm[460]: 2021-01-05T14:03:49.236Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,13,69,193,237,165,237,165,163,111,2,16,0,75,18,0,142,128]

Jan 05 14:03:49 raspberrypi npm[460]: 2021-01-05T14:03:49.255Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,13,69,193,237,165,237,165,163,111,2,16,0,75,18,0,142,128]
Jan 05 14:03:49 raspberrypi npm[460]: 2021-01-05T14:03:49.260Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 13 - 2 - 5 - 193 - [237,165,237,165,163,111,2,16,0,75,18,0,142] - 128
Jan 05 14:03:49 raspberrypi npm[460]: 2021-01-05T14:03:49.264Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - endDeviceAnnceInd - {"srcaddr":42477,"nwkaddr":42477,"ieeeaddr":"0x00124b0010026fa3","capabilities":142}
Jan 05 14:03:49 raspberrypi npm[460]: 2021-01-05T14:03:49.268Z zigbee-herdsman:controller:log Device announce '0x00124b0010026fa3'
Jan 05 14:03:49 raspberrypi npm[460]: Zigbee2MQTT:debug 2021-01-05 14:03:49: Device 'my_curtain' announced itself
Jan 05 14:03:49 raspberrypi npm[460]: Zigbee2MQTT:info  2021-01-05 14:03:49: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"my_curtain","ieee_address":"0x00124b0010026fa3"},"type":"device_announce"}'
Jan 05 14:03:49 raspberrypi npm[460]: 2021-01-05T14:03:49.296Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
Jan 05 14:03:49 raspberrypi npm[460]: Zigbee2MQTT:info  2021-01-05 14:03:49: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"my_curtain"},"type":"device_announced"}'
Jan 05 14:03:49 raspberrypi npm[460]: 2021-01-05T14:03:49.631Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102]

Jan 05 14:03:49 raspberrypi npm[460]: 2021-01-05T14:03:49.638Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102]
Jan 05 14:03:49 raspberrypi npm[460]: 2021-01-05T14:03:49.641Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102
Jan 05 14:03:49 raspberrypi npm[460]: 2021-01-05T14:03:49.644Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0}
Jan 05 14:03:49 raspberrypi npm[460]: 2021-01-05T14:03:49.648Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
[... cut ...]
Jan 05 14:03:53 raspberrypi npm[460]: Zigbee2MQTT:info  2021-01-05 14:03:53: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"my_curtain","ieee_address":"0x00124b0010026fa3"},"type":"device_announce"}'
Jan 05 14:03:53 raspberrypi npm[460]: 2021-01-05T14:03:53.689Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
Jan 05 14:03:53 raspberrypi npm[460]: Zigbee2MQTT:info  2021-01-05 14:03:53: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"my_curtain"},"type":"device_announced"}'
```

vapopov, 2021-01-05:
@Koenkk only with this

```
        fingerprint: [
                {type: 'Router', manufacturerID: 0},
        ],
```

vapopov, 2021-01-05:
In database.db btw now it is mentioned like this

```
{"id":4,"type":"Router","ieeeAddr":"0x00124b0010026fa3","nwkAddr":42477,"manufId":0,"epList":[1],"endpoints":{"1":{"epId":1,"inClusterList":[],"outClusterList":[],"clusters":{"closuresWindowCovering":{"attributes":{"currentPositionLiftPercentage":100}},"genBasic":{"attributes":{"dateCode":"2017-05-19","swBuildId":"v2.2.9"}}},"binds":[],"configuredReportings":[],"meta":{}}},"dateCode":"2017-05-19","swBuildId":"v2.2.9","interviewCompleted":true,"meta":{},"lastSeen":1609875027222}
```

Koenkk, 2021-01-06:
The problem is that the firmware of this device is buggy, when interviewing it advertises it has no endpoint (while it has endpoint 1). Manually changing your entry to below should fix it:

```
{"id":4,"type":"Router","ieeeAddr":"0x00124b0010026fa3","modelId":"093199ff04984948b4c78167c8e7f47e","nwkAddr":42477,"manufId":0,"epList":[1],"endpoints":{"1":{"epId":1,"inClusterList":[],"outClusterList":[],"clusters":{"closuresWindowCovering":{"attributes":{"currentPositionLiftPercentage":100}},"genBasic":{"attributes":{"dateCode":"2017-05-19","swBuildId":"v2.2.9"}}},"binds":[],"configuredReportings":[],"meta":{}}},"dateCode":"2017-05-19","swBuildId":"v2.2.9","interviewCompleted":true,"meta":{},"lastSeen":1609875027222}
```

(I've added `"modelId":"093199ff04984948b4c78167c8e7f47e",`)

closed 2021-01-06 by vapopov