Koenkk/zigbee2mqtt #5468: Reset coordinator with the frontend version 0.2.125 ?

zibous, 2020-12-30:
## What happened
I had a router (OSRAM Smart+ plug) failure today and the devices that were connected are offline.

[image]

## Debug info
Zigbee2MQTT version: 1.16.2-dev commit: 897f5a9
Adapter hardware: ConBee2/RaspBee2
Adapter firmware version: 26660700

1. What does "Reset coordinator" do with the frontend version 0.2.125? 
    **Will all devices be reset and do I have to reconnect them?**

2. Reconnect devices
I had a router failure today and the devices that were connected to it are no longer accessible or I am no longer receiving measurement data. 
**Do I have to reconnect the orphaned devices to another router or to the coordinator?**

Koenkk, 2020-12-30:
> What does "Reset coordinator" do with the frontend version 0.2.125?

@nurikk can you remove this reset option from the frontend? It's not working correctly and also not included in the new api. Seems to only cause confusion.

> I had a router failure today and the devices that were connected to it are no longer accessible or I am no longer receiving measurement data.

First try re-powering them/re-insert batteries. If that doesn't help repair.

nurikk, 2020-12-30:
I've added `config/reset` call for @Hacker-CB, if it not required anymore I can remove it from ui

Hacker-CB, 2020-12-30:
@nurikk I don't remeber I asked for this. 
I don't know what this option do. I never used it.
May be somebody else asked for this?

Hacker-CB, 2020-12-30:
Is this discussion is about "Reset Coordinator" option? Or other one?

timoline, 2020-12-30:
I think it comes from this discussion?
https://github.com/nurikk/z2m-frontend/issues/202

zibous, 2020-12-30:
this one: frontend version 0.2.125

[image]

nurikk, 2020-12-30:
> this one: frontend version 0.2.125
> 
> [image]

I have removed this feature in the latest update

zibous, 2021-01-01:
removed see: frontend version 0.2.129

closed 2021-01-01 by zibous