Koenkk/zigbee2mqtt #5473: Make report->poll smarter now we store reporting date

sjorge, 2020-12-30:
## Is your feature request related to a problem? Please describe
n/a

## Describe the solution you'd like
I think we clean this https://github.com/Koenkk/zigbee2mqtt/blob/d1842ab69ae62a8a59a0de1b0caa86f73d614763/lib/extension/report.js#L243 up to make it smarter... but been messing with it and can't come up with anything better.

I was trying to make it only poll when the actions was against a device/group where the device being controlled was a bulb/switch and did not have a reporting entry for genOnOff/genLevelCtrl/lighthingColorCtrl... but struggling with that.

Mostly because the mapping with the above list is confusing me I think.

## Describe alternatives you've considered
Leaving it as is, but having it not hardcoded to a bunch of known manufacturers is probably better.
As we had to add a few in the past because e.g. hue changed ID and such.

## Additional context
Add any other context or screenshots about the feature request here.

Koenkk, 2020-12-31:
I propose to wait for this as I'm currently planning to do some rework on this. I'm planning to deprecate the report feature. The `poll` code will be moved to `bind.js` (since it fits better there). Additionally bind.js will automatically setup the reportings when binding. This will make the report feature obsolete since it was mainly used for getting correct states of devices having bindings to and will have the benefit of reduced messages on the network (since only required reportings are being setup).

sjorge, 2020-12-31:
Sounds good, I'll close this.

closed 2020-12-31 by sjorge