Koenkk/zigbee2mqtt #5477: Ikea E1743 needs to be rebound after swapping batteries

sjorge, 2020-12-31:
## What happened
I had a fresh opened battery I used to test my new multi-meter. In stead of leaving it out of the pack somewhere, I grabbed my remote which was sitting at 9% battery and swapped it.

I drop the battery and had to search a for it, after the swap was complete I noticed the remote didn't work.
Well it did, it announced itself and was sending action just fine. It was no longer controlling the group I had it bound to.

After rebinding it worked OK.

## What did you expect to happen
Remote to keep working, or at least z2m to show that it was back on the default bindings.

## How to reproduce it (minimal and precise)
Take out the battery, wait for 30-60sec, insert the battery.

## Debug info
Zigbee2MQTT version: 1.16.2-dev commit: 92f3caa
Adapter hardware: CC2652P (zzhp-lite)
Adapter firmware version:  20201113

sjorge, 2020-12-31:
I noticed later that I still had Joining mode enabled, I don't think that would be an issue though.

MattWestb, 2020-12-31:
Sankta was coming with one multi-meter this year :-))))
Congratulations and welcome to the hardware team !!!

Its more or less well known that IKEA end device can eating batteries very fast but not on normal basis and if they is going completely down its very likely need being repaired.
Its also happenings on IKEA GW so its not generelle for "other" systems.
The latest EZSP release notes from Silabs have fixed some bugs that doing end devices not going down in power save mode and cant disabling the external flash (is used for OTA updates only).
Very likely its also making that the stack cant writing the security frame counter before its dying and also corrupting the NVR with the network parameters resulting in losing the network.
I can confirming that both draining battery and must repairing it happens on IKEA GW and on ZHA with EZSP coordinator. Other users on ZHA have getting it on Rasp/CornBee and TI coordinators.
Some of my IKEA controlling devices is still on original batteries and is online for over 2 years now and other is lasting some months but also having good router ner them is making they dont need using so much power for communicating and is making the batteries living longer.

PS: Also "high end brands" batteries can being old or bad batches but normally its commun with 3 for 2€ ones that the voltages is under 3V then taking them out of the package. One good new ones should having over 3.3V and under 2.8V it cant supplying the zigbee chip with enough power for transmitting and writing / updating the NVR parameters.

sjorge, 2020-12-31:
The remote was fine still before removing the battery and putting in a fresh one.
It happens consistently though, it was still sending the "action"... so maybe only binding/reporting is lost.

If that is indeed the case, we can probably like with the bulb, resetup them on deviceAnnounce.

Koenkk, 2020-12-31:
> If that is indeed the case, we can probably like with the bulb, resetup them on deviceAnnounce.

Sounds like a good solution, can you check it?

MattWestb, 2020-12-31:
I think its not so common that devices is losing network on battery changes but its still happening.
The question is it worth always reconfing on device annunce ? Then the device is draining the battery faster if its losing contact and need rejoining the network. ANd is not doing it fast its going in to sleep and the reconfing is failing = worse then leaving it untouched.

One Q ?
What was the voltage of your old battery then it was 9% left ?
Im curious how deep the chip can working OK.
I have some E1743 that was working under 10% but was föashing the LED then being operated and then i was replacing the battery and it was working OK.

sjorge, 2020-12-31:
@mattwestb I had a few working (sometimes needing one or two presses) around 1%, swapped the battery and it was still fine.

@Koenkk 
> Sounds like a good solution, can you check if this would work?

I manually rebound and re-applied reporting via the frontend, will wait a day to see if it has updated the battery %, it is still showing the old one. 

I remember it happening in the past too where even with a fresh battery it was not reporting the new battery % but I was still able to control the group... so it might just have different levels of what it forgets when the battery is removed for too long.

Once I know it's OK, I'll start experimenting with removing, waiting and seeing what work/doesn't works and go from there.

sjorge, 2021-01-01:
Sadly the battery has not been update, so I don't think reconfiguring reporting is going to work.
I will try to redo the genPowerCfg binding also, if that doesn't work I will do a full repair.

Either way, I'll do a PR and or I will update the docs for the device.

MattWestb, 2021-01-01:
Great to knowing that 2.7 = red blinking and double press = nearly dead.
HARDWARE HACKER !!!  :-)))

Then doing one new binding you need redoing the attribute reporting or the attribute reporting is broken and that is by zigbee standard.

Have doing some intense test with E1743 on ZHA with the reporting and binding last days because one remote was not checking in and reporting. Rebinding and configure attribute reporting looks doing the thing.

sjorge, 2021-01-01:
I guess we're hitting 2 bugs then, for some reason I can't unbind genPowerCfg to rebind it.

Ofcourse can't test any more now though as I just rejoined the device.

@Koenkk OK to add an onEvent handler like this:
<snip> just going to open a PR, thats faster and I have wrote the code anyway.

sjorge, 2021-01-01:
https://github.com/Koenkk/zigbee-herdsman-converters/pull/2006

sjorge, 2021-01-05:
I had another one that was around 28% last week, drop to 16% during the weekend.

Today it stopped responding and looks like it died during the night (as I used it before bed).

- put in a new cell
- click once on a button
<nothing happens>
- click once on the back button (paring?)
<device annoucement>
- click on, off, hold... now work again but the lights don't
- hit bind on the via the frontend
<lights work>

Matt is probably on to something, that the longer they are without power the more they forget?
As taking about the battery for ~1 minute doesn't seem to cause it to forget the group it is bound too, neither did ~5 min.

Do we want to redo the bindings too on device announcement?

MattWestb, 2021-01-05:
They is two scenarios that ZHA devs have doing both is connected to Silabs zigbee stack.
It was (fixed with the last release 2020.12 so no devices have getting the fixes yet) that the device cant going down in power saving mode in some circumstances and the other it having problem shutting down the flash and burning batteries (was the ground digging after bugs).

The theory of the repairing issue is that that the security framecunter is not written OK in the device before its dying and cant communicating OK until its being reseted and the other is that its going more bad and then writing to the NVR (its one part of the flash) its corrupting it then cant writing / deleting it OK and the devices is doing one reset of its network parameters to factory new = losing all config / settings.

If its possible checking the groupe binding and is OK on device announcement (old ZLL devises is making one random LL groupe on reset ZB3 dont and defaulting to 0X0000) = dont do any thing and if its lost i think its needed one complete initiation and also the customized groupe binding and attribute reporting.

closed 2021-02-13 by github-actions[bot]