Koenkk/zigbee2mqtt #5478: issue with ikea e1810

jerome70, 2020-12-31:
Hello
i am facing a issue with my last ikea E1810 for paring

# Bug Report
## What happened
something goes wrong :
info  2020-12-31 10:19:52: Device '0xbc33acfffe84e099' joined                                                                                                                                                  info  2020-12-31 10:19:52: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xbc33acfffe84e099","ieee_address":"0xbc33acfffe84e099"},"type":"device_joined"}'
info  2020-12-31 10:19:52: Starting interview of '0xbc33acfffe84e099'
info  2020-12-31 10:19:52: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xbc33acfffe84e099","ieee_address":"0xbc33acfffe84e099","status":"started"},"type":"device_inter$info  2020-12-31 10:19:52: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0xbc33acfffe84e099"},"type":"device_connected"}'
info  2020-12-31 10:19:52: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0xbc33acfffe84e099"},"type":"pairing"}'                              debug 2020-12-31 10:19:52: Device '0xbc33acfffe84e099' announced itself                                                                                                                                        info  2020-12-31 10:19:52: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xbc33acfffe84e099","ieee_address":"0xbc33acfffe84e099"},"type":"device_announce"}'              info  2020-12-31 10:19:52: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0xbc33acfffe84e099"},"type":"device_announced"}'                              debug 2020-12-31 10:20:08: Device '0xbc33acfffe84e099' announced itself                                                                                                                                        info  2020-12-31 10:20:08: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xbc33acfffe84e099","ieee_address":"0xbc33acfffe84e099"},"type":"device_announce"}'
info  2020-12-31 10:20:08: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0xbc33acfffe84e099"},"type":"device_announced"}'                              debug 2020-12-31 10:20:23: Received Zigbee message from 'routeur_3', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 8 with groupID 0                                             info  2020-12-31 10:20:23: MQTT publish: topic 'zigbee2mqtt/routeur_3', payload '{"led":false,"led_state":true,"linkquality":70,"state":true}'                                                                 error 2020-12-31 10:20:27: Failed to interview '0xbc33acfffe84e099', device has not successfully been paired   
warn  2020-12-31 10:22:11: Received message from unsupported device with Zigbee model 'undefined' and manufacturer name 'undefined'
warn  2020-12-31 10:22:11: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.              

{"id":45,"type":"EndDevice","ieeeAddr":"0xbc33acfffe84e099","nwkAddr":25701,"manufId":4476,"epList":[1],"endpoints":{"1":{"epId":1,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],    

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2MQTT version: 1.16.2-dev (commit #89272ce)
Adapter hardware: CC2531
Adapter firmware version: last source

Koenkk, 2020-12-31:
Looks like the device did not pair correctly. Try repairing it while holding it close to the coordinator.

jerome70, 2021-01-01:
Thanks so much for your answer,
i dit it just now , but same result .

my log:
nfo  2021-01-01 18:15:49: Device '0xbc33acfffe84e099' joined
info  2021-01-01 18:15:49: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xbc33acfffe84e099","ieee_address":"0xbc33acfffe84e099"},"type":"device_joined"}'
info  2021-01-01 18:15:49: Starting interview of '0xbc33acfffe84e099'
info  2021-01-01 18:15:49: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xbc33acfffe84e099","ieee_address":"0xbc33acfffe84e099","status":"started"},"type":"device_inter$info  2021-01-01 18:15:49: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0xbc33acfffe84e099"},"type":"device_connected"}'
info  2021-01-01 18:15:49: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0xbc33acfffe84e099"},"type":"pairing"}'                              debug 2021-01-01 18:15:49: Device '0xbc33acfffe84e099' announced itself                                                                                                                                        info  2021-01-01 18:15:49: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xbc33acfffe84e099","ieee_address":"0xbc33acfffe84e099"},"type":"device_announce"}'
debug 2021-01-01 18:15:49: Received Zigbee message from '0xbc33acfffe84e099', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
debug 2021-01-01 18:15:49: Skipping message, definition is undefined and still interviewing
info  2021-01-01 18:15:49: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0xbc33acfffe84e099"},"type":"device_announced"}'
debug 2021-01-01 18:15:49: Received Zigbee message from '0xbc33acfffe84e099', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
debug 2021-01-01 18:15:49: Skipping message, definition is undefined and still interviewing
debug 2021-01-01 18:15:49: Received Zigbee message from '0xbc33acfffe84e099', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
debug 2021-01-01 18:15:49: Skipping message, definition is undefined and still interviewing
debug 2021-01-01 18:15:50: Received Zigbee message from '0xbc33acfffe84e099', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
debug 2021-01-01 18:15:50: Skipping message, definition is undefined and still interviewing

and also 
debug 2021-01-01 18:18:34: Received Zigbee message from '0xbc33acfffe84e099', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
debug 2021-01-01 18:18:34: Skipping message, definition is undefined and still interviewing
debug 2021-01-01 18:18:38: Received Zigbee message from '0xbc33acfffe84e099', type 'readResponse', cluster 'genBasic', data '{"swBuildId":"2.3.014"}' from endpoint 1 with groupID 0
debug 2021-01-01 18:18:38: Skipping message, definition is undefined and still interviewing
info  2021-01-01 18:18:38: Successfully interviewed '0xbc33acfffe84e099', device has successfully been paired                                                                                                  warn  2021-01-01 18:18:38: Device '0xbc33acfffe84e099' with Zigbee model 'undefined' and manufacturer name 'IKEA of Sweden' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_$info  2021-01-01 18:18:38: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0xbc33acfffe84e099","ieee_address":"0xbc33acfffe84e099","status":"successful",$info  2021-01-01 18:18:39: MQTT publish: topic 'zigbee2mqtt/bridge/devices', payload '[{"date_code":"2018-09-26","definition":{"description":"[CC2530 router](http://ptvo.info/cc2530-based-zigbee-coordinator$info  2021-01-01 18:18:39: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"friendly_name":"0xbc33acfffe84e099","supported":false},"type":"pairing"}'
debug 2021-01-01 18:18:41: Received Zigbee message from 'routeur_3', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 8 with groupID 0
info  2021-01-01 18:18:41: MQTT publish: topic 'zigbee2mqtt/routeur_3', payload '{"led":true,"led_state":true,"linkquality":89,"state":true}'
debug 2021-01-01 18:18:43: Received Zigbee message from '0xbc33acfffe84e099', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
warn  2021-01-01 18:18:43: Received message from unsupported device with Zigbee model 'undefined' and manufacturer name 'IKEA of Sweden'
warn  2021-01-01 18:18:43: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
debug 2021-01-01 18:18:45: Received Zigbee message from '0xbc33acfffe84e099', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0                                               warn  2021-01-01 18:18:45: Received message from unsupported device with Zigbee model 'undefined' and manufacturer name 'IKEA of Sweden'                                                                       warn  2021-01-01 18:18:45: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.                                                                                                     debug 2021-01-01 18:18:47: Received Zi [... cut ...] pported device with Zigbee model 'undefined' and manufacturer name 'IKEA of Sweden'                                                                       warn  2021-01-01 18:18:47: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.                                                                                                     debug 2021-01-01 18:19:00: Saving state to file /opt/zigbee2mqtt/data/state.json      

tried resetting pi an remote , what can i do else ?

Koenkk, 2021-01-01:
An alternative way to fix it would be to set the modelId manually. Open the data/database.db, go to the line of the remote (one with `ieeeAddr` being `0xbc33acfffe84e099` and add `"modelId":"TRADFRI remote control,"` here. (right after `ieeeAddr`)

jerome70, 2021-01-02:
dear Koenkk

i tried this 

my database once your modification done and restarting
{"id":45,"type":"EndDevice","ieeeAddr":"0xbc33acfffe84e099","nwkAddr":51625,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Battery","modelId":"TRADFRI remote control,","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":2080,"inClusterList":[0,1,3,32,4096,64636],"outClusterList":[3,4,6,8,25,4096],"clusters":{"genBasic":{"attributes":{"manufacturerName":"IKEA of Sweden","powerSource":3}}},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":true,"meta":{},"lastSeen":1609585714759}

and my log:
warn  2021-01-02 12:16:01: Received message from unsupported device with Zigbee model 'TRADFRI remote control,' and manufacturer name 'IKEA of Sweden'
warn  2021-01-02 12:16:01: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.                                                                                                     debug 2021-01-02 12:16:02: Received Zigbee message from '0xbc33acfffe84e099', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
warn  2021-01-02 12:16:02: Received message from unsupported device with Zigbee model 'TRADFRI remote control,' and manufacturer name 'IKEA of Sweden'
warn  2021-01-02 12:16:02: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.

Koenkk, 2021-01-02:
Ah sorry, made a typo remove the trailing `,` so `TRADFRI remote control` and not `TRADFRI remote control,`

jerome70, 2021-01-03:
@Koenkk 
thanks for your message.
What i did : change battery ( for test) put the remote next the coordinator and everything works !
Do you think the battery level is an explaination ?
Thank you for your help and the amazing work on this project !

Koenkk, 2021-01-03:
@jerome70 yes, battery level is very important indeed.

closed 2021-01-03 by jerome70