Koenkk/zigbee2mqtt #5489: Incomplete implementation of IAS ACE causes multiple "keyfob" devices to send duplicate button events

olix74, 2021-01-01:
## What happened
I'm seeing multiple retries/messages of button events on multiple security remote controller keyfobs (at least Heiman HS1RC and Immax 07046L). Heiman semms to be sending four repeats every ~7s before giving up, Immax sending two events ~5s. Both seem to indicate a failed transaction with a red led afterwards. Regrettably each repeat comes with a seperate transaction number, so they can not be filtered out easiliy/ it is not possible to distinguish two consecutive button presses from the repeat.
 
## Why is this bad?
Some applications require one touch actions - first press -> turn on, second press -> turn off. These can not be implemented. Appart from that batteries are drained much faster due to repeats.

## What did you expect to happen
Avoid retransmitions. Transaction successfull indication on keyfob with green led (instead of transaction failed)

## How to reproduce it (minimal and precise)
Happens on every single button press on the Heiman remote and happens for all buttons except "SOS" on Immax remote.

## Debug info
Zigbee2MQTT version: regrettably I'm using Iobroker.adapter 1.4 as of Dec. 2020 which is using zigbee-herdsman library, but I'm near 100% sure that Zigbee2MQTT shows same behavior and since main development is happening here I appologize for opening reuqest here.
Adapter hardware: Conbee II
Adapter firmware version: 26680700 as of 30th of November 2020 /should be latest=

## Analysis and Proposal
I'm also really no expert in Zigbee. Nevertheless I had a look at https://zigbeealliance.org/wp-content/uploads/2019/12/07-5123-06-zigbee-cluster-library-specification.pdf which I assume to be part of the relevant zigbee spec.
As both devices are IAS ACE devices I'm xpecting that they require a responses from a security system, so it would be fine to "pretend" to be such a security system.
Here is my proposal for that after reading through the spec:

8.3 specifies IAS ACE,
8.3.2.4 the Commands Received (from Heiman and Immax remote )and
8.3.2.5 the expected/proposed responses to be generated (currently missing).

Proposal 1:
I assume that all buttons except SOS generate an "Arm Command" according to 8.3.2.4.1
As such they probably expect a response according to 8.3.2.4.1.5
Here it says "The device SHALL generate an "Arm Response command" (see 8.3.2.5.1)to indicate the resulting armed state."
8.3.2.5.1 explains this response.
Respone should be two bytes and is explained below:
Command ID --> 0x00 Arm Response
followed by following "Arm Notification" (enum8) content according to following table
- 0x01 Only Day/Home Zones Armed
- 0x00 All Zones Disarmed
- 0x03 All Zones Armed

Proposal 2:
In addition it MIGHT be necessary to send "Panel Status Changed Command" ID=0x04 according to 8.3.2.5.5
I would assume following responses

Panel Status: 0x01 Armed Stay, Seconds Remaining: 0x00, Audible Notification: 0x01 (default), Alarm Status: 0x00 (no Alarm)
Panel Status: 0x00 Panel disarmed, Seconds Remaining: 0x00, Audible Notification: 0x01 (default), Alarm Status: 0x00 (no Alarm)
Panel Status: 0x03 Armed away, Seconds Remaining: 0x00, Audible Notification: 0x01 (default), Alarm Status: 0x00 (no Alarm)
Nevertheless one could argue that this should only be triggered by a real IAS alarm system connected to zigbee2mqtt instead of being riggered directly by zigbee2mqtt.

Proposal 3:
Regrettably 8.3.2.4.3 does not explain the response to emergency button 3002.
I would propose to also send a "Panel Status Changed Command" ID=0x04 according to 8.3.2.5.5
Panel Status: 0x07 In alarm, Seconds Remaining: 0x00, Audible Notification: 0x01 (default), Alarm Status: 0x03 (Emergency)

Thank you, very much.
Oliver

Koenkk, 2021-01-01:
Thanks for the good investigation already. I don't know exactly which HS1RC you have, but here is an example of doing the arm response (proposal 1) for a `HS1RC-N`. You have to update the definition of your device in devices.js, the relevant part is in the `onEvent` method. See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html on how to modify devices.js

```js
{
    fingerprint: [{modelID: 'RC-N', manufacturerName: 'HEIMAN'}],
    model: 'HS1RC-N',
    vendor: 'HEIMAN',
    description: 'Smart remote controller',
    fromZigbee: [fz.battery, fz.legacy.heiman_smart_controller_armmode, fz.command_emergency],
    toZigbee: [],
    exposes: [e.battery(), e.action(['emergency', 'disarm', 'arm_partial_zones', 'arm_all_zones'])],
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
        await reporting.batteryPercentageRemaining(endpoint, {min: repInterval.MINUTES_5, max: repInterval.HOUR});
        await endpoint.read('genPowerCfg', ['batteryPercentageRemaining']);
    },
    onEvent: async (type, data, device) => {
        if (type === 'message' && data.type === 'commandArm' && data.cluster === 'ssIasAce') {
            console.log('\n\nResponding to command arm!\n\n');
            await device.getEndpoint(1).commandResponse(
                'ssIasAce', 'armRsp', {armnotification: 0}, {}, data.meta.zclTransactionSequenceNumber,
            );
        }
    },
},
```

olix74, 2021-01-01:
**Hi, 
Thank you for the fast response, a Great 2021 for you and thank you very much for all the effort you put into your zigbee implementation already!** 
Since I'm currently having issues with my Heiman remote (I got the EM version) I tried to "transfer" the above example to the Immax remote, but that was kind of "guessing" and so I failed in a way that the if clause is never true/reached. Do you see any obvious issue in the following code for Immax in devices.js or have any other proposals? Again I have to admit that I'm using the iobroker interface to zigbee-herdsman but I hope that shouldn't be a general issue.

Thank you, Oliver
```
    {
        zigbeeModel: ['Keyfob-ZB3.0'],
        model: '07046L',
        vendor: 'Immax',
        description: '4-Touch single click buttons',
        fromZigbee: [fz.legacy.immax_07046L_arm, fz.command_panic],
        exposes: [e.action(['disarm', 'arm_stay', 'arm_away', 'panic'])],
        toZigbee: [],
// added code snipet starting here
        meta: {configureKey: 1}, //not sure what this is for? is it needed? It didn't work with or without
        configure: async (device, coordinatorEndpoint, logger) => {
//          const endpoint = device.getEndpoint(1); //is this needed?
//          await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
//          await reporting.batteryPercentageRemaining(endpoint, {min: repInterval.MINUTES_5, max: repInterval.HOUR});
//          await endpoint.read('genPowerCfg', ['batteryPercentageRemaining']);
        },
        onEvent: async (type, data, device) => {
          if (type === 'message' && data.type === 'commandArm' && data.cluster === 'ssIasAce') {
//              meta.logger.debug('\n\nResponding to command arm!\n\n');
              await device.getEndpoint(1).commandResponse(
                'ssIasAce', 'armRsp', {armnotification: 0}, {}, data.meta.zclTransactionSequenceNumber,
              );
          }
      },
    },

```

Koenkk, 2021-01-01:
@kirovilya does iobroker call the onEvent method? What would be the best way to debug this from iobroker?

olix74, 2021-01-02:
Hello @Koenkk, I raised the question on how to debug iobroker in the iobroker forum, since a "node inspect" fails with "Starting inspector on 127.0.0.1:9229 failed: address already in use" and I can't see any console.log() either. Maybe I can find out if onEvent is called by doing a code review, but of course I would appreciate an answer from the experts ;-)
Best regards,
Oliver

asgothian, 2021-01-02:
> @kirovilya does iobroker call the onEvent method? What would be the best way to debug this from iobroker?

- The onEvent is called when a zigbee Event is received.
- The point to debug this would be from lib/zigbeecontroller.js where iobroker logging functions would be available, as well as access to js variables if the adapter is started in a debug context.

closed 2021-02-10 by github-actions[bot]

Michelobb, 2023-01-01:
Hi,

Not a programmer here, so I had some hard time to get this work properly.

Maybe this could help others, here's what I had to do :

1. create a blank new file in config/zigbee2mqtt, next to configuration.yaml.  Call it as you want, for example short name of your device.  Inmy case, it was HS1RC.js (Heimann Remote)

2. Paste the code below.  Codes here above are missing constant declaration (or maybe I didn't get this point, sorry, as I told, not a programmer).

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

module.exports = [
    {
        fingerprint: [{modelID: 'RC-EM', manufacturerName: 'HEIMAN'}],
        model: 'HS1RC-EM',
        vendor: 'HEIMAN',
        description: 'Smart remote controller',
        fromZigbee: [fz.battery, fz.command_arm, fz.command_emergency],
        toZigbee: [],
        exposes: [e.battery(), e.action(['emergency', 'disarm', 'arm_partial_zones', 'arm_all_zones'])],
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint, logger) => {
                const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
            await reporting.batteryPercentageRemaining(endpoint, {min: constants.repInterval.MINUTES_5, max: constants.repInterval.HOUR});
            await endpoint.read('genPowerCfg', ['batteryPercentageRemaining']);
        },
        onEvent: async (type, data, device) => {
            // Since arm command has a response zigbee-herdsman doesn't send a default response.
            // This causes the remote to repeat the arm command, so send a default response here.
            if (type === 'message' && data.type === 'commandArm' && data.cluster === 'ssIasAce') {
                await data.endpoint.defaultResponse(0, 0, 1281, data.meta.zclTransactionSequenceNumber);
            }
        },
    },
];
```

3. into configuration.yaml, add this :

```
external_converters:
  - HS1RC.js

```

And eventually this : 

```
advanced:
  log_level: debug

```

Restart Z2M.

Have fun (oh my God).

Michel