Koenkk/zigbee2mqtt #5494: Home Assistant MQTT discovery isn't working

biscuit-316, 2021-01-01:
## What happened
When I add a device to zigbee2mqtt, the device is not automatically added to Home Assistant.

## What did you expect to happen
I expected that after a reboot the light would appear in Home Assistant.

## How to reproduce it (minimal and precise)
When I add my first device to zigbee2mqtt it shows up in the web interface. It can be operated from there. But the device is not automatically added to Home Assistant.

When I use the ["Manual Home Assistant configuration"](https://www.zigbee2mqtt.io/devices/LED1733G7.html) the light appears in Home Assistant. And is fully controllable.

I had added the [MQTT Discovery](https://www.zigbee2mqtt.io/integration/home_assistant.html) details, although they are no longer needed. (MQTT Discovery is enabled by default).

The addon configuration contains: "homeassistant: true".

## Debug info
zigbee2mqtt addon in Home Assistant (version: 1.16.2.2)
Adapter hardware: CC2531
Mosquitto broker addon in Home Assistant (version: 5.1)
Home Assistant (version: 2020.12.2)
Home Assistant OS 5.9

IlmLV, 2021-01-01:
same issue for me

Home Assistant 2020.12.1 (docker)
Zigbee2MQTT version 1.17.0 (commit #07cdc9d)
Mosquitto version 1.6.12
Adapter: Conbee2

Koenkk, 2021-01-02:
When subscribing to `homeassistant/#` using a MQTT client and starting z2m, do you see any messages?

biscuit-316, 2021-01-02:
When I started listening for `homeassistant/#` and restart zigbee2mqtt. [7 messages](https://pastebin.com/iYUJg7GS) came in. so mqtt works, what's next?

Koenkk, 2021-01-02:
Those are the discovery messages, so looks like the problem is on the HA side.

biscuit-316, 2021-01-02:
okay, then i will raise a issue with home assistant.

Thanks for your quick responses.

closed 2021-01-02 by biscuit-316