Koenkk/zigbee2mqtt #5499: Xiaomi WSDCGQ11LM excessive battery usage

embcla, 2021-01-01:
## What happened
Installed zigbee2mqtt in HA, in 24h the battery of my Xiaomi WSDCGQ11LM went from 100% to 74%

## What did you expect to happen
I have another 7 such sensors hooked to a phoscon addon on another HA instance for 3 months, all of them above 80%.
## How to reproduce it (minimal and precise)
I'm not sure, I suspect there is a higher polling rate that's draining the battery.
## Debug info
Zigbee2MQTT version: 1.16.2.2
Adapter hardware: Conbee II
Adapter firmware version: 26660700

Boia11, 2021-01-01:
Hi, I had this problem with a RTCGQ11LM ... the battery drain was caused by loop of sending occupancy.
I solve with removing and replacing the battery and re-pairing device (removing device)

Koenkk, 2021-01-01:
Unlikely this is due to a z2m issue.

> I suspect there is a higher polling rate that's draining the battery.

This sensor is never polled (it doesn't even support it), the sensor reports when necessary.

embcla, 2021-01-01:
Interesting point

As all the others exhibit exactly the same behaviour, I will try this one sensor on the other instance.
If it's the sensor, it will keep doing the same.

closed 2021-02-09 by github-actions[bot]

pedroke, 2022-03-01:
I noticed exactly the same problem with my WSDCGQ11LM. In the past it reported much less frequently. Now it reports every change of temperature or humidity in 0.1C steps.. In the past it was I think in 0.5C steps.. I didn't change anything.. maybe just updated z2m.. So I'm not sure if it's coincidence. I cannot see anything suspicious in z2m code.. so I'm really not sure. But since also others noticed such problem.. It really have some dramatic impact on battery life... Isn't there any default reporting setting which has been changed recently or something like that which would be common for all devices and hidden somewhere?

maxi1134, 2023-06-10:
> WSDCGQ11LM

Hey there!
Have you ever found a solution?

MahrWe, 2025-07-10:
I had two of these for years, I remember the first battery change being like two years into using them. 

Now I changed the batteries on both three times in the past months, first i thought i had a bad batch of batteries.

I put of the z2m updates for a while, back then i had no availability for any devices.