Koenkk/zigbee2mqtt #5503: Prolight Smart Classic RGBW LED lamp E27

danielukm8, 2021-01-01:
I manage to add the light with this config:
`{
zigbeeModel: [‘PROLIGHT E27 WHITE AND COLOUR’],
model: ‘PROLIGHT E27 WHITE AND COLOUR’,
vendor: ‘Prolight’,
description: ‘RGB LED Downlight’,
supports: ‘RGB, Dimming’,
extend: gledopto.light_onoff_brightness_colortemp_colorxy
}`

The only problem i have now is that i can’t control the colors in the hue app i have only dimming and on/off

i try olso to change the list line to : `extend: generic.light_onoff_brightness_colortemp_colorxy` and it doesnt work olso

any ideas?

The lights you can see them in the link below, and im useing diyhue and hue app/hue essentials
https://www.hubo.be/nl/p/prolight-smart-classic-rgbw-led-lamp-e27-9-5w-dimbaar-2-stuks-afstandsbediening/917706.html

Koenkk, 2021-01-02:
If you can also not control the colors via the Hue hub (which is claimed on the hubo site) I guess these bulbs don't support it.

danielukm8, 2021-01-02:
when I connect them via original hue bridge they work  perfectly with colours  (except hue entertainment)
when I connect via ZigBee2mqtt>diyhue> hue android app/hue essentials I can't control the colours

the reason I want to make them work is to check if the diyhue can simulate the hue entertainment and olso the price now it's 50 e 1 pack with 2 bulbs and one remote and if you buy 1 pack you get one pack free.
Happy new year!

Koenkk, 2021-01-02:
Can you control the colour directly via MQTT? (example: `color_xy` of https://www.zigbee2mqtt.io/devices/81809_81813.html#light)

danielukm8, 2021-01-02:
I works when i publish with mqtt explorer

danielukm8, 2021-01-02:
I find it strange i cant find any information on their website
https://danielukm.stackstorage.com/s/nrVYyn3KYMWloq1k
https://danielukm.stackstorage.com/s/1momMTdiNvCrLfXz

Koenkk, 2021-01-03:
Someone else added support for this device (https://github.com/Koenkk/zigbee-herdsman-converters/pull/2007). It's supported in the latest dev branch now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html).

Since you can correctly control the colors from MQTT it looks that the Zigbee2MQTT part works correctly therefore I will close this.

closed 2021-01-03 by Koenkk