Koenkk/zigbee2mqtt #5504: Moving from PI to docker

onlymejosh, 2021-01-01:
Hey there!

## What happened
I am trying to migrate from a PI to a new server as well as moving to docker.
So far I have:
- Copied the contents of the working the pis  `/opt/zigbee2mqtt/data` to a folder called `zigbee2mqtt` on the new server. 
My `docker-compose`  looks like this:

```
mosquitto:
    container_name: mosquitto
    image: eclipse-mosquitto:1.6.10
    restart: always
    ports:
        - '1883:1883'
    volumes:
      - ./mosquitto/config/mosquitto.conf:/mosquitto/config/mosquitto.conf
      - ./mosquitto/data:/mosquitto/data
      - ./mosquitto/log:/mosquitto/log
zigbee2mqtt:
    container_name: zigbee2mqtt
    depends_on:
      - mosquitto
    image: koenkk/zigbee2mqtt:1.16.2
    volumes:
      - ./zigbee2mqtt:/app/data
      - /run/udev:/run/udev:ro
    devices:
      - /dev/ttyUSB0:/dev/ttyUSB0
    restart: always
    privileged: true
    environment:
      - TZ=Europe/London
    network_mode: host
```

`zigbee2mqtt` folder looks like this:
```
log/
configuration.yaml
coordinator_backup.json
databse.db
databse.db.backup
state.json
```

`configuration.yaml`
```
homeassistant: true
permit_join: true
frontend:
  port: 8080
advanced:
  log_level: debug
  pan_id: 6760
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://haus.local:1883'
serial:
  port: /dev/ttyUSB0
devices:
```

I get errors saying the pan id is in use already. However changing the id I still result in 0 connected devices

```
zigbee2mqtt      | Zigbee2MQTT:info  2021-01-01 22:59:09: zigbee-herdsman started
zigbee2mqtt      | Zigbee2MQTT:info  2021-01-01 22:59:09: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20200805,"transportrev":2},"type":"zStack3x0"}'
zigbee2mqtt      | Zigbee2MQTT:debug 2021-01-01 22:59:09: Zigbee network parameters: {"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6761}
zigbee2mqtt      | Zigbee2MQTT:info  2021-01-01 22:59:09: Currently 0 devices are joined:
zigbee2mqtt      | Zigbee2MQTT:warn  2021-01-01 22:59:09: `permit_join` set to  `true` in configuration.yaml.
zigbee2mqtt      | Zigbee2MQTT:warn  2021-01-01 22:59:09: Allowing new devices to join.
zigbee2mqtt      | Zigbee2MQTT:warn  2021-01-01 22:59:09: Set `permit_join` to `false` once you joined all devices.
zigbee2mqtt      | Zigbee2MQTT:info  2021-01-01 22:59:09: Zigbee: allowing new devices to join.
zigbee2mqtt      | Zigbee2MQTT:info  2021-01-01 22:59:09: Started frontend on port 0.0.0.0:8080
zigbee2mqtt      | Zigbee2MQTT:info  2021-01-01 22:59:10: Connecting to MQTT server at mqtt://haus.local:1883
zigbee2mqtt      | Zigbee2MQTT:debug 2021-01-01 22:59:10: Using MQTT anonymous login
zigbee2mqtt      | Zigbee2MQTT:info  2021-01-01 22:59:10: Connected to MQTT server
zigbee2mqtt      | Zigbee2MQTT:info  2021-01-01 22:59:10: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
zigbee2mqtt      | Zigbee2MQTT:info  2021-01-01 22:59:10: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"04c15f7","coordinator":{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20200805,"transportrev":2},"type":"zStack3x0"},"log_level":"debug","network":{"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6761},"permit_join":true,"version":"1.16.2"}'
```

DoI need to do a full repair of all of my deices?

## What did you expect to happen
Devices appear as they did before

## Debug info
Zigbee2MQTT version: 1.16.2
Adapter hardware: zig-a-zig-ah!
Adapter firmware version:

```
zigbee2mqtt      | Zigbee2MQTT:debug 2021-01-01 22:58:58: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6761},"serialPort":{"path":"/dev/ttyUSB0"}}'
```

kevincaradant, 2021-01-01:
Hi

Did you have your list of  devices into `configuration.yaml` ? And `database.yaml` ? 

The channel is it the same than before ? (channel 11 here) ?

onlymejosh, 2021-01-01:
Hey,

Yep configuration and database.db are both correct by the looks of it.

I have flipped the pan_id around so it may be that I copied the wrong output. 

I haven't changed the channel so I am guessing it is correct.

I have tried to plug the pi version back in for now, but that is now not working either! This is running on `6760`

```
Zigbee2MQTT:debug 2021-01-01 23:29:45: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","databasePath":"/opt/zigbee2mqtt/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6760},"serialPort":{"path":"/dev/ttyUSB0"}}'

Zigbee2MQTT:error 2021-01-01 23:30:47: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-01-01 23:30:47: Failed to start zigbee
Zigbee2MQTT:error 2021-01-01 23:30:47: Exiting...
Zigbee2MQTT:error 2021-01-01 23:30:47: Error: AREQ - ZDO - stateChangeInd after 60000ms
    at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
```

rjasse, 2021-01-03:
Dear all,
I have the same type of problem with my new CC2652RB sitck.
1) change pan_id
2) start npm on my test-Laptop
3) pair only one router device (3 differnt tested: OSRAM-PLUG, SIVERCREST-PLUG, ZBMINI)
4) playing arround with the device is responding okay.
5) STOP zigbee2mqtt
6) restart FAILS with above message.

Network sniffing & herdsman-log:
[image]
[zb2.log](https://github.com/Koenkk/zigbee2mqtt/files/5762338/zb2.log)

I'll discuss the Sniffer-Output:
a) the 1. Beacon request (red) is seen after starting zigbee2mqtt.
b) The Router respons with the (blue) Superframe
c) the Router? resend the Beacon and respond to the 2. Superframe.
D) the zigbee-hersman receive the 2. superframe and is going to an unsuspected state followed by timeout.
note) all frames with MAC payload (yellow) sampled come from the paired router with stopped coordinator - ignore them

for me it shows as an timing issue with fast enviroments (for me: ubuntu-Laptop or docker on QNAP TS453)
Slower enviroments or an included USB-HUB can help to play correctly (can't test myself)

Hope this analysis helps for debugging.
BR, Ruediger

closed 2021-02-10 by github-actions[bot]