Koenkk/zigbee2mqtt #5507: (BUG) SEA801-Zigbee/SEA802-Zigbee not all functions and no converters

salopette, 2021-01-02:
https://github.com/nurikk/z2m-frontend/issues/235

Hello, the SEA801-Zigbee thermostat is missing some functions in the front end and no converters are available anywhere.
No matter which update arrow I press, there is always a message that no converter is available.
Can you fix it?

THX!

Koenkk, 2021-01-02:
Reading of these properties is indeed not supported so the refresh icon shouldn't show. Fixed this now.

Changes will be available in the latest dev branch tomorrow (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

salopette, 2021-01-02:
Thanks, can you also integrate the points in the frontend?

    "window_detection": "ON"

    "local_temperature_calibration": 0,

[image]

`{"id":54,"type":"EndDevice","ieeeAddr":"0x5c0272fffec23f51","nwkAddr":59238,"manufId":4098,"manufName":"_TZE200_c88teujp","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_c88teujp","powerSource":3,"zclVersion":3,"appVersion":85,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00124b001ca168ea","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":85,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1609613323547}
`

Koenkk, 2021-01-03:
- window_detection is already there. I believe it's the second state switch. @nurrik in case a device has multiple "states" this might seem confusing, maybe also add the property there?
- local_temperature_calibration -> added

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

salopette, 2021-01-03:
[image]

salopette, 2021-01-03:
running state is always with NA, I don't know what that is.

closed 2021-02-10 by github-actions[bot]

danielk117, 2021-02-24:
Hi,
`child_lock` does not really work. I could disable it over mqtt but not enable it. 
activating over mqtt shows `LOCKED` but the device isnt. I need to activate it on the device.