Koenkk/zigbee2mqtt #5511: Zigbee2MQTT doesn't start after update to 1.17

drivingmx, 2021-01-02:
## What happened
Zigbee2MQTT doesn't start after updating to 1.17
Error Log from Portainer.
>Using '/app/data' as data directory
> zigbee2mqtt@1.17.0 start /app
> node index.js
Zigbee2MQTT:info  2021-01-02 20:07:49: Logging to console and directory: '/app/data/log/2021-01-02.20-07-48' filename: log.txt
Zigbee2MQTT:info  2021-01-02 20:07:49: Starting Zigbee2MQTT version 1.17.0 (commit #07cdc9d)
Zigbee2MQTT:info  2021-01-02 20:07:49: Starting zigbee-herdsman (0.13.46)
Zigbee2MQTT:error 2021-01-02 20:08:59: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-01-02 20:08:59: Failed to start zigbee
Zigbee2MQTT:error 2021-01-02 20:08:59: Exiting...
Zigbee2MQTT:error 2021-01-02 20:08:59: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:94:31)
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.17.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.17.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2021-01-02T19_08_59_881Z-debug.log

## What did you expect to happen
It should start as usual

## How to reproduce it (minimal and precise)
every start of the docker container.

## Debug info
Zigbee2MQTT version: 1.17
Adapter hardware: CC2531
Adapter firmware version: ? (the former zigbee2mqtt version worked fine with it)

Arjenlodder, 2021-01-02:
I have this sometimes too (also with previous versions of Zigbee2MQTT). For me, stopping Zigbee2MQTT, unplugging the adapter, plugging it back in and starting Zigbee2MQTT always works with this exact error.

drivingmx, 2021-01-03:
I read about this fix. I tried this with a few variations but didn't work at all for me.

Koenkk, 2021-01-03:
@drivingmx try reflashing your coordinator

drivingmx, 2021-01-03:
I flashed my coordinator with the latest source routing firmware.
Now it works fine again. 
Any clue what was happening? Do I have to expect this behavior again?

Koenkk, 2021-01-03:
@drivingmx CC2531 sometimes crashes, source routing fw might improve it: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing . If you still get crashes often you have to upgrade to a more powerful coordinator (https://www.zigbee2mqtt.io/getting_started/what_do_i_need.html#supported-zigbee-adapter)

Closing this as OP issue has been solved.

closed 2021-01-03 by Koenkk

brammp, 2021-02-02:
Hello All,
I had the same issue i upgrade z2m 1.16.2 i just upgrade to 1.17.1 on my cc2531 with FW 20190619
i coud not connect to devices, tried re-pairing, when i enabled accept new devices it reported 'can not accept devices in Inter-PAN mode'

i had to upgrade the cordinator firmware to get evrything working(evrything is working great now), coud you please add "when using cc253* requires firmware 20201128 or newer " to the release notes so other people do not have to spend unnecessary debuging the system.

also the version in the cc2531 guide is not updated, might be handy to update those to the newest version of FW  
-Greetz