Koenkk/zigbee2mqtt #5516: Livolo Dimmer Switch (new Type)

Piscator74, 2021-01-02:
I have the following [Livolo Dimmer Switch](https://www.aliexpress.com/item/1005001665327663.html?spm=a2g0s.9042311.0.0.1aff4c4dAC95Tk) (new type) that I would like to add to Z2M.

When I pair the device (with the `loglevel: debug` option) I get the following.
* The device is identified as a "Livolo Zigbee switch (1 and 2 gang) (TI0001)" (but it is a dimmer obviously)
* None of the button presses on the dimmer are recognized. There is no info output  (so: no state_left, state_right or linkquality).
* Trying to control it as a 'normal' livolo switch (from OpenHAB) to turn it on/off also does nothing (I have roughly 15 'normal' Livolo switches and that works great)

Looking in detail in the log when pairing, I have the following:
```
info  2021-01-02 21:21:18: Device '0x00124b0021745db3' joined
info  2021-01-02 21:21:18: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b0021745db3","ieee_address":"0x00124b0021745db3"},"type":"device_joined"}'
info  2021-01-02 21:21:18: Starting interview of '0x00124b0021745db3'
info  2021-01-02 21:21:18: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b0021745db3","ieee_address":"0x00124b0021745db3","status":"started"},"type":"device_interview"}'
info  2021-01-02 21:21:18: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x00124b0021745db3"},"type":"device_connected"}'
info  2021-01-02 21:21:18: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x00124b0021745db3"},"type":"pairing"}'
debug 2021-01-02 21:21:18: Device '0x00124b0021745db3' announced itself
info  2021-01-02 21:21:18: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b0021745db3","ieee_address":"0x00124b0021745db3"},"type":"device_announce"}'
info  2021-01-02 21:21:18: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x00124b0021745db3"},"type":"device_announced"}'
debug 2021-01-02 21:21:19: Received Zigbee message from '0x00124b0021745db3', type 'readResponse', cluster 'genBasic', data '{"modelId":"TI0001          "}' from endpoint 6 with groupID 0
debug 2021-01-02 21:21:19: No converter available for 'TI0001' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"TI0001          "}'
debug 2021-01-02 21:21:19: Received Zigbee message from '0x00124b0021745db3', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"LIVOLO\u0000\u0010TI0001  "}' from endpoint 6 with groupID 0
debug 2021-01-02 21:21:19: No converter available for 'TI0001' with cluster 'genBasic' and type 'readResponse' and data '{"manufacturerName":"LIVOLO\u0000\u0010TI0001  "}'
debug 2021-01-02 21:21:19: Received Zigbee message from '0x00124b0021745db3', type 'readResponse', cluster 'genBasic', data '{"powerSource":1}' from endpoint 6 with groupID 0
debug 2021-01-02 21:21:19: No converter available for 'TI0001' with cluster 'genBasic' and type 'readResponse' and data '{"powerSource":1}'
debug 2021-01-02 21:21:19: Received Zigbee message from '0x00124b0021745db3', type 'readResponse', cluster 'genBasic', data '{"zclVersion":0}' from endpoint 6 with groupID 0
debug 2021-01-02 21:21:19: No converter available for 'TI0001' with cluster 'genBasic' and type 'readResponse' and data '{"zclVersion":0}'
debug 2021-01-02 21:21:19: Received Zigbee message from '0x00124b0021745db3', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 6 with groupID 0
debug 2021-01-02 21:21:19: No converter available for 'TI0001' with cluster 'genBasic' and type 'readResponse' and data '{}'
debug 2021-01-02 21:21:19: Received Zigbee message from '0x00124b0021745db3', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 6 with groupID 0
debug 2021-01-02 21:21:19: No converter available for 'TI0001' with cluster 'genBasic' and type 'readResponse' and data '{}'
debug 2021-01-02 21:21:20: Received Zigbee message from '0x00124b0021745db3', type 'readResponse', cluster 'genBasic', data '{"hwVersion":0}' from endpoint 6 with groupID 0
debug 2021-01-02 21:21:20: No converter available for 'TI0001' with cluster 'genBasic' and type 'readResponse' and data '{"hwVersion":0}'
debug 2021-01-02 21:21:20: Received Zigbee message from '0x00124b0021745db3', type 'readResponse', cluster 'genBasic', data '{"dateCode":"20060831        "}' from endpoint 6 with groupID 0
debug 2021-01-02 21:21:20: No converter available for 'TI0001' with cluster 'genBasic' and type 'readResponse' and data '{"dateCode":"20060831        "}'
debug 2021-01-02 21:21:20: Received Zigbee message from '0x00124b0021745db3', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,179,93,116,33,0,75,18,0,2,0,20,201,1,11],"type":"Buffer"}' from endpoint 6 with groupID 0
debug 2021-01-02 21:21:20: Received Zigbee message from '0x00124b0021745db3', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 6 with groupID 0
debug 2021-01-02 21:21:20: No converter available for 'TI0001' with cluster 'genBasic' and type 'readResponse' and data '{}'
info  2021-01-02 21:21:20: Successfully interviewed '0x00124b0021745db3', device has successfully been paired
info  2021-01-02 21:21:20: Device '0x00124b0021745db3' is supported, identified as: Livolo Zigbee switch (1 and 2 gang) [work in progress](https://github.com/Koenkk/zigbee2mqtt/issues/592) (TI0001)
info  2021-01-02 21:21:20: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Zigbee switch (1 and 2 gang) [work in progress](https://github.com/Koenkk/zigbee2mqtt/issues/592)","exposes":[{"endpoint":"left","features":[{"access":7,"description":"On/off state of the switch","endpoint":"left","name":"state","property":"state_left","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"endpoint":"right","features":[{"access":7,"description":"On/off state of the switch","endpoint":"right","name":"state","property":"state_right","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"TI0001","vendor":"Livolo"},"friendly_name":"0x00124b0021745db3","ieee_address":"0x00124b0021745db3","status":"successful","supported":true},"type":"device_interview"}'
info  2021-01-02 21:21:20: Configuring '0x00124b0021745db3'
info  2021-01-02 21:21:20: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"Zigbee switch (1 and 2 gang) [work in progress](https://github.com/Koenkk/zigbee2mqtt/issues/592)","friendly_name":"0x00124b0021745db3","model":"TI0001","supported":true,"vendor":"Livolo"},"type":"pairing"}'
info  2021-01-02 21:21:20: Successfully configured '0x00124b0021745db3'
```

The interesting line to me seems the following.
```
debug 2021-01-02 21:21:20: No converter available for 'TI0001' with cluster 'genBasic' and type 'readResponse' and data '{}'
```

It does not seem to have any data (???). I am not sure where to go exactly from here, but I suppose the data is 'hidden' in the raw data. When I just use the ON/OFF toggle, I get the following raw data in the log (actions are commented)
```
# do nothing
debug 2021-01-02 21:21:20: Received Zigbee message from '0x00124b0021745db3', type 'raw', cluster 'genPowerCfg', data '{"data":[124,210,21,216,128,179,93,116,33,0,75,18,0,19,20,0],"type":"Buffer"}' from endpoint 6 with groupID 0
debug 2021-01-02 21:21:21: Received Zigbee message from '0x00124b0021745db3', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,179,93,116,33,0,75,18,0,2,0,20,201,1,11],"type":"Buffer"}' from endpoint 6 with groupID 0
debug 2021-01-02 21:21:23: Received Zigbee message from '0x00124b0021745db3', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,179,93,116,33,0,75,18,0,2,0,20,201,1,11],"type":"Buffer"}' from endpoint 6 with groupID 0
debug 2021-01-02 21:21:24: Received Zigbee message from '0x00124b0021745db3', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,179,93,116,33,0,75,18,0,2,0,20,201,1,11],"type":"Buffer"}' from endpoint 6 with groupID 0
[... cut ...]
debug 2021-01-02 21:21:47: Received Zigbee message from '0x00124b0021745db3', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,179,93,116,33,0,75,18,0,2,0,20,201,1,11],"type":"Buffer"}' from endpoint 6 with groupID 0
debug 2021-01-02 21:21:48: Received Zigbee message from '0x00124b0021745db3', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,179,93,116,33,0,75,18,0,2,0,20,201,1,11],"type":"Buffer"}' from endpoint 6 with groupID 0
```

Any clues / pointers how I could add support for this device?

Koenkk, 2021-01-03:
Livolo devices don't use the default zigbee commands, it has it's own protocol. This make adding these devices very hard. You will need the original gateway (and a lot of time) and sniff the traffic with that: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html . This should make it possible to reverse engineer the used protocol. I suggest to read #592 first, it might contain useful tips on how to get started.

whitepail, 2021-01-03:
I was able to pair it and control it with on/off command (adding some extra to devices.js and fromZigbee.js) but now I'm stuck with "heartbeat" problem: every n seconds dimmer stop sending events and responding to zigbee commands.
Looks like it waits for some special heartbeat packet from original livolo gateway to resume working.

If someone could share wireshark dump for communication between dimmer and original gateway - it'll be great help.

Piscator74, 2021-01-04:
@koenkk. Unfortunately I do not have the original Livolo gateway and I have limited time ;-), so thanks for the warning. I was hoping I could add it using the existing Livolo implementation and the raw packets as an example. But a "heartbeat" as whitepail describes does not sound so promising...

I did dive a bit in the code and I was wondering if I could detect the dimmer as a separate / new device. That I could figure out using the existing code (see below: fromZigbee.js). I suppose I need to base this dimmer device on: ```preset.light_onoff_brightness``` ?

```
    livolo_switch_state_raw: {
        cluster: 'genPowerCfg',
        type: ['raw'],
        convert: (model, msg, publish, options, meta) => {
            /*
            header                ieee address            info data
            new socket
            [124,210,21,216,128,  199,147,3,24,0,75,18,0,  19,7,0]       after interview
            [122,209,             199,147,3,24,0,75,18,0,  7,1,6,1,0,11] off
            [122,209,             199,147,3,24,0,75,18,0,  7,1,6,1,1,11] on
            
            new switch
            [124,210,21,216,128,  228,41,3,24,0,75,18,0,  19,1,0]       after interview
            [122,209,             228,41,3,24,0,75,18,0,  7,1,0,1,0,11] off
            [122,209,             228,41,3,24,0,75,18,0,  7,1,0,1,1,11] on
            
            old switch
            [124,210,21,216,128,  170, 10,2,24,0,75,18,0,  17,0,1] after interview
            [124,210,21,216,0,     18, 15,5,24,0,75,18,0,  34,0,0] left: 0, right: 0
            [124,210,21,216,0,     18, 15,5,24,0,75,18,0,  34,0,1] left: 1, right: 0
            [124,210,21,216,0,     18, 15,5,24,0,75,18,0,  34,0,2] left: 0, right: 1
            [124,210,21,216,0,     18, 15,5,24,0,75,18,0,  34,0,3] left: 1, right: 1
            
            new dimmer
            [122,209,179,93,116,   33, 0,75,18,0,2,0,20,  201,1,11] after interview
            */
            const malformedHeader = Buffer.from([0x7c, 0xd2, 0x15, 0xd8, 0x00]);
            const infoHeader = Buffer.from([0x7c, 0xd2, 0x15, 0xd8, 0x80]);
            // status of old devices
            if (msg.data.indexOf(malformedHeader) === 0) {
                const status = msg.data[15];
                return {
                    state_left: status & 1 ? 'ON' : 'OFF',
                    state_right: status & 2 ? 'ON' : 'OFF',
                };
            }
            // info about device
            if (msg.data.indexOf(infoHeader) === 0) {
                if (msg.data.includes(Buffer.from([19, 7, 0]), 13)) {
                    // new socket, hack
                    meta.device.modelID = 'TI0001-socket';
                    meta.device.save();
                }
                if (msg.data.includes(Buffer.from([19, 1, 0]), 13)) {
                    // new switch, hack
                    meta.device.modelID = 'TI0001-switch';
                    meta.device.save();
                }
            }
            
            // patch for new livolo dimmer
            const infoHeaderDimmer = Buffer.from([0x7a, 0xd1, 0xb3, 0x5d, 0x74]);
            if (msg.data.indexOf(infoHeaderDimmer) === 0) {
                if (meta.logger) meta.logger.debug('Detected Livolo Dimmer!');
                // new dimmer, hack
                meta.device.modelID = 'TI0001-dimmer';
                meta.device.save();
            }
        },
    },
```

@whitepail. Could you share the code / changes  you have so far?

Piscator74, 2021-01-04:
I have the following working:
   * Read ON/OFF status
   * Read DIMMER level
   * Control ON/OFF status  (it works similar as ```livolo_socket_switch_on_off``` in toZigbee.js)

I guess that was the low-hanging fruit. What is not working is:
   * It only works for a limited time. That's the timeout @whitepail was describing. (just about a minute).
   * I have no idea how to control the DIMMER level.
I suppose I am stuck here. Without the original gateway and sniffing the traffic, there is no way to continue.

whitepail, 2021-01-11:
Already ordered original gateway from livolo. 
Will share the dump as soon as GW will arrive.

Rincovanderbeek, 2021-01-13:
Great that you guys are working on this.  I installed a Zig-a-Zag with Zigbee2Mqtt but unfortunately the Livolo switches I've got (1 and 2 gang, new type) are not detected. Either I send them back or hopefully you find a way to get it working :-)

Rincovanderbeek, 2021-01-19:
Hi Koen, I understand your point that checking out this Livolo zigbee would take a lot of time.
Hopefully somewhere in the future it'll be an option.
Thx. for your reply.

whitepail, 2021-01-20:
Adding raw data for futher analysis.
Pairing - packets 982-1167

dimming packets
8446 - setting to minimum (0%)
8574 - 20%
8657 - 39%
8913 - 100%

Didn't find any special "heartbeat" packets, looks like after good pairing it's all ok.
To pair with real gateway I tried more than 30 times before success. 

Byte on printscreen - dimmer level
[image]

wireshark file
[dimmer pairing and dimming.zip](https://github.com/Koenkk/zigbee2mqtt/files/5842825/dimmer.pairing.and.dimming.zip)

Piscator74, 2021-01-20:
Thanks for the log, whitepail. At least can I make some sense out of it.
At least, this should give a start to get a step further. When I have some time  ;-)

### Recorded responses
Some of my recorded responses from the dimmer to zigbee2mqtt (with debug level) relate to your log
```
header                ieee address            info data
[124,210,21,216,128,  179,93,116,33,0,75,18,0, 19,20,0]          after interview
[122,209,             179,93,116,33,0,75,18,0, 2,0,20,201,1,11]  idle (every second)

[122,209,             179,93,116,33,0,75,18,0, 13,1,2,1,13,11]   on
[122,209,             179,93,116,33,0,75,18,0, 13,1,2,0,13,11]   off

when switching level when the device is ON: level message followed by an ON message
[122,209,             179,93,116,33,0,75,18,0,  5,1,3,0,1,11]     level 0
[122,209,             179,93,116,33,0,75,18,0,  5,1,3,0,12,11]    level 1
[122,209,             179,93,116,33,0,75,18,0,  5,1,3,0,24,11]    level 2 
[122,209,             179,93,116,33,0,75,18,0,  5,1,3,0,36,11]    level 3
[122,209,             179,93,116,33,0,75,18,0,  5,1,3,0,48,11]    level 4 
[122,209,             179,93,116,33,0,75,18,0,  5,1,3,0,61,11]    level 5 
[122,209,             179,93,116,33,0,75,18,0,  5,1,3,0,73,11]    level 6 
[122,209,             179,93,116,33,0,75,18,0,  5,1,3,0,85,11]    level 7
[122,209,             179,93,116,33,0,75,18,0,  13,1,2,1,13,11]   level message followed by an ON message

when switching the device ON (when it is on level 7), it looks like the "level value + 1" is read
[122,209,             179,93,116,33,0,75,18,0,  5,1,3,0,86,11]    level 7
[122,209,             179,93,116,33,0,75,18,0,  13,1,2,1,13,11]   level message followed by an ON message
```

### Level communication
Translating the level message to HEX
```
[122,209,             179,93,116,33,0,75,18,0,  5,  1, 3, 0,85,11] -->
[7a, d1,              x,  x, x,  x, x,x,  x,x,  05,01,03,00,ZZ,0b] (and ZZ is the level)
```

So looking at package 8657 (set dimmer level to 39% (=HEX 27),
you see the dimmer responds in package 8667 with the level
```
[7a, d1,              x,  x, x,  x, x,x,  x,x,  05,01,03,00,27,0b]   <-- 39%
```

### Pairing / 'idle' message after interview.
Translating the idle message to HEX
```
[122,209,             179,93,116,33,0,75,18,0,   2, 0,20,201, 1,11] -->
[7a, d1,              x,  x, x,  x, x,x,  x,x,  02,00,14, c9,01,0b]
```

Device --> Gateway.  I can find that message back in package 1048
Gateway --> Device.  Response from gateway in package 1066 hopefully on package 1048

whitepail, 2021-01-21:
Adding some suplementary info regarding livolo gateway.

Gateway based on two chipsets:
CC2530 (in form of SZ05-L-PRO-2). Chip has firmware based on SampleLight app+Coordinator from TI Zstack home 1.2.2. Firmware attached [firmware.zip](https://github.com/Koenkk/zigbee2mqtt/files/5850914/firmware.zip)

CC2530 communicates to second chip STM32F412 (MXChip p53-emw3166) via serial using non-standard protocol.
STM32F412 is running MicoOS 3.6.0 (sources: https://code.aliyun.com/work_prj/mico-os/tree/3.6.0) + linkkit (https://github.com/aliyun/iotkit-embedded/).

Linkkit is connected to Alibaba IOT cloud (public.iot-as-mqtt.ap-southeast-1.aliyuncs.com) using Demo account (insecure).

Board has three debug interfaces:
1) CC2530 debug interface for CC-PROG. Can be used for flashing/dumping software from CC2530.
2) TTL UART of STM32F412 with debug console output, allowing to sniff all communications between chips and communication to cloud.
3) JTAG interface of STM32F412, allowing to upload/download firmware.

[image]

Two ways of using this data:
1) Use cc2530 firmware for flashing cheap 2530 module from ali and prepare serial app/converter for further use without livolo gateway. Will allow use of livolo devices in the future without need to mass-rewrite of z2m/shephard.
2) Flash STM32 with adopted slsys firmware for future use as universal zigbee gateway.

Piscator74, 2021-02-01:
That sounds good, whitepail! Could you share some code? How do you set and readback the dimmer levels?
I hope we can figure out the "configuration part", so the dimmer does not need to be connected with an original gateway. But is is a start. I hope to have some time soon, so I'll be able to look into this as well.

whitepail, 2021-02-02:
Added two pull requests.
First for writeUndiv method (to support "Write Attributes Undivided" packet as Livolo uses those ZCL Command Id bytes to identify different messages).
Second is change to devices.js + adding special methods for dimmer to converters.

After pairing device will be shown in web frontend as TI0001, need to open the device in web frontend and press "Reconfigure" button. 
[image]

Piscator74, 2021-03-06:
Just a message to not close this issue. I'll be looking at it when I have some time ;-)

aesedepece, 2021-03-30:
What's the protocol limitation?

whitepail, 2021-03-30:
I've tried to reuse "readResponse" login message that @aesedepece you emulated for cover to bypass "heartbeat problem".
Looks like it doesn't help for dimmer as message format differs between Z2M and Livolo GW.
In livolo gw sniff log it has format
7c d2 1a e9 01 02 08 *01* 22 *01* 00 00 00 00 00 00
Emulated by z2m sniffed 
7c d2 1a e9 01 02 08 *00* 22 *00* 00 00 00 00

The difference is attr_status. If you set attr_status to "01", attribute value will be deleted according to foundation.ts

Piscator74, 2021-03-31:
I have the Livolo Dimmer now also working with zigbee2mqtt :-)

My first try was trying to set it through the 'state' feature. There seems to be an issue with the following code (commented out). When I replace it with the following code, setting and getting the 'ON' and 'OFF' status works fine.
```
    livolo_dimmer_state: {
        cluster: 'genPowerCfg',
        type: ['raw'],
        convert: (model, msg, publish, options, meta) => {
            const stateHeader = Buffer.from([122, 209]);
            if (msg.data.indexOf(stateHeader) === 0) {
                // code in 1.18.1 with an issue
                //if (msg.data[10] === 7) {
                //    const status = msg.data[14];
                //    return {state: status & 1 ? 'ON' : 'OFF'};
                //}
                // code that fixes the problem 
                if (msg.data[10] === 13) {
                    const status = msg.data[13];
                    return {state: status & 1 ? 'ON' : 'OFF'};
                } else if (msg.data[10] === 5) { // TODO: Unknown dp, assumed value type
                    const value = msg.data[14] * 10;
                    return {brightness: Math.round((value / 1000) * 255), brightness_percent: Math.round(value / 10), level: value};
                }
            }
        },
    },
```

Piscator74, 2021-05-07:
Just a message to remove the stale label.
I am still planning to commit the code above when I have some time.

closed 2021-06-15 by github-actions[bot]

resh1309, 2021-11-13:
Can I ask a question regarding the Livolo switch. Will this fix be working only with zigbee2mqtt or do you need the Livolo gateway to connect to zigbee2mqtt?

wizake, 2022-02-16:
@Piscator74  did you manage to find some time to work on the livolo curtain switch ?

Piscator74, 2024-01-27:
For the livolo curtain switch, see https://github.com/Koenkk/zigbee2mqtt/discussions/8525