Koenkk/zigbee2mqtt #5517: Another LIDL/Silvercrest support request

RhinoRich, 2021-01-02:
Sincere apologies as this issue seems to have been covered before and resolved, but the resolutions aren't working for me to add the HG06337 sockets. Also, apologies in advance for any breaches in protocol in raising the issue. I'm very new to this.

I'm using the HA add-on and have followed the steps to find the right devices.js file for my build (1.16.2.2). The correct devices.js files is from 30 November so precedes the launch of Lidl hardware, but I pasted in the multi-country fingerprint found on one of the earlier threads. The add-on re-started fine but still with unsupported sockets. 

Then I found https://www.zigbee2mqtt.io/devices/HG06337.html page and added the HA manual configuration to the add-on configuration, changing the friendly name to the one I'd set. 

Same result:

Zigbee2MQTT:info  2021-01-02 23:00:03: Currently 6 devices are joined:
Zigbee2MQTT:info  2021-01-02 23:00:03: 0x588e81fffed3802c (0x588e81fffed3802c): TS0505A - TuYa RGB+CCT light controller (Router)
Zigbee2MQTT:info  2021-01-02 23:00:03: 0x588e81fffed9a995 (0x588e81fffed9a995): TS0505A - TuYa RGB+CCT light controller (Router)
Zigbee2MQTT:info  2021-01-02 23:00:03: 0x842e14fffe1df25a (0x842e14fffe1df25a): TS0505A - TuYa RGB+CCT light controller (Router)
Zigbee2MQTT:info  2021-01-02 23:00:03: 0x588e81fffee38c23 (0x588e81fffee38c23): TS0505A - TuYa RGB+CCT light controller (Router)
Zigbee2MQTT:info  2021-01-02 23:00:03: Lidl Socket 2 (0x842e14fffef8e702): Not supported (Router)
Zigbee2MQTT:info  2021-01-02 23:00:03: Lidl Socket 1 (0x842e14fffef59c48): Not supported (Router)

kevincaradant, 2021-01-03:
Hi

Not concerned my self by this issue, but did you try to update Z2M on the latest version 1.17.0 hopping the devices.js file could be updated and by consequence your device could be recognized ? :)

RhinoRich, 2021-01-03:
> Hi
> 
> Not concerned my self by this issue, but did you try to update Z2M on the latest version 1.17.0 hopping the devices.js file could be updated and by consequence your device could be recognized ? :)

1.16.2 is the latest version for Hassio

RhinoRich, 2021-01-03:
Solved. My built was giving an error for `extend: present.switch` so I changed it to 

`    // Lidl
    {
        fingerprint: [{modelID: 'TS011F', manufacturerName: '_TZ3000_00mk2xzy'}],
        model: 'HG06337',
        vendor: 'Lidl',
        description: 'Silvercrest smart plug (EU, CH, FR, BS)',
        extend: generic.switch,
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(11);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
            await reporting.onOff(endpoint);
        },
     },`

and that's now all working.

closed 2021-01-03 by RhinoRich