Koenkk/zigbee2mqtt #5518: BAC-002ALZB

francisgambin, 2021-01-02:
I am not able to run the converter

## Information about the device + link
BAC-002ALZB is similar to https://www.zigbee2mqtt.io/devices/BHT-002-GCLZB.html
https://h5.es.aliexpress.com/item/1005001452232622.html
## data/database.db entry of the device
'0x60a423fffeb280cf':
  friendly_name: 'TermostatoDespacho'

kevincaradant, 2021-01-03:
Hi

Try to start Z2M in debug mode and add logs here. 
https://www.zigbee2mqtt.io/information/debug.html

It will be necessary when people will try to help you ')

francisgambin, 2021-01-03:
Hi and Thanks!

info  2021-01-02 23:55:40: Logging to console and directory: '/share/zigbee2mqtt/log/2021-01-02.23-55-39' filename: log.txt
info  2021-01-02 23:55:40: Starting Zigbee2MQTT version 1.16.2 (commit #1.16.2)
info  2021-01-02 23:55:40: Starting zigbee-herdsman (0.13.37)
info  2021-01-02 23:55:43: zigbee-herdsman started
info  2021-01-02 23:55:43: Coordinator firmware version: '{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190619,"transportrev":2},"type":"zStack12"}'
info  2021-01-02 23:55:43: Currently 3 devices are joined:
info  2021-01-02 23:55:43: RouterZigbeePasillo (0x00124b001c5a7d19): CC2530.ROUTER - Custom devices (DiY) [CC2530 router](http://ptvo.info/cc2530-based-zigbee-coordinator-and-router-112/) (Router)
info  2021-01-02 23:55:43: TemperaturaBajo (0x5c0272fffebbe76a): TS0201 - TuYa Temperature & humidity sensor with display (EndDevice)
info  2021-01-02 23:55:43: termostatoDespacho (0x60a423fffeb280cf): Not supported (Router)
info  2021-01-02 23:55:43: Zigbee: disabling joining new devices.
info  2021-01-02 23:55:43: Started frontend on port 0.0.0.0:8099
warn  2021-01-02 23:55:43: Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_dzuqwsyg'
warn  2021-01-02 23:55:43: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
info  2021-01-02 23:55:43: Connecting to MQTT server at mqtt://192.168.1.14
info  2021-01-02 23:55:44: Connected to MQTT server
info  2021-01-02 23:55:44: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
info  2021-01-02 23:55:44: MQTT publish: topic 'zigbee2mqtt/RouterZigbeePasillo', payload '{"led":false}'
info  2021-01-02 23:55:44: MQTT publish: topic 'zigbee2mqtt/TemperaturaBajo', payload '{"battery":50,"humidity":37.5,"temperature":18.4,"voltage":2600}'
info  2021-01-02 23:55:44: MQTT publish: topic 'zigbee2mqtt/bridge/info', payload '{"commit":"1.16.2","config":{"advanced":{"adapter_concurrent":null,"adapter_delay":null,"availability_blacklist":[],"availability_blocklist":[],"availability_passlist":[],"availability_timeout":0,"availability_whitelist":[],"cache_state":true,"cache_state_persistent":true,"cache_state_send_on_startup":true,"channel":11,"elapsed":false,"ext_pan_id":[221,221,221,221,221,221,221,221],"homeassistant_discovery_topic":"homeassistant","homeassistant_legacy_triggers":true,"homeassistant_status_topic":"hass/status","last_seen":"disable","legacy_api":true,"log_directory":"/share/zigbee2mqtt/log/%TIMESTAMP%","log_file":"log.txt","log_level":"info","log_output":["console","file"],"log_rotation":true,"log_syslog":{},"pan_id":55019,"report":false,"soft_reset_timeout":0,"timestamp_format":"YYYY-MM-DD HH:mm:ss"},"ban":[],"blocklist":[],"device_options":{},"devices":{"0x00124b001c5a7d19":{"friendly_name":"RouterZigbeePasillo"},"0x5c0272fffebbe76a":{"friendly_name":"TemperaturaBajo"},"0x60a423fffeb280cf":{"friendly_name":"termostatoDespacho"}},"experimental":{"new_api":true,"output":"json"},"external_converters":[],"frontend":{"port":8099},"groups":{},"homeassistant":true,"map_options":{"graphviz":{"colors":{"fill":{"coordinator":"#e04e5d","enddevice":"#fff8ce","router":"#4ea3e0"},"font":{"coordinator":"#ffffff","enddevice":"#000000","router":"#ffffff"},"line":{"active":"#009900","inactive":"#994444"}}}},"mqtt":{"base_topic":"zigbee2mqtt","force_disable_retain":false,"include_device_information":false,"server":"mqtt://192.168.1.14","user":"mqtt"},"passlist":[],"permit_join":false,"queue":{},"serial":{"disable_led":false,"port":"/dev/ttyACM0"},"whitelist":[]},"coordinator":{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190619,"transportrev":2},"type":"zStack12"},"log_level":"info","network":{"channel":11,"extended_pan_id":"0xdddddddddddddddd","pan_id":55019},"permit_join":false,"version":"1.16.2"}'
info  2021-01-02 23:55:44: MQTT publish: topic 'zigbee2mqtt/bridge/devices', payload '[{"definition":null,"endpoints":{"1":{"bindings":[],"clusters":{"input":[],"output":[]}},"11":{"bindings":[],"clusters":{"input":["ssIasAce"],"output":["ssIasZone","ssIasWd"]}},"110":{"bindings":[],"clusters":{"input":[],"output":[]}},"12":{"bindings":[],"clusters":{"input":[],"output":[]}},"13":{"bindings":[],"clusters":{"input":["genOta"],"output":[]}},"2":{"bindings":[],"clusters":{"input":[],"output":[]}},"242":{"bindings":[],"clusters":{"input":[],"output":[]}},"3":{"bindings":[],"clusters":{"input":[],"output":[]}},"4":{"bindings":[],"clusters":{"input":[],"output":[]}},"47":{"bindings":[],"clusters":{"input":[],"output":[]}},"5":{"bindings":[],"clusters":{"input":[],"output":[]}},"6":{"bindings":[],"clusters":{"input":[],"output":[]}},"8":{"bindings":[],"clusters":{"input":[],"output":[]}}},"friendly_name":"Coordinator","ieee_address":"0x00124b001949b178","interview_completed":true,"interviewing":false,"network_address":0,"supported":false,"type":"Coordinator"},{"date_code":"2018-09-26","definition":{"description":"[CC2530 router](http://ptvo.info/cc2530-based-zigbee-coordinator-and-router-112/)","exposes":[{"access":1,"name":"led","property":"led","type":"binary","value_off":false,"value_on":true},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"CC2530.ROUTER","vendor":"Custom devices (DiY)"},"endpoints":{"8":{"bindings":[],"clusters":{"input":["genBasic","genOnOff"],"output":["genBasic","ge [... cut ...] "unique_id":"0x5c0272fffebbe76a_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
info  2021-01-02 23:55:46: MQTT publish: topic 'zigbee2mqtt/RouterZigbeePasillo', payload '{"led":false,"linkquality":18}'
warn  2021-01-02 23:56:02: Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_dzuqwsyg'
warn  2021-01-02 23:56:02: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.

francisgambin, 2021-01-04:
INFO: Successfully interviewed '0x60a423fffeb280cf', device has successfully been paired
WARNING: Device '0x60a423fffeb280cf' with Zigbee model 'TS0601' and manufacturer name '_TZE200_dzuqwsyg' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0x60a423fffeb280cf","ieee_address":"0x60a423fffeb280cf","status":"successful","supported":false},"type":"device_interview"}'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/devices', payload '[{"definition":null,"endpoints":{"1":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]},"11":{"bindings":[],"clusters":{"input":["ssIasAce"],"output":["ssIasZone","ssIasWd"]},"configured_reportings":[]},"110":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]},"12":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]},"13":{"bindings":[],"clusters":{"input":["genOta"],"output":[]},"configured_reportings":[]},"2":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]},"242":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]},"3":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]},"4":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]},"47":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]},"5":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]},"6":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]},"8":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]}},"friendly_name":"Coordinator","ieee_address":"0x00124b001949b178","interview_completed":true,"interviewing":false,"network_address":0,"supported":false,"type":"Coordinator"},{"date_code":"2018-09-26","definition":{"description":"[CC2530 router](http://ptvo.info/cc2530-based-zigbee-coordinator-and-router-112/)","exposes":[{"access":1,"name":"led","property":"led","type":"binary","value_off":false,"value_on":true},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"CC2530.ROUTER","vendor":"Custom devices (DiY)"},"endpoints":{"8":{"bindings":[],"clusters":{"input":["genBasic","genOnOff"],"output":["genBasic","genOnOff"]},"configured_reportings":[]}},"friendly_name":"RouterZigbeePasillo","ieee_address":"0x00124b001c5a7d19","interview_completed":true,"interviewing":false,"model_id":"lumi.router","network_address":28755,"power_source":"Mains (single phase)","supported":true,"type":"Router"},{"date_code":"","definition":{"description":"Temperature & humidity sensor with display","exposes":[{"access":1,"description":"Remaining battery in %","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"description":"Measured temperature value","name":"temperature","property":"temperature","type":"numeric","unit":"°C"},{"access":1,"description":"Measured relative humidity","name":"humidity","property":"humidity","type":"numeric","unit":"%"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"TS0201","vendor":"TuYa"},"endpoints":{"1":{"bindings":[],"clusters":{"input":["genBasic","genPowerCfg","msTemperatureMeasurement","msRelativeHumidity"],"output":["genOta"]},"configured_reportings":[]}},"friendly_name":"TemperaturaBajo","ieee_address":"0x5c0272fffebbe76a","interview_completed":true,"interviewing":false,"model_id":"TS0201","network_address":38924,"supported":true,"type":"EndDevice"},{"date_code":"","definition":null,"endpoints":{"1":{"bindings":[],"clusters":{"input":["genBasic","genGroups","genScenes","manuSpecificTuya"],"output":["genOta","genTime"]},"configured_reportings":[]}},"friendly_name":"0x60a423fffeb280cf","ieee_address":"0x60a423fffeb280cf","interview_completed":true,"interviewing":false,"model_id":"TS0601","network_address":20210,"power_source":"Mains (single phase)","supported":false,"type":"Router"}]'

francisgambin, 2021-01-04:
Sorry closed for error

closed 2021-02-12 by github-actions[bot]

juan11perez, 2021-03-18:
Good day. 
I created the below device config based on [Moes BHT-002-GCLZB](https://www.zigbee2mqtt.io/devices/BHT-002-GCLZB.html) and it does the job. 
Only issue is tidiness. Instead of showing heating in the dashboard, it should show cooling. 

I tried to create copies of the converters, replacing heating with cooling but i kept getting errors.

`     {
        fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_dzuqwsyg'}],
        model: 'BAC-002-ALZB',
        vendor: 'HKGK',
        description: 'BAC series Thermostat',
        fromZigbee: [fz.moes_thermostat],
        toZigbee: [tz.moes_thermostat_child_lock, tz.moes_thermostat_current_heating_setpoint, tz.moes_thermostat_mode,
            tz.moes_thermostat_standby1, tz.moes_thermostat_sensor, tz.moes_thermostat_calibration,
            tz.moes_thermostat_deadzone_temperature, tz.moes_thermostat_max_temperature_limit],
        exposes: [e.child_lock(), e.deadzone_temperature(), e.max_temperature_limit(),
            exposes.climate().withSetpoint('current_heating_setpoint', 5, 30, 1, ea.STATE_SET)
                .withLocalTemperature(ea.STATE).withLocalTemperatureCalibration(ea.STATE_SET)
                .withSystemMode(['off', 'cool'], ea.STATE_SET).withRunningState(['idle', 'heat', 'cool'], ea.STATE)
                .withPreset(['hold', 'program']).withSensor(['IN', 'AL', 'OU'], ea.STATE_SET)],
        onEvent: tuya.onEventSetLocalTime,
    },`

Also added below in the toZigbee converter file, to at least get cool in the drop-down menu.

`   moes_thermostat_standby1: {
        key: ['system_mode'],
        convertSet: async (entity, key, value, meta) => {
            await tuya.sendDataPointBool(entity, tuya.dataPoints.state, value === 'cool');
        },
    },  `

Any guidance to tidy this up is appreciated.

a2nt, 2022-07-12:
Thermostat works ok, but doesn't allows to control fan speed using zigbee2mqtt, while Smart Life does allows to do it.