Koenkk/zigbee2mqtt #5524: Mijia Smoke Sensor got some issues

kevincaradant, 2021-01-03:
```
ERROR: Publish 'set' 'selftest' to 'smoke_couloir' failed: 'Error: Write 0x00158d00045afdb9/1 ssIasZone({"65521":{"value":50397184,"type":35}}, {"timeout":35000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null}) failed (Data request failed with error: 'MAC transaction expired' (240))'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'selftest' to 'smoke_couloir' failed: 'Error: Write 0x00158d00045afdb9/1 ssIasZone({\"65521\":{\"value\":50397184,\"type\":35}}, {\"timeout\":35000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":4447,\"transactionSequenceNumber\":null}) failed (Data request failed with error: 'MAC transaction expired' (240))'","meta":{"friendly_name":"smoke_couloir"},"type":"zigbee_publish_error"}'
```

And 
```
ERROR: Publish 'set' 'sensitivity' to 'smoke_couloir' failed: 'Error: Write 0x00158d00045afdb9/1 ssIasZone({"65521":{"value":67174400,"type":35}}, {"timeout":35000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null}) failed (Data request failed with error: 'MAC transaction expired' (240))'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'sensitivity' to 'smoke_couloir' failed: 'Error: Write 0x00158d00045afdb9/1 ssIasZone({\"65521\":{\"value\":67174400,\"type\":35}}, {\"timeout\":35000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":4447,\"transactionSequenceNumber\":null}) failed (Data request failed with error: 'MAC transaction expired' (240))'","meta":{"friendly_name":"smoke_couloir"},"type":"zigbee_publish_error"}'
```

And sometimes when I switch between 2 sensibility or start a test on the smoke sensor, that's seems to work but with a delay between 2 and 15 seconds later.
```
zigbee2mqtt    | Zigbee2MQTT:info  2021-01-03 13:31:45: MQTT publish: topic 'zigbee2mqtt/smoke_couloir', payload '{"ac_status":false,"battery":100,"battery_low":false,"enrolled":false,"linkquality":141,"restore_reports":false,"sensitivity":"low","smoke":false,"smoke_density":0,"supervision_reports":false,"tamper":false,"trouble":false,"voltage":3055}'
zigbee2mqtt    | Zigbee2MQTT:error 2021-01-03 13:31:46: No converter available for 'get' 'sensitivity' ()
zigbee2mqtt    | Zigbee2MQTT:error 2021-01-03 13:31:46: No converter available for 'get' 'sensitivity' ()
INFO: MQTT publish: topic 'zigbee2mqtt/smoke_couloir', payload '{"ac_status":false,"battery":100,"battery_low":false,"enrolled":false,"linkquality":111,"restore_reports":false,"sensitivity":"low","smoke":false,"smoke_density":0,"supervision_reports":false,"tamper":false,"trouble":false,"voltage":3055}'
```

Any idea why this happens only on the smoke sensor ? Thank you 

[image]

Koenkk, 2021-01-03:
- Can you provide the networkmap?
- Can you add your coordinator/firmware like requested in the issue template?

kevincaradant, 2021-01-03:
Yep Sorry I didn't paste the latest picture about coordinator

[image]

And this is the map:
[image]

(smoke_couloir) is directly connected on the coordinator

Koenkk, 2021-01-03:
There have been a lot of reports/investigation already about this: #4285, #4569, #2368. It happens because the smoke sensor only polls its parent every 30 seconds. Most routers (and our coordinator) don't wait for such long and will raise a timeout. I've experimented with increasing the timeout in the coordinator firmware but this was not an acceptable solution.

I suggest to pair the device to a Xiaomi router (so in the networkmap the smoke sensor shows a direct connection to the xiaomi router). It is known that the smoke sensor doesn't change parents after that. Via the frontend you can force it to pair via a certain device.

kevincaradant, 2021-01-03:
Hi

OK thanks for the feedback, so what you suggest is to force the smoke sensor to be on one of my router (bitron video in my case) ? 

how can I do that ? 

Is it here ?

[image]

kevincaradant, 2021-01-03:
I saw in Home Assistant these error also about the smoke sensor. 
 Is it normal ?
 
 ```
 home-assistant   | Traceback (most recent call last):
home-assistant   |   File "/usr/src/homeassistant/homeassistant/components/mqtt/__init__.py", line 1304, in discovery_callback
home-assistant   |     debug_info.update_entity_discovery_data(self.hass, payload, self.entity_id)
home-assistant   |   File "/usr/src/homeassistant/homeassistant/components/mqtt/debug_info.py", line 89, in update_entity_discovery_data
home-assistant   |     entity_info = hass.data[DATA_MQTT_DEBUG_INFO]["entities"][entity_id]
home-assistant   | KeyError: 'sensor.0x00158d000402f92c_smoke_density'
home-assistant   | 
home-assistant   | 2021-01-03 19:21:07 ERROR (MainThread) [homeassistant.util.logging] Exception in discovery_callback when dispatching 'mqtt_discovery_updated_('sensor', '0x00158d000402f92c linkquality')': ({'availability': [{'topic': 'zigbee2mqtt/bridge/state'}], 'device': {'identifiers': ['zigbee2mqtt_0x00158d000402f92c'], 'manufacturer': 'Xiaomi', 'model': 'MiJia Honeywell smoke detector (JTYJ-GD-01LM/BW)', 'name': 'smoke_livingroom', 'sw_version': 'Zigbee2MQTT 1.17.0-dev'}, 'icon': 'mdi:signal', 'json_attributes_topic': 'zigbee2mqtt/smoke_livingroom', 'name': 'smoke_livingroom_linkquality', 'state_topic': 'zigbee2mqtt/smoke_livingroom', 'unique_id': '0x00158d000402f92c_linkquality_zigbee2mqtt', 'unit_of_measurement': 'lqi', 'value_template': '{{ value_json.linkquality }}', 'platform': 'mqtt'},)
home-assistant   | Traceback (most recent call last):
home-assistant   |   File "/usr/src/homeassistant/homeassistant/components/mqtt/__init__.py", line 1304, in discovery_callback
home-assistant   |     debug_info.update_entity_discovery_data(self.hass, payload, self.entity_id)
home-assistant   |   File "/usr/src/homeassistant/homeassistant/components/mqtt/debug_info.py", line 89, in update_entity_discovery_data
home-assistant   |     entity_info = hass.data[DATA_MQTT_DEBUG_INFO]["entities"][entity_id]
home-assistant   | KeyError: 'sensor.0x00158d000402f92c_linkquality'
```

Koenkk, 2021-01-03:
> I saw in Home Assistant these errors also about the smoke sensor.
Is it normal ?

Looks like https://github.com/Koenkk/zigbee2mqtt/issues/5150 (known HA issue).

> (bitron video in my case) ?

I don't know if this device uses a larger timeout (of 30 seconds like described in https://github.com/Koenkk/zigbee2mqtt/issues/5524#issuecomment-753630227). The only ones that do this for sure are the Xiaomi devices itself.

> how can I do that ?

[image]

> But to be honest I still need to wait like 15-30 seconds before the self test o sensibility change. We can't do nothing for that ? Even if the sensor is connected on a router and not directly on coordinator now ?

The smoke sensor only polls every 30 seconds, so no.

kevincaradant, 2021-01-03:
Alright :)

So if I don't have any error in Z2M using my routers and the poll is for sure 30s all the time on Smoke Sensor. I can't do nothing else to fix it (hardware restriction)

I think I can close this issue :)

Thank You

closed 2021-01-03 by kevincaradant