Koenkk/zigbee2mqtt #5526: TuYa valve TS0601: Problem setting eco_temperature value

kremik, 2021-01-03:
## What happened
When I set eco_temperature value via `zigbee2mqtt/FRIENDLY_NAME/set` it is ignored and not set. 
```
{
      "eco_temperature": 18
}
```

There is no way how to set eco_temperature. I have 2 valves of this type and happens with both of them.

There is nothing in the log, sometimes (but not exactly in the same time) I can see this error:
`Jan 03 13:52:55 raspberrypi4 npm[429]: zigbee-herdsman-converters:tuyaThermostat: NOT RECOGNIZED DP #115 with data {"status":6,"transid":152,"dp":115,"datatype":1,"fn":0,"data":{"type":"Buffer","data":[0]}}`

## What did you expect to happen

I expect to have eco_temperature set to 18, but the value remains the same as before.

## How to reproduce it (minimal and precise)
Send the set command with values above and see the devices response.

## Debug info
Zigbee2MQTT version: 1.16.2
Adapter hardware: CC2531
Adapter firmware version: 3.2.6
Accessory:
https://www.zigbee2mqtt.io/devices/TS0601_thermostat.html

kremik, 2021-01-03:
Sorry, I see it is fixed in 1.17.0 after upgrade.

closed 2021-01-03 by kremik

Xites, 2021-01-21:
> Sorry, I see it is fixed in 1.17.0 after upgrade.

Maybe, but now the decimals result in zero and a text format is needed.
{
      "eco_temperature": "18"
}
Result: 18

{
      "eco_temperature": 18
}
Result: 0

{
      "eco_temperature": "18.5"
}
Result: 0